kellerza/sunsynk #1: Inverter Model

w1tw0lf, 2021-11-11:
Would the code need to be adopted for 8kw unit ? I see you based the code on 5kw unit.

kellerza, 2021-11-16:
Likely not @w1tw0lf . The  protocol document I used was not specific on the 5kW and 8kW models.

Have you tried it yet?

w1tw0lf, 2021-11-16:
Not yet, as I am looking to go this route. Just want to get 8kw unit. 5kw unit might not be efficient for me. Just want something that works with home assistant. 

If I do, we can work on to get it to work? Know my way around code, just not expert.

kellerza, 2021-11-16:
Yes, if there is an issue we should be able to iron it out.

If you browse around Powerforum [here](https://powerforum.co.za/topic/8451-sunsynk-inverter-monitoring) you will see some people have the 8kW. Many of them use the Node-RED integration, but this Addon should be a bit easier to get up and running

w1tw0lf, 2021-11-16:
Will have a look once installed. Sure it will work as is or with some tweaks.  Have a look at my profile. Built a plug in for unifi/home assistant.

kellerza, 2022-01-25:
Hi @w1tw0lf did you get your 8kW sunsynk installed?

w1tw0lf, 2022-01-25:
Hi. Not yet. Planning is for May.

kellerza, 2022-01-25:
Good luck with the install and feel free to open a new issue if needed!

closed 2022-01-25 by kellerza