kellerza/sunsynk #2: Using Deye Inverter

maumauns, 2021-12-11:
As I know is the same inverter rebranded.
But im getting some errors.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-12-11 17:25:11,574 INFO    Loading HASS OS configuration
2021-12-11 17:25:11,577 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2021-12-11 17:25:11,578 INFO    Filter *step used for ['battery_soc']
2021-12-11 17:25:11,579 DEBUG   Using selector: EpollSelector
2021-12-11 17:25:11,583 DEBUG   Connecting.
2021-12-11 17:25:11,619 DEBUG   Client connected to modbus server
2021-12-11 17:25:11,619 INFO    Protocol made connection.
2021-12-11 17:25:11,622 INFO    Connected to /dev/ttyUSB0
2021-12-11 17:25:11,623 DEBUG   send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
2021-12-11 17:25:11,626 DEBUG   Adding transaction 1
2021-12-11 17:25:11,709 DEBUG   recv: 0x1 0x3 0xa 0x32 0x31 0x30 0x36 0x32 0x39 0x37 0x30 0x32 0x30 0xa 0x1d 0xff
2021-12-11 17:25:11,711 DEBUG   Getting Frame - 0x3 0xa 0x32 0x31 0x30 0x36 0x32 0x39 0x37 0x30 0x32 0x30
2021-12-11 17:25:11,712 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2021-12-11 17:25:11,714 DEBUG   Frame advanced, resetting header!!
2021-12-11 17:25:11,715 DEBUG   Getting transaction 1
2021-12-11 17:25:11,719 DEBUG   Request registers: [3, 4, 5, 6, 7] glen=5. Response [12849, 12342, 12857, 14128, 12848] len=5. regs={3: 12849, 4: 12342, 5: 12857, 6: 14128, 7: 12848}
2021-12-11 17:25:11,720 INFO    ############################################################
2021-12-11 17:25:11,721 INFO                   SMA serial number '2106297020'
2021-12-11 17:25:11,722 INFO    ############################################################
2021-12-11 17:25:11,749 INFO    MQTT: Connection successful
2021-12-11 17:25:12,235 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/total_pv_power/config, {"name": "Total PV Power", "stat_t": "SUNSYNK/status/2106297020/total_pv_power", "unit_of_meas": "kWh", "uniq_id": "2106297020_total_pv_power", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-11 17:25:12,275 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/total_battery_charge/config, {"name": "Total Battery Charge", "stat_t": "SUNSYNK/status/2106297020/total_battery_charge", "unit_of_meas": "kWh", "uniq_id": "2106297020_total_battery_charge", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-11 17:25:12,283 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/total_battery_discharge/config, {"name": "Total Battery Discharge", "stat_t": "SUNSYNK/status/2106297020/total_battery_discharge", "unit_of_meas": "kWh", "uniq_id": "2106297020_total_battery_discharge", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-11 17:25:12,297 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/total_grid_export/config, {"name": "Total Grid Export", "stat_t": "SUNSYNK/status/2106297020/total_grid_export", "unit_of_meas": "kWh", "uniq_id": "2106297020_total_grid_export", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-11 17:25:12,310 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/total_grid_import/config, {"name": "Total Grid Import", "stat_t": "SUNSYNK/status/2106297020/total_grid_import", "unit_of_meas": "kWh", "uniq_id": "2106297020_total_grid_import", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-11 17:25:12,323 DEBUG   PUBLISH 1R homeassistant/sensor/2106297020/battery_soc/config, {"name": "Battery SOC", "stat_t": "SUNSYNK/status/2106297020/battery_soc", "unit_of_meas": "%", "uniq_id": "2106297020_battery_soc", "dev": {"ids": ["sunsynk_2106297020"], "name": "Sunsynk Inverter 2106297020", "mdl": "Inverter 2106297020", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "battery"}
2021-12-11 17:25:13,343 DEBUG   send: 0x1 0x3 0x0 0x48 0x0 0xb 0x84 0x1b
2021-12-11 17:25:13,350 DEBUG   Adding transaction 1
2021-12-11 17:25:13,441 DEBUG   recv: 0x1 0x3 0x16 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0x0 0x28 0x2 0x3d 0x17 0x6d 0x0 0x0 0x1 0x76 0x0 0x0 0x8a 0x19 0xff
2021-12-11 17:25:13,442 DEBUG   Frame check failed, ignoring!!
2021-12-11 17:25:13,443 DEBUG   Resetting frame - Current Frame in buffer - 0xff 0x1 0x3 0x16 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0x0 0x28 0x2 0x3d 0x17 0x6d 0x0 0x0 0x1 0x76 0x0 0x0 0x8a 0x19 0xff
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 214, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 185, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

maumauns, 2021-12-11:
I think i solved the problem, installed modbus tcp addon and now its working.

kellerza, 2022-01-25:
Great, the mbusd addon in this repo might have a larger value for `TIMEOUT` (which you can also set in the settings)

closed 2022-01-25 by kellerza