kellerza/sunsynk #3: Connection Issues

ApoclypseBoy, 2021-12-20:
Hi Kellerza

I'm Having issues with the sensors not being available.

I installed the plugin and configured it on a Pi3b and home assistant core image from them.

The log shows that I'm connected successfully and I did get some output to the dashboard after setting up a card in lovalace.

Now the app just runs for a second or two once you start it ( You can see the cpu load ) and than just stops.

Please advise on what to do from here.

TIA

AB

kellerza, 2022-01-25:
Difficult to say without any logs. Would need the logs from the addon at least, you get it in the supervisor. 

Add a line with three backticks (```) before and after your logs to format it

dirkackerman, 2022-01-31:
Kellerza,

I'm running your main HEAD from a docker container but seems to be experiencing comunication issues with the Sunsynk 8kW inverter. It seems as if the unit comms times out:

```
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-01-31 22:40:59,128 INFO    Loading HASS OS configuration
2022-01-31 22:40:59,129 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2022-01-31 22:40:59,130 INFO    Filter *step used for ['battery_soc']
2022-01-31 22:40:59,139 INFO    Protocol made connection.
2022-01-31 22:40:59,140 INFO    Connected to /dev/ttyUSB0
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 185, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 146, in main
    await SUNSYNK.read([ssdefs.serial])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(grp[0], glen, unit=self.unit)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
```
docker-compose file entry:
```
    hass_sunsynk:
        container_name: hass_sunsynk
        build:
            context: ../addOnRepos/sunsynk/hass-addon-sunsynk/
            args:
                BUILD_FROM: homeassistant/amd64-base-python:latest
        depends_on:
            - mqtt
            - dns
        networks:
            saru:
                ipv4_address: 172.24.0.13
        dns: 172.24.0.2
        env_file:
            - hosting.env
        volumes:
            - "/dev/ttyUSB0:/dev/ttyUSB0:rw"
            - "./sunsynk/options.json:/data/options.json:ro"
        privileged: true
        restart: unless-stopped
```
and options.json (id and host name replaced with #):

```

    "PORT": "/dev/ttyUSB0",
    "PORT_ADDRESS": "",
    "SUNSYNK_ID": "#########",
    "SENSORS": ["total_pv_power","total_battery_charge", "total_battery_discharge", "total_grid_export", "total_grid_import", "battery_soc"],
    "SENSOR_PREFIX": "ss9n",
    "TIMEOUT": 10,
    "MQTT_HOST": "##########",
    "MQTT_PORT": 1883,
    "MQTT_USERNAME": "",
    "MQTT_PASSWORD": "",
    "DEBUG": 1
  }

```

I've connected the pins 1 and 2 to a RS485 USB converter as per your wiring diagram to the RS485 port on the inverter (the BMS is connected to the CAN port).

Do you have any tips to debug this comms issue?

Thanks

kellerza, 2022-02-01:
Thanks for confirming @dirkackerman !

Just to confirm: you used the port labelled **RS485** on your Sunsynk 8.8kW, with the same wiring I have on the main readme (Pin 1 - B, Pin 2 - A) 

I like your referenced [converter](https://www.robotics.org.za/index.php?route=product/product&product_id=5947&search=rs48), since it has a GND pin as well. Did you connect GND? (might be pin3 from what I've seen)

dirkackerman, 2022-02-01:
I've created a [pull request](https://github.com/kellerza/sunsynk/pull/8) with this info should you like to review it.

closed 2022-02-01 by kellerza