kellerza/sunsynk #4: Unable to read sensor data

lispwarez, 2021-12-21:
Hi

I am able to connect to the inverter as I can read the serial number on initial connection but it fails to read any sensor data. I have enabled debug logging and the results are below. Please advise on how to fix

```
2021-12-21 11:09:15,707 INFO    Loading HASS OS configuration
2021-12-21 11:09:15,709 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2021-12-21 11:09:15,709 INFO    Filter *step used for ['battery_soc']
2021-12-21 11:09:15,710 INFO    Filter last used for ['grid_ct_load', 'inverter_output', 'load_power']
2021-12-21 11:09:15,711 DEBUG   Using selector: EpollSelector
2021-12-21 11:09:15,714 DEBUG   Connecting.
2021-12-21 11:09:15,752 DEBUG   Client connected to modbus server
2021-12-21 11:09:15,753 INFO    Protocol made connection.
2021-12-21 11:09:15,755 INFO    Connected to /dev/ttyUSB0
2021-12-21 11:09:15,756 DEBUG   send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
2021-12-21 11:09:15,758 DEBUG   Adding transaction 1
2021-12-21 11:09:15,821 DEBUG   recv: 0x1 0x3 0xa 0x32 0x31 0x30 0x36 0x32 0x39 0x37 0x32 0x32 0x38 0xaa 0x1b 0xff
2021-12-21 11:09:15,822 DEBUG   Getting Frame - 0x3 0xa 0x32 0x31 0x30 0x36 0x32 0x39 0x37 0x32 0x32 0x38
2021-12-21 11:09:15,822 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2021-12-21 11:09:15,823 DEBUG   Frame advanced, resetting header!!
2021-12-21 11:09:15,824 DEBUG   Getting transaction 1
2021-12-21 11:09:15,826 DEBUG   Request registers: [3, 4, 5, 6, 7] glen=5. Response [12849, 12342, 12857, 14130, 12856] len=5. regs={3: 12849, 4: 12342, 5: 12857, 6: 14130, 7: 12856}
2021-12-21 11:09:15,827 INFO    ############################################################
2021-12-21 11:09:15,827 INFO                   SMA serial number '2106297228'
2021-12-21 11:09:15,828 INFO    ############################################################
2021-12-21 11:09:15,864 INFO    MQTT: Connection successful
2021-12-21 11:09:16,334 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/total_pv_power/config, {"name": "Total PV Power", "stat_t": "SUNSYNK/status/2106297228/total_pv_power", "unit_of_meas": "kWh", "uniq_id": "2106297228_total_pv_power", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-21 11:09:16,357 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/total_battery_charge/config, {"name": "Total Battery Charge", "stat_t": "SUNSYNK/status/2106297228/total_battery_charge", "unit_of_meas": "kWh", "uniq_id": "2106297228_total_battery_charge", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-21 11:09:16,362 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/total_battery_discharge/config, {"name": "Total Battery Discharge", "stat_t": "SUNSYNK/status/2106297228/total_battery_discharge", "unit_of_meas": "kWh", "uniq_id": "2106297228_total_battery_discharge", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-21 11:09:16,367 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/total_grid_export/config, {"name": "Total Grid Export", "stat_t": "SUNSYNK/status/2106297228/total_grid_export", "unit_of_meas": "kWh", "uniq_id": "2106297228_total_grid_export", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-21 11:09:16,372 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/total_grid_import/config, {"name": "Total Grid Import", "stat_t": "SUNSYNK/status/2106297228/total_grid_import", "unit_of_meas": "kWh", "uniq_id": "2106297228_total_grid_import", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "energy", "stat_cla": "total_increasing"}
2021-12-21 11:09:16,378 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/battery_soc/config, {"name": "Battery SOC", "stat_t": "SUNSYNK/status/2106297228/battery_soc", "unit_of_meas": "%", "uniq_id": "2106297228_battery_soc", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "battery"}
2021-12-21 11:09:16,383 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/grid_ct_load/config, {"name": "Grid CT load", "stat_t": "SUNSYNK/status/2106297228/grid_ct_load", "unit_of_meas": "W", "uniq_id": "2106297228_grid_ct_load", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "power"}
2021-12-21 11:09:16,388 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/inverter_output/config, {"name": "Inverter Output", "stat_t": "SUNSYNK/status/2106297228/inverter_output", "unit_of_meas": "W", "uniq_id": "2106297228_inverter_output", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "power"}
2021-12-21 11:09:16,393 DEBUG   PUBLISH 1R homeassistant/sensor/2106297228/load_power/config, {"name": "Load Power", "stat_t": "SUNSYNK/status/2106297228/load_power", "unit_of_meas": "W", "uniq_id": "2106297228_load_power", "dev": {"ids": ["sunsynk_2106297228"], "name": "Sunsynk Inverter 2106297228", "mdl": "Inverter 2106297228", "mf": "Sunsynk"}, "exp_aft": 301, "dev_cla": "power"}
2021-12-21 11:09:17,402 DEBUG   send: 0x1 0x3 0x0 0x48 0x0 0xb 0x84 0x1b
2021-12-21 11:09:17,407 DEBUG   Adding transaction 1
2021-12-21 11:09:17,472 DEBUG   recv: 0x1 0x3 0x16 0x0 0x14 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x27 0x0 0x0 0x2 0xbf 0x13 0x89 0x0 0x0 0x0 0x5 0x0 0x0 0x1f 0x7c
2021-12-21 11:09:17,474 DEBUG   Frame check failed, ignoring!!
2021-12-21 11:09:17,475 DEBUG   Resetting frame - Current Frame in buffer - 0xff 0x1 0x3 0x16 0x0 0x14 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x27 0x0 0x0 0x2 0xbf 0x13 0x89 0x0 0x0 0x0 0x5 0x0 0x0 0x1f 0x7c
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 214, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 185, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
```

kellerza, 2022-01-25:
Can you try adding a TIMEOUT of 10 seconds to your config? The default is 5

kellerza, 2022-01-29:
The new default will be 10

closed 2022-01-29 by kellerza