kellerza/sunsynk #5: modbus tcp serial bridge?

pepelillo, 2022-01-13:
I would like to know if it's possible to use a serial bridge with this addon.
Something like using and ESP and a ttl to rs485 converter and a ESPeasy firmware or similar.

I've found this project https://github.com/pawelka/sofarsolar_esp_rs485_logger with similar approach but using firmware for sofar inverter. would like to use a simple bridge to be used.
Some ideas?
Thanks

kellerza, 2022-01-25:
This sunsynk repo includes [mbusd](https://github.com/kellerza/sunsynk/tree/main/hass-addon-mbusd), which is a Modbus TCP to RTU bridge

Searching for an esp variant, I've found this
https://github.com/emelianov/modbus-esp8266/tree/master/examples/Bridge

The author discusses it here as well: https://github.com/emelianov/modbus-esp8266/issues/19

Please let us know if you found a working solution!

kellerza, 2022-01-27:
For mbusd mentioned above you can use any Raspberry Pi. If you run it without the HomeAssistant OS, you could even use a Model 1B

The [USR-W630](https://reliablestore.co.za/products/usr-w630-industrial-serial-to-wifi-and-ethernet-converter-supports-2-ethernet-ports-modbus-rtu-dc-adapter-and-terminal-power-supply-5-36-v?variant=40024388108449&sfdr_ptcid=28846_617_672307057&sfdr_hash=412ad5f003960dda27df1c5ac24f48e6) was also recommended on the Powerforum.

closed 2022-01-27 by kellerza