kellerza/sunsynk #7: Wiring Diagram RS485 RJ45

tuanha2000vn, 2022-01-29:
[image]
[image]
Hi, I'm trying to connect Home Assistant to Deye inverter using USB-RS485-RJ45 cable.

What the correct way to wire the cable?

Option 1:  USB-RS485 A/D+ to RJ45 1 - USB-RS485 B/D1 to RJ45 8
Option 2:  USB-RS485 A/D+ to RJ45 8 - USB-RS485 B/D1 to RJ45 1

Thank you.

tuanha2000vn, 2022-01-29:
[image]

Found this on the user manual, that mean this?
RJ45 1 and 8 to RS485 A/D+
RJ45 2 and 7 to RS485 B/D-

kellerza, 2022-01-29:
I documented my wiring [here](https://github.com/kellerza/sunsynk#hardware)

Below is a 568A plug...

[image]

kellerza, 2022-02-01:
The Readme now has some more info...

closed 2022-02-01 by kellerza