kellerza/sunsynk #10: USB to RS485 adapter issues?

reedy, 2022-02-03:
**Issue related to**
[ ] New Sensors / Sensors not working (existing definitions available [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)
[x] Sunsynk Home Assistant Add-On
[ ] mbusd Home Assistant Add-On
[ ] Python sunsynk library

**Describe the issue/bug**
Not so much a bug (necessarily), but I guess a support request.

Recently had a Sunsynk 3.6 kWh inverter installed.

I bought one of https://www.ebay.co.uk/itm/393575748700 "RS485-150U" hoping it would work (being premade), but when starting the addon it seems to timeout.

I don't know if it's due to the wrong pinout on the RJ45 end, but I did try `/dev/ttyACM0`, `/dev/ttyAMA0` and `/dev/serial1`; all of which failed similarly.

I have plugged it into the BMS485 (top left) port; it seems for a lot of things the 5.5 and 3.6 are similar.

Got a different adapter coming tomorrow from Amazon - https://smile.amazon.co.uk/dp/B06Y1JTGZX which will make it easier to make sure the pinout is right, so this request might be invalid...

Happy to try a few other things (and/or doing this outside Home Assistant.. On a vanilla Ubunutu/Raspberry Pi OS/Debian rpi install), such as the library as a standalone if it helps.

Thanks!

**Expected behavior**
Not crash on startup? :)

**Home Assistant version (if applicable)**
2022.2.0

**Additional context**
Add any other context about the problem here.

**Logs**

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-03 18:20:43,458 INFO    Loading HASS OS configuration
2022-02-03 18:20:43,460 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2022-02-03 18:20:43,460 INFO    Filter *step used for ['battery_soc']
2022-02-03 18:20:43,469 INFO    Protocol made connection.
2022-02-03 18:20:43,470 INFO    Connected to /dev/serial1
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 185, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 146, in main
    await SUNSYNK.read([ssdefs.serial])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(grp[0], glen, unit=self.unit)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

dirkackerman, 2022-02-03:
I had simular output [here](https://github.com/kellerza/sunsynk/issues/3#issuecomment-1026198502).

Check your connections and make sure you are using the correct device in linux. I believe USD to serial converters should generally be /dev/ttyUSBx in most distros. Maybe unplug the device and reconnect it and compare the `ls /dev` output to see the added device's name?

reedy, 2022-02-03:
Thanks for the reply. The dev entry is definitely `/dev/ttyACM0` for it.

[image]

^ is from Home Assistant. Using a vanilla RasberryPI OS image adding/removing tht USB cable makes `/dev/ttyAMA0` remove and disappear.

So it's definitely the "correct" logical device, but I'm not sure the USB device is completely correct for this purpose. The fact it's AMA (from a bit of Googling) suggests it might not be a serial console like I'm thinking it is (it was a punt based on availability on ebay) - https://arduino.stackexchange.com/questions/19648/difference-between-dev-ttyacm0-and-dev-ttys0-arduino-ide-ports-under-linux

>/dev/ttyACM0 is a USB [communication device](https://en.wikipedia.org/wiki/USB_communications_device_class) (CDC) of sub-type "abstract control model" (ACM). That is what the Arduino is.

Which probably means I should stop wasting time on it :). Especially as what should be the right device should be here tomorrow. At what it cost, it's probably not even worth bothering to return with the cost of the return postage...

Will follow up again tomorrow when the delivery person has been!

reedy, 2022-02-03:
I guess at this point, my only feedback might this part of the code might benefit from a bit better error handling. Trap it and show a better error message, before showing the stack trace anyway

kellerza, 2022-02-04:
Will improve that error if it fails on reading the serial number!

closed 2022-02-04 by kellerza

reedy, 2022-02-04:
Yay, so with https://smile.amazon.co.uk/dp/B06Y1JTGZX and me making up a new (longer) cable.. it works!

Just need to try and work out the dashboards/sensors now (I'm new to HA too)...

reedy, 2022-02-04:
Seem to be having similar issues to https://github.com/kellerza/sunsynk/issues/13 even only with 1 or 2 sensors

```
PORT: /dev/ttyUSB0
PORT_ADDRESS: ''
SUNSYNK_ID: '210615XXXX'
SENSORS:
  - battery_voltage
  - battery_soc
SENSOR_PREFIX: Sunsynk
TIMEOUT: 10
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: mqtt
MQTT_PASSWORD: redacted
DEBUG: 0
```

And then the logs...

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-04 17:28:56,880 INFO    Loading HASS OS configuration
2022-02-04 17:28:56,881 INFO    Filter *step used for ['battery_voltage', 'battery_soc']
2022-02-04 17:28:56,896 INFO    Protocol made connection.
2022-02-04 17:28:56,899 INFO    Connected to /dev/ttyUSB0
2022-02-04 17:28:56,955 INFO    ############################################################
2022-02-04 17:28:56,956 INFO                   SMA serial number '210615XXXX'
2022-02-04 17:28:56,957 INFO    ############################################################
2022-02-04 17:28:56,975 INFO    MQTT: Connection successful
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 185, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 156, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(grp[0], glen, unit=self.unit)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

reedy, 2022-02-06:
Same on `2022.02.05-0.1.0`...

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-06 13:29:37,754 INFO    Loading HASS OS configuration
2022-02-06 13:29:37,756 INFO    Filter *step used for ['battery_voltage']
2022-02-06 13:29:37,756 INFO    Filter *last used for ['battery_soc']
2022-02-06 13:29:37,764 INFO    Protocol made connection.
2022-02-06 13:29:37,764 INFO    Connected to /dev/ttyUSB0
2022-02-06 13:29:37,817 INFO    ############################################################
2022-02-06 13:29:37,818 INFO                Inverter serial number '210615XXXX'
2022-02-06 13:29:37,818 INFO    ############################################################
2022-02-06 13:29:37,856 INFO    MQTT: Connection successful
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 204, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 170, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

reedy, 2022-02-08:
And after https://github.com/kellerza/sunsynk/commit/dea9864091d006dbd7438b8e4bb08e996cc477c2

```
2022-02-08 01:26:05,458 INFO    MQTT: Connection successful
2022-02-08 01:26:26,475 ERROR   Timeout on initial read. Sensors: ['Battery voltage:step:mean', 'Battery SOC:last']
2022-02-08 01:26:46,551 ERROR   Timeout reading single sensor: Battery SOC:last
2022-02-08 01:27:08,586 ERROR   TimeOut 
2022-02-08 01:27:29,611 ERROR   TimeOut 
2022-02-08 01:27:56,654 ERROR   TimeOut 
2022-02-08 01:28:17,679 ERROR   TimeOut
```

Think I need to turn up the debug settings

reedy, 2022-02-08:
With `DEBUG: 2`

```
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/usr/local/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/usr/local/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/usr/local/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: %d format: a number is required, not NoneType
Call stack:
  File "/usr/src/app/./run.py", line 214, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 629, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 596, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 1882, in _run_once
    handle._run()
  File "/usr/local/lib/python3.9/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/app/./run.py", line 178, in main
    await SUNSYNK.read([fil.sensor])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 116, in read
    update_sensors(sensors, regs)
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sensor.py", line 124, in update_sensors
    sen.update_value()
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sensor.py", line 60, in update_value
    _LOGGER.debug(
Message: '%s low=%d high=%d value=%d%s'
Arguments: ('Battery SOC', 46, 0, None, '%')
2022-02-08 01:31:33,290 DEBUG   PUBLISH 0 SUNSYNK/status/2106154277/battery_voltage, 52.08
2022-02-08 01:31:33,295 DEBUG   PUBLISH 0 SUNSYNK/status/2106154277/battery_soc, 46
2022-02-08 01:31:33,300 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x2 0x74 0x2d
2022-02-08 01:31:33,301 DEBUG   Adding transaction 1
2022-02-08 01:31:33,348 DEBUG   recv: 0x1 0x3 0x4 0x14 0x58 0x0 0x2e 0xfe 0xc
2022-02-08 01:31:33,348 DEBUG   Getting Frame - 0x3 0x4 0x14 0x58 0x0 0x2e
2022-02-08 01:31:33,349 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:31:33,349 DEBUG   Frame advanced, resetting header!!
2022-02-08 01:31:33,349 DEBUG   Getting transaction 1
2022-02-08 01:31:33,351 DEBUG   Request registers: [183, 184] glen=2. Response [5208, 46] len=2. regs={183: 5208, 184: 46}
2022-02-08 01:31:33,351 DEBUG   Battery voltage low=5208 high=0 value=52V
2022-02-08 01:31:33,352 DEBUG   Battery SOC low=46 high=0 value=46%
2022-02-08 01:31:33,352 DEBUG   Battery SOC:last over 1 samples = 46
2022-02-08 01:31:33,352 DEBUG   PUBLISH 0 SUNSYNK/status/2106154277/battery_soc, 46
2022-02-08 01:31:34,307 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:31:34,310 DEBUG   Adding transaction 1
2022-02-08 01:31:34,356 DEBUG   recv: 0x40 0x20 0x20 0x11 0x65 0xf3 0xff
2022-02-08 01:31:34,357 DEBUG   Frame check failed, ignoring!!
2022-02-08 01:31:34,358 DEBUG   Resetting frame - Current Frame in buffer - 0x40 0x20 0x20 0x11 0x65 0xf3 0xff
2022-02-08 01:31:54,335 ERROR   TimeOut 
2022-02-08 01:31:54,339 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:31:54,341 DEBUG   Adding transaction 1
2022-02-08 01:31:54,384 DEBUG   recv: 0x1 0x3 0x2 0x14 0x59 0x77 0x7e
2022-02-08 01:31:54,385 DEBUG   Getting Frame - 0x3 0x2 0x14 0x59
2022-02-08 01:31:54,385 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:31:54,387 DEBUG   Frame advanced, resetting header!!
2022-02-08 01:31:54,387 DEBUG   Getting transaction 1
2022-02-08 01:31:54,390 DEBUG   Request registers: [183] glen=1. Response [5209] len=1. regs={183: 5209}
2022-02-08 01:31:54,390 DEBUG   Battery voltage low=5209 high=0 value=52V
2022-02-08 01:31:55,345 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:31:55,347 DEBUG   Adding transaction 1
2022-02-08 01:31:55,394 DEBUG   recv: 0x1 0x3 0x2 0x14 0x59 0x77 0x7e
2022-02-08 01:31:55,395 DEBUG   Getting Frame - 0x3 0x2 0x14 0x59
2022-02-08 01:31:55,396 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:31:55,396 DEBUG   Frame advanced, resetting header!!
[... 16 lines cut ...]
2022-02-08 01:32:16,438 DEBUG   Request registers: [183] glen=1. Response [5209] len=1. regs={183: 5209}
2022-02-08 01:32:16,439 DEBUG   Battery voltage low=5209 high=0 value=52V
2022-02-08 01:32:17,386 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:32:17,388 DEBUG   Adding transaction 1
2022-02-08 01:32:17,433 DEBUG   recv: 0x1 0x3 0x2 0x14 0x59 0x77 0x7e
2022-02-08 01:32:17,434 DEBUG   Getting Frame - 0x3 0x2 0x14 0x59
2022-02-08 01:32:17,434 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:32:17,435 DEBUG   Frame advanced, resetting header!!
2022-02-08 01:32:17,435 DEBUG   Getting transaction 1
2022-02-08 01:32:17,438 DEBUG   Request registers: [183] glen=1. Response [5209] len=1. regs={183: 5209}
2022-02-08 01:32:17,438 DEBUG   Battery voltage low=5209 high=0 value=52V
2022-02-08 01:32:18,392 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:32:18,394 DEBUG   Adding transaction 1
2022-02-08 01:32:18,444 DEBUG   recv: 0x20 0x20 0x20 0x51 0x75 0xf9 0xff
2022-02-08 01:32:18,445 DEBUG   Frame check failed, ignoring!!
2022-02-08 01:32:18,445 DEBUG   Resetting frame - Current Frame in buffer - 0x20 0x20 0x20 0x51 0x75 0xf9 0xff
```

Ok... So with DEBUG enabled, things work?

This is obviously good, but a bit strange... Some race type conditions, and debug mode helps there?

[image]

Any ideas @kellerza? :)

reedy, 2022-02-08:
Logs get truncated quite quickly..

```
2022-02-08 01:37:35,692 DEBUG   send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
2022-02-08 01:37:35,694 DEBUG   Adding transaction 1
2022-02-08 01:37:35,746 DEBUG   recv: 0x1 0x3 0xa 0x32 0x31 0x30 0x36 0x31 0x35 0x34 0x32 0x37 0x37 0xf9 0x39 0xff
2022-02-08 01:37:35,748 DEBUG   Getting Frame - 0x3 0xa 0x32 0x31 0x30 0x36 0x31 0x35 0x34 0x32 0x37 0x37
2022-02-08 01:37:35,749 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:37:35,750 DEBUG   Frame advanced, resetting header!!
2022-02-08 01:37:35,751 DEBUG   Getting transaction 1
2022-02-08 01:37:35,754 DEBUG   Request registers: [3, 4, 5, 6, 7] glen=5. Response [12849, 12342, 12597, 13362, 14135] len=5. regs={3: 12849, 4: 12342, 5: 12597, 6: 13362, 7: 14135}
2022-02-08 01:37:35,755 INFO    ############################################################
2022-02-08 01:37:35,755 INFO                Inverter serial number '210615XXXX'
2022-02-08 01:37:35,756 INFO    ############################################################
2022-02-08 01:37:35,789 INFO    MQTT: Connection successful
2022-02-08 01:37:36,264 DEBUG   PUBLISH 1R SUNSYNK/status/210615XXXX/availability, online
2022-02-08 01:37:36,276 DEBUG   publish homeassistant/sensor/210615XXXX/total_pv_power/config
2022-02-08 01:37:36,277 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/total_pv_power/config, {"unique_id": "210615XXXX_total_pv_power", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/total_pv_power", "name": "Sunsynk Total PV Power", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing"}
2022-02-08 01:37:36,283 DEBUG   publish homeassistant/sensor/210615XXXX/total_battery_charge/config
2022-02-08 01:37:36,284 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/total_battery_charge/config, {"unique_id": "210615XXXX_total_battery_charge", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/total_battery_charge", "name": "Sunsynk Total Battery Charge", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing"}
2022-02-08 01:37:36,288 DEBUG   publish homeassistant/sensor/210615XXXX/total_battery_discharge/config
2022-02-08 01:37:36,289 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/total_battery_discharge/config, {"unique_id": "210615XXXX_total_battery_discharge", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/total_battery_discharge", "name": "Sunsynk Total Battery Discharge", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing"}
2022-02-08 01:37:36,293 DEBUG   Rx retained msg: topic=homeassistant/sensor/210615XXXX/battery_soc/config -- device=210615XXXX
2022-02-08 01:37:36,293 DEBUG   Rx retained msg: topic=homeassistant/sensor/210615XXXX/battery_voltage/config -- device=210615XXXX
2022-02-08 01:37:36,294 WARNING Removing HASS MQTT discovery info homeassistant/sensor/210615XXXX/battery_voltage/config
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 954, in _bootstrap_inner
2022-02-08 01:37:36,308 DEBUG   publish homeassistant/sensor/210615XXXX/total_grid_export/config
2022-02-08 01:37:36,309 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/total_grid_export/config, {"unique_id": "210615XXXX_total_grid_export", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/total_grid_export", "name": "Sunsynk Total Grid Export", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing"}
    self.run()
  File "/usr/local/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 3452, in _thread_main
2022-02-08 01:37:36,318 DEBUG   publish homeassistant/sensor/210615XXXX/total_grid_import/config
2022-02-08 01:37:36,319 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/total_grid_import/config, {"unique_id": "210615XXXX_total_grid_import", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/total_grid_import", "name": "Sunsynk Total Grid Import", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "energy", "unit_of_measurement": "kWh", "state_class": "total_increasing"}
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python3.9/site-packages/paho/mqtt/client.py", line 1181, in loop
[... cut ...]
Message: '%s low=%d high=%d value=%d%s'
Arguments: ('Total Grid Import', 772, 0, None, 'kWh')
2022-02-08 01:39:56,598 DEBUG   send: 0x1 0x3 0x0 0x60 0x0 0x2 0xc4 0x15
2022-02-08 01:39:56,600 DEBUG   Adding transaction 1
2022-02-08 01:39:56,653 DEBUG   recv: 0x1 0x3 0x4 0x6 0x1a 0x0 0x0 0xdb 0x7c
2022-02-08 01:39:56,654 DEBUG   Frame check failed, ignoring!!
2022-02-08 01:39:56,655 DEBUG   Resetting frame - Current Frame in buffer - 0xff 0x1 0x3 0x4 0x6 0x1a 0x0 0x0 0xdb 0x7c
2022-02-08 01:40:16,626 ERROR   TimeOut
```

kellerza, 2022-02-08:
What happens with only battery_voltage as a start. Frist try without debug and then enable it later

reedy, 2022-02-08:
As per https://github.com/kellerza/sunsynk/issues/10#issuecomment-1030200209

```
  - battery_voltage
  - battery_soc
```

These without debug mode didn't work. See https://github.com/kellerza/sunsynk/issues/10#issuecomment-1032118584 and then with `DEBUG` set we got https://github.com/kellerza/sunsynk/issues/10#issuecomment-1032122302

I'll try just `battery_voltage` and no debug...

reedy, 2022-02-08:
Just `battery_voltage` and no debug:

```
2022-02-08 01:54:45,332 INFO    MQTT: Connection successful
2022-02-08 01:54:45,832 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/total_pv_power/config
2022-02-08 01:54:45,833 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/total_battery_charge/config
2022-02-08 01:54:45,835 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/total_battery_discharge/config
2022-02-08 01:54:45,836 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/total_grid_export/config
2022-02-08 01:54:45,837 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/total_grid_import/config
2022-02-08 01:54:45,838 WARNING Removing HASS MQTT discovery info homeassistant/sensor/2106154277/battery_soc/config
2022-02-08 01:55:06,353 ERROR   Timeout on initial read. Sensors: ['Battery voltage:step:mean']
2022-02-08 01:55:26,431 ERROR   TimeOut 
2022-02-08 01:55:47,456 ERROR   TimeOut
```

etc etc.

Adding back `DEBUG: 2`, we start getting the voltage being reported...

[image]

```
2022-02-08 01:58:55,969 INFO    MQTT: Connection successful
2022-02-08 01:58:56,461 DEBUG   PUBLISH 1R SUNSYNK/status/210615XXXX/availability, online
2022-02-08 01:58:56,485 DEBUG   publish homeassistant/sensor/210615XXXX/battery_voltage/config
2022-02-08 01:58:56,486 DEBUG   PUBLISH 1R homeassistant/sensor/210615XXXX/battery_voltage/config, {"unique_id": "210615XXXX_battery_voltage", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/battery_voltage", "name": "Sunsynk Battery voltage", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "device_class": "voltage", "unit_of_measurement": "V"}
2022-02-08 01:58:56,492 DEBUG   publish homeassistant/select/210615XXXX/system_mode/config
2022-02-08 01:58:56,493 DEBUG   PUBLISH 1R homeassistant/select/210615XXXX/system_mode/config, {"unique_id": "210615XXXX_system_mode", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/system_mode", "name": "Sunsynk System Mode", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "command_topic": "SUNSYNK/status/210615XXXX/system_mode_set", "options": ["UPDATE"]}
2022-02-08 01:58:56,498 DEBUG   publish homeassistant/select/210615XXXX/system_mode_voltages/config
2022-02-08 01:58:56,499 DEBUG   PUBLISH 1R homeassistant/select/210615XXXX/system_mode_voltages/config, {"unique_id": "210615XXXX_system_mode_voltages", "device": {"identifiers": ["210615XXXX"], "manufacturer": "Sunsynk", "model": "Inverter 210615XXXX", "name": "Sunsynk Inverter 210615XXXX"}, "state_topic": "SUNSYNK/status/210615XXXX/system_mode_voltages", "name": "Sunsynk System Mode Voltages", "availability": [{"topic": "SUNSYNK/status/210615XXXX/availability"}], "command_topic": "SUNSYNK/status/210615XXXX/system_mode_voltages_set", "options": ["UPDATE"]}
2022-02-08 01:58:56,503 DEBUG   Rx retained msg: topic=homeassistant/sensor/210615XXXX/battery_voltage/config -- device=210615XXXX
2022-02-08 01:58:56,504 DEBUG   Rx retained msg: topic=homeassistant/select/210615XXXX/system_mode/config -- device=210615XXXX
2022-02-08 01:58:56,505 DEBUG   Rx retained msg: topic=homeassistant/select/210615XXXX/system_mode_voltages/config -- device=210615XXXX
2022-02-08 01:58:57,007 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:58:57,009 DEBUG   Adding transaction 1
2022-02-08 01:58:57,055 DEBUG   recv: 0x1 0x3 0x2 0x14 0x64 0xb6 0xaf 0xff
2022-02-08 01:58:57,056 DEBUG   Frame check failed, ignoring!!
2022-02-08 01:58:57,056 DEBUG   Resetting frame - Current Frame in buffer - 0xff 0x1 0x3 0x2 0x14 0x64 0xb6 0xaf 0xff
2022-02-08 01:59:17,034 ERROR   Timeout on initial read. Sensors: ['Battery voltage:step:mean']
2022-02-08 01:59:17,048 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:59:17,049 DEBUG   Adding transaction 1
2022-02-08 01:59:17,094 DEBUG   recv: 0x1 0x3 0x2 0x14 0x64 0xb6 0xaf 0xff
2022-02-08 01:59:17,095 DEBUG   Getting Frame - 0x3 0x2 0x14 0x64
2022-02-08 01:59:17,095 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-08 01:59:17,096 DEBUG   Frame advanced, resetting header!!
2022-02-08 01:59:17,096 DEBUG   Getting transaction 1
2022-02-08 01:59:17,099 DEBUG   Request registers: [183] glen=1. Response [5220] len=1. regs={183: 5220}
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 172, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[... cut ...]
2022-02-08 01:59:37,198 DEBUG   Request registers: [183] glen=1. Response [5220] len=1. regs={183: 5220}
2022-02-08 01:59:37,199 DEBUG   Battery voltage low=5220 high=0 value=52V
2022-02-08 01:59:38,151 DEBUG   send: 0x1 0x3 0x0 0xb7 0x0 0x1 0x34 0x2c
2022-02-08 01:59:38,154 DEBUG   Adding transaction 1
2022-02-08 01:59:38,203 DEBUG   recv: 0x1 0x3 0x2 0x14 0x64 0xb6 0xaf 0xff
2022-02-08 01:59:38,204 DEBUG   Frame check failed, ignoring!!
2022-02-08 01:59:38,204 DEBUG   Resetting frame - Current Frame in buffer - 0xff 0x1 0x3 0x2 0x14 0x64 0xb6 0xaf 0xff
```

reedy, 2022-02-08:
[image]

Will see what it reports over the next few hours

kellerza, 2022-02-08:
Ok, try to update in 5 min when the CI/CD job is finished and then at least debugging should work better

reedy, 2022-02-08:
[image]

Looks like it was fairly consistenly getting values overnight.

Just updated to `2022.02.08-0.1.1`

Without debug enabled, it's still regularly timing out it seems

```
2022-02-08 12:28:46,245 INFO    MQTT: Connection successful
2022-02-08 12:29:07,255 ERROR   Read error: Timeout
2022-02-08 12:29:07,256 INFO    Retrying individual sensors: ['Battery voltage:step:mean']
2022-02-08 12:29:32,363 ERROR   Read error: Timeout
2022-02-08 12:29:53,389 ERROR   Read error: Timeout
2022-02-08 12:30:14,397 ERROR   Read error: Timeout
2022-02-08 12:30:35,422 ERROR   Read error: Timeout
2022-02-08 12:30:56,447 ERROR   Read error: Timeout
2022-02-08 12:31:17,472 ERROR   Read error: Timeout
2022-02-08 12:31:38,492 ERROR   Read error: Timeout
2022-02-08 12:31:59,517 ERROR   Read error: Timeout
2022-02-08 12:32:20,542 ERROR   Read error: Timeout
2022-02-08 12:32:43,558 ERROR   Read error: Timeout
2022-02-08 12:33:13,609 ERROR   Read error: Timeout
2022-02-08 12:33:34,621 ERROR   Read error: Timeout
```

Lets turn debug back on now...

reedy, 2022-02-08:
[image]

It looks like the various improvements you've made are definitely helping! 👍

kellerza, 2022-02-08:
I still don't like your Timeouts, it seems to occur way too often! I probably get <5 a day...

Can you try do this through the mbusd bridge addon?

Things to check on your cable:
- Do you run this next wire next to any AC or to a lesser extend DC cables? (Maybe remove this completely from other cable trunking to be sure) 
- Make sure you use a twisted pair (gr-gr-white)
- Make sure the crimp is good
- Do you use a good quality CAT5e/CAT6?
- Shielded might be an option (ground on the one side only)

Another thing might be to try another RS485 converter, like [this](https://www.robotics.org.za/index.php?route=product/product&product_id=5947&search=rs485+usb) with pin 3 to ground

reedy, 2022-02-08:
I can try to get the mbusd addon up and running.

>Do you run this next wire next to any AC or to a lesser extend DC cables? (Maybe remove this completely from other cable trunking to be sure)

Nope. Experience has told me not to bother doing stuff like that until you know it's all working. The cable currently goes out the front of the inverter (the removeable panel is off), into the USB adapter plugged into the RPI sat on top of a grounded metal cabinet, so nowhere near any DC or AC cables.

Cable is definitely twisted pair (see photos in https://github.com/kellerza/sunsynk/pull/14 same for cable out of inverter setup)

Crimp seems to be fine, has continuity.

I'm not sure if it is good quality cable. It's CAT5e, and the tail end of the box I've been using on and off for a while. I do have some (much newer) external grade CAT5e I can try making a cable up from instead. Electricians used cable from the same reel to do the inverter <-> battery communication cable.

The plugs I know are of decent quality.

Haven't got any shielded cable (currently).

Happy to try another converter if I can find somewhere to source one... That one ordered from ZA isn't that bad price wise, just shy of 245 ZAR shipped to the UK, but not sure how quick it might be. No useful google-able part number to try and find one at another supplier, unfortunately.

That or finding a similar FTDI one (ie "known good brand") somewhere....

reedy, 2022-02-08:
Looks like the mbusd version just works a lot better...

Immediately showing extra data

[image]

reedy, 2022-02-08:
Logs are generally quieter too..

```
[14:39:12] INFO: Starting mbusd -d -L - -v 2 -p /dev/ttyUSB0 -s 9600 -m 8N1 -P 502
08 Feb 2022 14:39:12 mbusd-0.5.1 started...
08 Feb 2022 14:39:12 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
08 Feb 2022 14:40:53 conn_open(): accepting connection from 172.30.33.1
```

```
2022-02-08 14:40:53,209 INFO    MQTT: Connection successful
2022-02-08 14:47:00,150 ERROR   Read Error: failed to read
2022-02-08 15:00:23,106 ERROR   Read Error: failed to read
```

reedy, 2022-02-08:
Modbus is still identical

```
2022-02-08 14:40:53,209 INFO    MQTT: Connection successful
2022-02-08 14:47:00,150 ERROR   Read Error: failed to read
2022-02-08 15:00:23,106 ERROR   Read Error: failed to read
2022-02-08 15:21:31,376 ERROR   Read Error: failed to read
2022-02-08 16:13:04,565 ERROR   Read Error: failed to read
2022-02-08 16:25:44,407 ERROR   Read Error: failed to read
2022-02-08 16:28:53,181 ERROR   Read Error: failed to read
2022-02-08 16:31:23,685 ERROR   Read Error: failed to read
2022-02-08 16:44:26,874 ERROR   Read Error: failed to read
2022-02-08 16:49:28,650 ERROR   Read Error: failed to read
2022-02-08 17:02:24,722 ERROR   Read Error: failed to read
2022-02-08 17:13:18,058 ERROR   Read Error: failed to read
2022-02-08 17:20:13,180 ERROR   Read Error: failed to read
2022-02-08 17:36:38,801 ERROR   Read Error: failed to read
2022-02-08 17:54:21,920 ERROR   Read Error: failed to read
2022-02-08 18:01:20,268 ERROR   Read Error: failed to read
2022-02-08 18:08:50,095 ERROR   Read Error: failed to read
2022-02-08 18:09:46,323 ERROR   Read Error: failed to read
2022-02-08 18:27:31,178 ERROR   Read Error: failed to read
2022-02-08 18:29:42,796 ERROR   Read Error: failed to read
2022-02-08 18:44:00,296 ERROR   Read Error: failed to read
2022-02-08 19:16:09,142 ERROR   Read Error: failed to read
2022-02-08 19:20:17,497 ERROR   Read Error: failed to read
2022-02-08 19:31:28,224 ERROR   Read Error: failed to read
```

kellerza, 2022-02-09:
Maybe this [converter](https://www.aliexpress.com/item/32913685259.html)? 

Can you try measure if there is voltage difference between the inverter ground (pin 3 on the rj-45 likely orange-white?) and the USB ground (maybe you can try measure to the metal shield)

The only other option I can think of is to increase the retry timer on mbusd (but its not configurable today)

reedy, 2022-02-09:
> Maybe this [converter](https://www.aliexpress.com/item/32913685259.html)?

Thanks for digging it up. Ordered, should be here in a month...

kellerza, 2022-02-18:
@reedy you should also try the (dev) version of the addon for improved read&reconnects