kellerza/sunsynk #13: Connect to Deye SUN-3.6/5K-SG03LP1-EU

tuanha2000vn, 2022-02-04:
Hi, it's possible to connect to Deye Inverter?

[image]
This is the inverter

[image]
Network to RS 485

[image]
[image]
[image]
Blue (line 4) to B and Blue/White (line 5) to A according to the document

[image]

USB found in home assistant

Sunsynk Inverter Add-on Config
```
PORT: /dev/ttyUSB0
PORT_ADDRESS: ''
SUNSYNK_ID: '21031292XX'
SENSORS:
  - total_pv_power
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - battery_soc
SENSOR_PREFIX: rs485
TIMEOUT: 10
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: xxxxxx
MQTT_PASSWORD: '111111'
DEBUG: 0
```

I'm not sure what wrong the log only show this
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-04 20:11:11,734 INFO    Loading HASS OS configuration
2022-02-04 20:11:11,740 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2022-02-04 20:11:11,740 INFO    Filter *step used for ['battery_soc']
2022-02-04 20:11:11,751 WARNING Failed to connect: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 185, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 144, in main
    await SUNSYNK.connect(timeout=OPT.timeout)
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 79, in connect
    raise ConnectionError
ConnectionError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

Thank you in advance

tuanha2000vn, 2022-02-04:
Update, look like the addon connected to inverter using line 1-2 to B-A and plug into BMS RS 485 instead of Meter RS 485

However there still error, please help.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-04 21:00:14,827 INFO    Loading HASS OS configuration
2022-02-04 21:00:14,830 INFO    Filter *last used for ['total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import']
2022-02-04 21:00:14,830 INFO    Filter *step used for ['battery_soc']
2022-02-04 21:00:14,862 INFO    Protocol made connection.
2022-02-04 21:00:14,863 INFO    Connected to /dev/ttyUSB0
2022-02-04 21:00:14,939 INFO    ############################################################
2022-02-04 21:00:14,939 INFO                   SMA serial number '2103129246'
2022-02-04 21:00:14,940 INFO    ############################################################
2022-02-04 21:00:14,957 INFO    MQTT: Connection successful
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 489, in wait_for
    fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 185, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 156, in main
    await SUNSYNK.read([f.sensor for f in SENSORS])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 100, in read
    r_r = await self.client.read_holding_registers(grp[0], glen, unit=self.unit)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 35, in execute
    resp = await asyncio.wait_for(req, timeout=self._timeout)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 491, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

kellerza, 2022-02-04:
Your connection now seems the same as we have documented on the main readme of this repo

can you try less sensors? (Only try 1 for now)

tuanha2000vn, 2022-02-04:
Thank you, I've managed to connect to the inverter using Port Address instead of Port

```
PORT: /dev/ttyUSB0
PORT_ADDRESS: 192.168.1.120:502
SUNSYNK_ID: '21031292XX'
SENSORS:
  - battery_voltage
  - battery_soc
SENSOR_PREFIX: Sunsynk
TIMEOUT: 10
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: xxxxxx
MQTT_PASSWORD: '111111'
DEBUG: 2
```
There a few MQTT sensor issue, need to restart to clear them all.

I'll report about the progress soon.

Thank for the amazing addon.

kellerza, 2022-02-04:
Ok, glad you got it working, I’m assuming you are also using mbusd on the same host?

tuanha2000vn, 2022-02-04:
[image]

Sometime all sensors became unvailable, maybe 1 of the sensor fail to read/convert to float/int cause the issue?

Maybe my network cable too long (15 meter) or the baurate can be set higher than 9600?

Anyway I got near install update that amazing.

kellerza, 2022-02-05:
15 meter is not long, as long as your data lines are on a twister pair (like orange/orange-white)

Lower baud rate is also better for long distances, but in this case we have to match the inverter.

See the linked issue above for available

I’ll look into the filter, that should not happen

MQTT do unpublish sensors you remove, but also hardened this and will publish an update to the addon over the weekend

kellerza, 2022-02-05:
> Anyway I got near instant sensor value update, that amazing.

glad that is working :-) 

a big part is the combination of reading often, and only publishing on changes with the filters

kellerza, 2022-02-05:
The filter issue seems to be caused by sd_status or fault, which are both test fields. Try force their filters to last for now

```yaml
- sd_status:last
- fault:last
```

kellerza, 2022-02-08:
Text will now be accepted by the filters, so you should not need the filter overrides

closed 2022-02-08 by kellerza