kellerza/sunsynk #15: Sensor Unavailable

tuanha2000vn, 2022-02-05:
Sometime, all the sensor became unvailable. What can I do to fix this issue?

This is my setup:

USB RS-845 to Inverter via RJ45 cable **15 meter long**.

[image]
Modbus TCP to Modbus RTU Gateway Add-on
```
[12:14:07] INFO: Starting mbusd -d -L - -v 2 -p /dev/ttyUSB0 -s 9600 -m 8N1 -P 502
05 Feb 2022 12:14:07 mbusd-0.5.1 started...
05 Feb 2022 12:14:07 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
05 Feb 2022 12:14:14 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:24:06 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:24:19 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:27:02 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:27:23 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:33:55 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:34:26 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:36:19 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:36:31 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:41:04 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:41:33 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 12:45:18 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 12:45:37 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 13:16:17 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 13:16:22 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 13:32:27 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 13:32:57 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 13:40:01 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 13:40:33 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 13:50:43 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 13:50:59 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 13:56:32 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 13:56:44 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:05:50 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:06:19 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:16:27 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:16:52 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:17:01 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:17:23 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:29:44 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:29:59 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:40:16 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:40:33 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 14:42:09 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 14:42:35 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:01:26 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:01:47 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:09:25 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:09:51 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:25:12 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:25:30 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:29:19 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:29:32 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:29:59 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:30:06 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:38:57 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:39:09 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:41:46 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:42:11 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:42:26 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:42:42 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:43:46 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:44:13 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:47:40 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:47:48 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:51:36 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:51:49 conn_open(): accepting connection from 192.168.1.120
05 Feb 2022 15:52:04 conn_close(): closing connection from 192.168.1.120
05 Feb 2022 15:52:20 conn_open(): accepting connection from 192.168.1.120
```
[image]
Sunsynk Inverter Add-on
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-02-05 15:52:20,417 INFO    Loading HASS OS configuration
2022-02-05 15:52:20,418 WARNING Your config includes PORT and PORT_ADDRESS. PORT_ADDRESS will be used
2022-02-05 15:52:20,418 INFO    Filter now used for ['day_battery_charge', 'day_battery_discharge', 'grid_voltage', 'inverter_voltage', 'grid_load', 'inverter_output', 'load_power', 'pv1_power', 'pv2_power', 'battery_power', 'battery_charge']
2022-02-05 15:52:20,419 INFO    Filter *step used for ['day_grid_import', 'day_grid_export', 'day_pv_energy']
2022-02-05 15:52:20,435 INFO    Protocol made connection.
2022-02-05 15:52:20,437 INFO    Connected to 192.168.1.120:502.
2022-02-05 15:52:20,581 INFO    ############################################################
2022-02-05 15:52:20,582 INFO                   SMA serial number '2103129246'
2022-02-05 15:52:20,582 INFO    ############################################################
2022-02-05 15:52:20,599 INFO    MQTT: Connection successful
```

kellerza, 2022-02-05:
As soon as the sunsynk inverter addon restarts, or looses connection to the mqtt server all sensors will become unavailable. 

Since your addon&mqtt server is on the same host (core-mosquitto) the network is stable, but I expect the addon restarted (likely die to a timeout reading).

do you have the watchdog enabled?

Does the sunsynk addon’s log timestamp indicate the addon restarted?

tuanha2000vn, 2022-02-05:
Watchdog enabled

No addon restarted because I was watching it.

Maybe it's my cable, I'll keep watching

lispwarez, 2022-02-05:
My addon restarts multiple times a day which causes my sensor states to go "unavailable" which in turn causes havoc with my automations
@kellerza you on discord?

tuanha2000vn, 2022-02-06:
I watch closely the addon log and it look like the watchdog restart cause the unvailable issue. Turning watch dog off to find out why in restarting...

tuanha2000vn, 2022-02-06:
```
2022-02-07 03:18:07,371 DEBUG   Day Grid Import low=15 high=0 value=1kWh
2022-02-07 03:18:07,372 DEBUG   Day Grid Export low=0 high=0 value=0kWh
2022-02-07 03:18:07,372 DEBUG   send: 0x5 0x84 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x54 0x0 0x1
2022-02-07 03:18:07,372 DEBUG   Adding transaction 1412
2022-02-07 03:18:08,057 DEBUG   recv: 0x5 0x84 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x10
2022-02-07 03:18:08,059 DEBUG   Processing: 0x5 0x84 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x10
2022-02-07 03:18:08,060 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2022-02-07 03:18:08,061 DEBUG   Getting transaction 1412
2022-02-07 03:18:08,064 DEBUG   Request registers: [84] glen=1. Response [16] len=1. regs={84: 16}
2022-02-07 03:18:08,065 DEBUG   Day Load Power low=16 high=0 value=1kWh
2022-02-07 03:18:08,066 DEBUG   send: 0x5 0x85 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x6c 0x0 0x1
2022-02-07 03:18:08,067 DEBUG   Adding transaction 1413
2022-02-07 03:18:10,232 DEBUG   recv: 0x5 0x85 0x0 0x0 0x0 0x3 0x1 0x83 0x4
2022-02-07 03:18:10,232 DEBUG   Processing: 0x5 0x85 0x0 0x0 0x0 0x3 0x1 0x83 0x4
2022-02-07 03:18:10,233 DEBUG   Factory Response[131]
2022-02-07 03:18:10,234 DEBUG   Getting transaction 1413
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 204, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 190, in main
    await polltask
  File "/usr/src/app/./run.py", line 182, in poll_sensors
    await SUNSYNK.read([f.sensor for f in fsensors])
  File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 104, in read
    raise Exception("failed to read")
Exception: failed to read
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```
Look like this part cause the crash issue
```
File "/usr/local/lib/python3.9/site-packages/sunsynk/sunsynk.py", line 104, in read
    raise Exception("failed to read")
```

kellerza, 2022-02-08:
The latest version catches a general exception, so it should just retry, but I'm not 100% why this failed. If it occurs often, we can look at it again

(For reference, over the last day and a half I got 7 Timeouts, but non of these exceptions)

closed 2022-02-08 by kellerza