kellerza/sunsynk #21: ModuleNotFoundError: No module named 'pymodbus'

reedy, 2022-02-21:
### Issue related to

Sunsynk Home Assistant Add-On

### Describe the issue/bug
Attempting to start the add on fails with `ModuleNotFoundError`.

### Your environment

- Home Assistant version: 2022.2.9
- Addon version: 2022.02.18-0.1.3

### Logs

```log
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 17, in <module>
    from pymodbus.exceptions import ModbusIOException  # type: ignore
ModuleNotFoundError: No module named 'pymodbus'
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

kellerza, 2022-02-22:
You can try to update to the latest version

reedy, 2022-02-22:
That looks to have fixed that error, but we get different errorrs now

```log
2022-02-22 12:48:38,696 INFO    MQTT: Connection successful
2022-02-22 12:48:45,624 INFO    Protocol lost connection.
2022-02-22 12:48:50,082 ERROR   Read Error: 'NoneType' object has no attribute 'write'
2022-02-22 12:49:00,101 ERROR   Read Error: 'NoneType' object has no attribute 'write'
2022-02-22 12:49:10,120 ERROR   Read Error: 'NoneType' object has no attribute 'write'
2022-02-22 12:49:20,138 ERROR   Read Error: 'NoneType' object has no attribute 'write'
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 158, in read
    await SUNSYNK.read(sensors)
  File "/usr/local/lib/python3.9/site-packages/sunsynk/pysunsynk.py", line 101, in read
    r_r = await self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 34, in execute
    req = self._execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 141, in _execute
    self.write_transport(packet)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py", line 131, in write_transport
    return self.transport.write(packet)
AttributeError: 'NoneType' object has no attribute 'write'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 255, in <module>
    LOOP.run_until_complete(main(LOOP))
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/usr/src/app/./run.py", line 241, in main
    await polltask
  File "/usr/src/app/./run.py", line 233, in poll_sensors
    if await read([f.sensor for f in fsensors]):
  File "/usr/src/app/./run.py", line 171, in read
    raise Exception(f"Multiple Modbus read errors: {err}") from err
Exception: Multiple Modbus read errors: 'NoneType' object has no attribute 'write'
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

kellerza, 2022-02-22:
Sorry @reedy I made quite significant changes and obviously not a lot of testing here, can you try the “dev” version. I’ll try to combine these two later

reedy, 2022-02-22:
Yeah, switched over now :)

Minor config changes needed, but looks to be up and running now again.

I should work out how to do the fancy dashboard things rather than just having the list of sensors

kellerza, 2022-02-22:
Energy management is the first step - [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-dev/DOCS.md#energy-management) (kWh)

The other one is the `power-distribution-card` that you can install via HACS. You can use it for displaying all the instantaneous power measurements (Watt)

closed 2022-02-22 by kellerza