kellerza/sunsynk #24: Sensor Listing for Dummy - Not an issue

stoffies00711, 2022-03-11:
Hi @kellerza 

First of all...  Thanks for the integration.  

Been using an outdated version for a while now.  I finally updated it and got it working perfectly again.  Just one question...
The old version's documentation had a link to a list of sensors to enter in your config file such as - sensor.pv1_power or sensor.battery_soc and so on.  I just took what I needed from there and included it in my config page.  
The new link shows sensor definitions but I'm not yet clever enough to make sense of it.  How can I get a simple list of sensors or a line to include all sensors?  

Thanks again for the help.

[image]

stoffies00711, 2022-03-11:
I copied some sensors from the definition page and added some "undescores" between the spaces and they work so far.  Thanks!

kellerza, 2022-03-11:
Great👍🏻 

I’ll add the option that you can use the sensor name without the underscores in the config as well…

stoffies00711, 2022-03-11:
Don't worry. I think the underscores makes it easier to work with.  Maybe just add a simple list of the sensors with the documentation.  Either way...  The integration works great.  Thanks!!