kellerza/sunsynk #26: Document working adapters

reedy, 2022-03-20:
Following up on https://github.com/kellerza/sunsynk/issues/10#issuecomment-1033486797

https://www.aliexpress.com/item/32913685259.html works just fine on the SunSynk 3.8.

TODO for myself to document this in the README, and maybe look at a better way of documenting this...

A table and some checks for "tested working"?

kellerza, 2022-07-12:
Hi @reedy, not sure if you're still considering working on this, else can we close this issue?

N1c084, 2022-07-16:
Hi 
Maybe this need to be added to the compatible hardware in readme also?

For information, my system working fine with a RTU to TCP IP converter: [HF5142](https://fr.aliexpress.com/item/32945650524.html?gatewayAdapt=glo2fra)

PORT: 102.xxx.xxx.xxx: port
DRIVER: pymobus
SUNSYNK_ID: 'xxxxxx'
SENSOR_PREFIX: deye2
SENSORS:
  - total_pv_power
  - total_battery_charge

Is it possible to change the name of inverter in HA from sunsynk to other brand?

Thanks

kellerza, 2022-07-16:
I’ve updated the readme recently. You are welcome to create a PR to add additional adaptors

Not sure where you want the name to be changed. Toaday you can only changethe prefix (needs to be done before you enable any sensors)

closed 2022-08-17 by kellerza