kellerza/sunsynk #27: FR: Expand scope to include sunsunk OEM'd inverters like Sol-Ark

poldim, 2022-03-21:
I've got a Sol Ark 12k which is a Sunsynk in wool's clothing.  I did read that some of the registers are different.  [Here is the list I have](https://docs.google.com/spreadsheets/d/1JKPSJ3RWAkGTsEz0LB08PZZ-XCBtGfTnJy5yqc7GmP4/edit?usp=sharing) what works on my config and matches that from [the other HA addon](https://github.com/pbix/HA-solark-PV/blob/main/custom_components/solark/hub.py).

kellerza, 2022-03-22:
How many of these are actually different? 

The list you shared includes all registers, so not really sure what the expectation/delta is here

poldim, 2022-04-13:
I did a quick check and surprisingly most of them are a match.  It looks like you have 16 points that are called power but the units are energy so those should be updated
[image]

kellerza, 2022-04-14:
Thanks, yes, the addon started life from a translated Chinese modbus spec, so they all started their life as mostly `time x power`, so day/month/year/total and power/import/export/charge/discharge. Although it might not be 100% correct, I also did not want to change all the energy sensors (it effects everyone's HA energy stats). I did fix/deprecate some confusing power ones a while back.

166 is the aux port, where you would connect a generator - maybe they are the same?

I've never had the need for 200-202, and they seem duplicates of 84-86. Since you only need to feed totals into HA energy management, not sure how useful they are

If we prepend these 4 with "Solark " we could probably add definitions? Or any other suggestions?

poldim, 2022-04-14:
I know breaking changes suck but the option to have incorrect names seems like a worse one.  Obviously it's your call.

The Chinese translation/label is "Gen makes the power of micro-reverse input /  Gen Do Micro Inverse Power Input".  I just checked mine and it's still at zero but I'm currently not connected to my gen/smartload breaker.

200+ appear to be configuration of the battery system charging which are different from energy used in 84-86
[image]

A prefix on the entity name is fine, you can also disable them by default if you think they will be rarely used.

kellerza, 2022-07-11:
Implemented your suggestion in version `0.1.6` of the library. Should makes its way into the dev version of the addon first

closed 2022-07-20 by kellerza