kellerza/sunsynk #28: ValueError: unsupported scheme '' for /dev/ttyUSB0

McAllstar, 2022-03-25:
### Issue related to the Python sunsynk library
ValueError: unsupported scheme '' for /dev/ttyUSB0
### Describe the issue/bug
when i start the plugin I get this error listed below.

File "/usr/local/lib/python3.9/site-packages/connio/__init__.py", line 21, in connection_for_url
    raise ValueError("unsupported scheme {!r} for {}".format(scheme, url))
ValueError: unsupported scheme '' for /dev/ttyUSB0
A clear and concise description of what the bug is.

### Expected behavior
expect the plugin to start. 
A clear and concise description of what you expected to happen.
expect the plugin to start and run 
### Logs (if applicable)
[image]

```log
paste logs in 3x backticks...
```

McAllstar, 2022-03-25:
2022-03-25 15:28:43,615 INFO    Connected to /dev/ttyUSB0
2022-03-25 15:28:43,619 DEBUG   send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
2022-03-25 15:28:43,621 DEBUG   Adding transaction 1
2022-03-25 15:28:53,634 ERROR   Timeout reading: 
2022-03-25 15:28:53,635 INFO    ############################################################
2022-03-25 15:28:53,636 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2022-03-25 15:28:53,637 INFO    ############################################################
2022-03-25 15:28:53,638 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.

McAllstar, 2022-03-25:
thats if i change it to pymodbus

closed 2022-03-25 by McAllstar

kellerza, 2022-03-25:
Umodbus requires the port to start with `serial://` so you end up with `serial:///dev/ttyUSB0`

McAllstar, 2022-03-28:
> CAUTION - EXTERNAL SENDER - Please be careful when opening links and attachments. Nedbank - IT Information Security Department (ISD)

Hi

Hope you are well. Any idea why I could be getting this? I did a complete re-install of HA and only installed mosquito etc and then the Sunsynk addon and I get this below.

Thanks Brett

***@***.***

It seems as if it makes the connection but it’s unable to read. My cabling is correct, and I tried using umodbus with serial: ///dev/ttyUSB0 and also pymodbus /dev/ttyUSB0sr

Nedbank disclaimer and confidentiality notice:

This email may contain information that is confidential, privileged or otherwise protected from disclosure. If you are not an intended recipient of this email or all or some of the information contained therein, do not duplicate or redistribute it by any means. Please delete it and any attachments and notify the sender that you have received it in error. Unless specifically indicated, this email is neither an offer or a solicitation to buy or sell any securities, investment products or other financial product or service, nor is it an official confirmation of any transaction or an official statement of Nedbank. Any views or opinions presented are solely those of the author and do not necessarily represent those of Nedbank. Nedbank Ltd Reg No 1951/000009/06.

The following link displays the names of the Nedbank Board of Directors and Company Secretary. [http://www.nedbank.co.za/terms/DirectorsNedbank.htm]

If you do not want to click on a link, please type the relevant address in your browser
********************

kellerza, 2022-03-28:
The only other way to do this is to use mbusd as a tcp to rtu gateway then connect to tcp://ip:port

McAllstar, 2022-03-28:
Ok thanks a mil. I’ll try figure it out.

I enabled debug logging and go the below error again.

***@***.***