kellerza/sunsynk #29: HA add-on do not install

nardusleroux, 2022-04-02:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Install of add-on does not complete successfully.

**Expected behavior**
Add-on expected to install via GUI, configure add-on and start add-on.

### Your environment

- Home Assistant version: core-2022.3.8
- Supervisor version: supervisor-2022.03.5
- Addon version: armv7-addon-hass-addon-sunsynk:2022.03.04b-0.1.4
- Host OS: Raspbian GNU/Linux 10 (buster)
- Docker version: 20.10.8

### Logs

Supervisor log:
```
22-04-02 12:20:35 INFO (SyncWorker_7) [supervisor.docker.addon] Starting build for 4bb530c7/armv7-addon-hass-addon-sunsynk:2022.03.04b-0.1.4
22-04-02 12:20:40 ERROR (SyncWorker_7) [supervisor.docker.addon] Can't build 4bb530c7/armv7-addon-hass-addon-sunsynk:2022.03.04b-0.1.4: The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1 sunsynk[pymodbus,umodbus]==0.1.4' returned a non-zero code: 1
22-04-02 12:20:40 ERROR (SyncWorker_7) [supervisor.docker.addon] Build log: 
Step 1/13 : ARG BUILD_FROM
Step 2/13 : FROM ${BUILD_FROM}
 ---> 1c9cf6b54197
Step 3/13 : RUN pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1 sunsynk[pymodbus,umodbus]==0.1.4
 ---> Running in 6e61df029649
Fatal Python error: init_interp_main: can't initialize time
Python runtime state: core initialized
PermissionError: [Errno 1] Operation not permitted
Current thread 0xb6f82390 (most recent call first):
<no Python frame>

Removing intermediate container 6e61df029649

```

Core log:
```
2022-04-02 12:08:37 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/4bb530c7_hass-addon-sunsynk/install - The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1 sunsynk[pymodbus,umodbus]==0.1.4' returned a non-zero code: 1
```

kellerza, 2022-04-02:
It might be libseccomp2 related on your OS?

https://community.home-assistant.io/t/migration-to-2021-7-fails-fatal-python-error-init-interp-main-cant-initialize-time/320648

nardusleroux, 2022-04-03:
Indeed upgrading libseccomp solves this. Thank you.

closed 2022-04-03 by nardusleroux