kellerza/sunsynk #31: Sunsynk Docker

tuanha2000vn, 2022-04-08:
I LOVE this addon. The only downside is it require Home Assistant Supervised installation. On my Synology it mean you have install a virtual machine which is resource consuming.

It's possible to run this via Docker Container?

Thank you.

kellerza, 2022-04-08:
It should be possible to run this via docker, you just have to creeate your own config test file

If you map a manually crafted "config.json" to the root of the docker container docker container it should work...

```
git clone https://github.com/kellerza/sunsynk.git
cd sunsynk/hass-addon-sunsynk
docker build -t sunsynk
cd ..   # change to the path containing config.json
docker run -v $(pwd)/config.json:/config.json --name sunsynk sunsynk
```

tuanha2000vn, 2022-04-08:
Thank, wish  you publish it to docker hub someday

kellerza, 2022-04-08:
It uses the exact same config mechanism!

you are welcome to do some more research on the topic and come with a proposal 😉

Ivan-L, 2022-05-29:
@tuanha2000vn I also struggled with this as I ran Home Assistant Core on my Synology NAS in a Docker container. I then tried running HASSOS in a virtual machine but as you mentioned, that was a lot slower and more resource intensive. Finally I found this, which has been working very well for me:
https://gist.github.com/maeneak/851e883eca7cddd7114f7eaed201ca9d
I now run a Home Assistant Supervised installation on my NAS using Docker without the overhead of a VM.

[image]

closed 2022-07-12 by kellerza

jpmeijers, 2023-06-06:
> ```
> git clone https://github.com/kellerza/sunsynk.git
> cd sunsynk/hass-addon-sunsynk
> docker build -t sunsynk
> cd ..   # change to the path containing config.json
> docker run -v $(pwd)/config.json:/config.json --name sunsynk sunsynk
> ```

These steps do not seem to work anymore. @kellerza what is your current preferred workflow for testing locally?

PR #150 (2023-06-06): Add docker-compose to run as standalone container
Fixes #31 

* Added docker-compose to run both sunsynk-multi as well as mbusd
* Cleaned up sunsynk-multi Dockerfile, as list of Python dependencies should be in a requirements.txt file