kellerza/sunsynk #33: Sunsynk data not reading

marvil1961, 2022-04-11:
I have HA installed on a Ubuntu server inside virtualbox as per the installation instructions ha installation on Linux (https://www.home-assistant.io/installation/linux/#install-home-assistant-operating-system)

Have battled to get the usb to RS485 converter to be seen by HA but found that the usb device must first be enabled in Virtualbox and were then quite exited that HA will work with my 8kW Sunsynk Inverter but it was not meant to be - communication is still not happening. Cable connection I think is correct Oninverter Pin 1&2 to B&A on Usb converter

on the Modbus TCP to Modbus RTU Gateway Add-on my Log info is as follows:
DEVICE: /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0

[19:21:07] INFO: Starting mbusd -d -L - -v 3 -p /dev/ttyUSB0 -s 9600 -m 8N1 -P 502
11 Apr 2022 19:21:07 mbusd-0.5.1 started...
11 Apr 2022 19:21:07 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
11 Apr 2022 19:21:59 tty: error in read() (port closed)
11 Apr 2022 19:21:59 trying to re-open tty...
11 Apr 2022 19:21:59 tty re-opening...
11 Apr 2022 19:21:59 tty: trying to open /dev/ttyUSB0 (speed 9600 mode 8N1)
11 Apr 2022 19:21:59 tty re-opened.

The above looks like this part is working

on the Sunsynk Inverter Add-on the log file looks like this and I get Timeout Readings 

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-04-11 19:26:04,237 INFO    Loading HASS OS configuration
2022-04-11 19:26:04,294 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_power
2022-04-11 19:26:04,305 INFO    Protocol made connection.
2022-04-11 19:26:04,306 INFO    Connected to /dev/ttyUSB0
2022-04-11 19:26:04,364 INFO    ############################################################
2022-04-11 19:26:04,365 INFO                Inverter serial number '2111255555'
2022-04-11 19:26:04,365 INFO    ############################################################
2022-04-11 19:26:04,366 INFO    MQTT: Connecting to mqtt_user@core-mosquitto:1883
2022-04-11 19:26:04,383 INFO    MQTT: Connection successful
2022-04-11 19:26:15,412 ERROR   Timeout reading: 
2022-04-11 19:26:15,413 INFO    Retrying individual sensors: ['Total PV Power:last', 'Total Battery Charge:last', 'Total Battery Discharge:last', 'Total Grid Export:last', 'Total Grid Import:last', 'Battery SOC:last']
2022-04-11 19:26:25,450 ERROR   Timeout reading: Total PV Power
2022-04-11 19:26:35,489 ERROR   Timeout reading: Total Battery Charge
2022-04-11 19:26:45,520 ERROR   Timeout reading: Total Battery Discharge
2022-04-11 19:26:55,558 ERROR   Timeout reading: Total Grid Export
2022-04-11 19:27:05,598 ERROR   Timeout reading: Total Grid Import
2022-04-11 19:27:15,700 ERROR   Timeout reading:

marvil1961, 2022-04-13:
Thank you I did remove mbusb but still no connection, on Sunsynk Addon I get the following logging details

2022-04-13 06:36:32,724 INFO    Loading HASS OS configuration
2022-04-13 06:36:32,752 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_power
2022-04-13 06:36:32,801 INFO    Protocol made connection.
2022-04-13 06:36:32,802 INFO    Connected to /dev/ttyUSB0
2022-04-13 06:36:42,806 ERROR   Timeout reading: 
2022-04-13 06:36:42,809 INFO    ############################################################
2022-04-13 06:36:42,809 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2022-04-13 06:36:42,809 INFO    ############################################################
2022-04-13 06:36:42,809 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.

I think my problem is that I can not read any thing from the serial port
line 5 & 6 above

2022-04-13 06:36:32,802 INFO    Connected to /dev/ttyUSB0
2022-04-13 06:36:42,806 ERROR   Timeout reading:

kellerza, 2022-04-13:
In your initial logs you did have a connection

```
2022-04-11 19:26:04,305 INFO Protocol made connection.
2022-04-11 19:26:04,306 INFO Connected to /dev/ttyUSB0
2022-04-11 19:26:04,364 INFO ############################################################
2022-04-11 19:26:04,365 INFO Inverter serial number '2111255555'
2022-04-11 19:26:04,365 INFO ############################################################
```

Are you sure you don't have anything else trying to use the serial port?
Also worth double checking your wiring. Read the README in the repository. What is your cable length?

kellerza, 2022-07-12:
Hi @mikee47, it would be great if you can do a small writeup of this idea and add it to the readme - [here](https://github.com/kellerza/sunsynk#fault-finding)

closed 2022-07-16 by kellerza