kellerza/sunsynk #37: Enhancement: Writable sensors

Ivan-L, 2022-05-29:
### Enhancement related to

Sunsynk Home Assistant Add-On

### Description

Hi @kellerza, I see you recently introduced the concept of a RWSensor so I can see that you are planning to support this in the future. Have you got a branch or any in-progress work for this? If not, I'd like to take a stab at it to introduce a missing sensor and make it writable based on a [Select Entity](https://developers.home-assistant.io/docs/core/entity/select/). The sensor is number 243 which allows you to toggle between Battery or Load priority for PV production.

Happy to hear your thoughts!

kellerza, 2022-05-30:
Writing registers are supported by profiles today. So if you create a new profile and add your sensor to the array it should work.

Actual values being written is stored in yaml files under /share/addon_name/. Select UPDATE in the frontend and the current value (or multiple values in case of many sensors) will be written

See [here](https://github.com/kellerza/sunsynk/blob/v0.1.5/hass-addon-sunsynk-dev/profiles.py#L224)

Ivan-L, 2022-05-30:
Yeah I have been using profiles for a while, they are extremely useful!
[image]

What I was thinking was that having certain entities editable can be complementary to profiles and actually make creating them easier from Home Assistant. For example, whilst on vacation (therefore no access to the inverter), any time I want to modify the battery percentage of one of the programs (Prog1 Capacity-Prog6 Capacity) I need to access the /share/addon_name, copy and paste one of the profile files, edit the values in it, rename it and then reload. If however I could edit them via the Home Assistant UI, that would be so much easier. Same for the above sensor 243 - currently a profile would need to exist already for the register to be writable.

Here is where the inspiration came from, having device config entities for some/all writable inverter registers:
https://developers.home-assistant.io/blog/2021/10/26/config-entity/#entity-categories

kellerza, 2022-05-31:
Ok, that is an interesting idea. `entity_category` seems to be part of the MQTT autodiscovery options, so it should be possible.

If you decide to add all RWSesnors / descendants they will also have to be read as well, and for that the round robin filter should be ok

Ivan-L, 2022-05-31:
I'll do a proof of concept first to see if the idea is viable and report here.

Ivan-L, 2022-06-04:
Ok it is starting to take shape. As an experiment I converted the Prog 1 to 6 Battery Capacity sensors to `NumberEntity` instead of `SensorEntity`. The default min, max and step values (0, 100, 1) for this entity type map perfectly to the battery SOC settings. I also set the `entity_category` for these sensors to `config`.

Here is the result on the Device that this add-on creates:
[image]
Here is a [link](https://www.home-assistant.io/blog/2021/11/03/release-202111/#entity-categorization) explaining the extra "Configuration" card and the implications.
The profile picker is still a primary entity and is therefore in the topmost "Controls" card:
[image]

Now going into the `NumberSelect` entity in detail, it worked out of the box for loading up the existing register values.
I had to create it in mqtt.py:
```python
@attr.define
class NumberEntity(Entity):
    """A HomeAssistant Number entity."""

    command_topic: str = attr.field(default=None, validator=required)

    on_change: Callable = attr.field(default=None)

    _path = "number"
```

For now the logic of when to create the `NumberEntity` in run.py is very rudimentary and needs to change, but here it is for discussion:
```python
if isinstance(sensor, RWSensor) and not isinstance(sensor, TimeRWSensor):
    ents.append(
        NumberEntity(
            name=f"{OPT.sensor_prefix} {sensor.name}".strip(),
            entity_category="config",
            state_topic=f"{SS_TOPIC}/{OPT.sunsynk_id}/{sensor.id}",
            command_topic=f"{SS_TOPIC}/{OPT.sunsynk_id}/{sensor.id}_set",
            unit_of_measurement=sensor.unit,
            unique_id=f"{OPT.sunsynk_id}_{sensor.id}",
            device=dev,
            on_change=create_on_change_handler(filt, lambda val: int(val))
        )
    )
else:
    ents.append(
        SensorEntity(
            name=f"{OPT.sensor_prefix} {sensor.name}".strip(),
            state_topic=f"{SS_TOPIC}/{OPT.sunsynk_id}/{sensor.id}",
            unit_of_measurement=sensor.unit,
            unique_id=f"{OPT.sunsynk_id}_{sensor.id}",
            device=dev,
        )
    )
```
The interesting parts of the `NumberEntity` are that the `entity_category` is set, there is a `command_topic` and there is an `on_change` handler. There are other attributes that I'd like to set based for `NumberEntity`, namely `min_value`, `max_value`, `step` and `mode`, depending on the sensor. These parameters could be defined per sensor in definitions.py of the sunsynk module.

When the entity value changes, the register is written to the inverter, read back again and then the result is published. Error handling could be improved.
```python
async def write_sensor(
    filter: Filter, value
) -> bool:
    """Write sensor with the Modbus interface."""

    sensor = filter.sensor
    newv = ensure_tuple(value)
    if newv == sensor.reg_value:
        return False
    
    _LOGGER.info(
        "Writing sensor %s: %s=%s  [old %s]", sensor.name, sensor.id, newv, sensor.reg_value
    )
    sensor.reg_value = newv
    await SUNSYNK.write_sensor(sensor)
    await read_sensors([sensor])
    await publish_sensors([filter], force=True)

    return True
```

Some quirks with the `NumberEntity` include:
- The icon changed to a slider, whereas it was previously a battery for the Prog sensors - see my screenshot in my previous message under "Time of use"
- When you click on the entity name and the modal pops up, _both_ the slider and a text box which is squashed are displayed:<br />[image]<br />We'd probably want to pick either `slider` or `box` and not have the `mode` set to `auto`

Now because I had some time I took things a bit further.

I had previously created a card using the [multiple-entity-row](https://github.com/benct/lovelace-multiple-entity-row) HACS add-on. I simply adjusted it to display the new `number.*` entities for the Prog registers (which were previously `sensor.*` entities. It looks the same, however tapping on the battery percentages allows you to edit them in a dialog like in the screenshot above - cool!
[image]

Lastly - and this is probably going to be controversial - now that some entities are writable in Home Assistant, one can potentially use Scenes to achieve the same functionality that Profiles do. For example, you can create a Scene which sets certain entities to certain values, just like Profiles do, the difference being that you would configure them via the Home Assistant UI. This does need some thinking / discussion.

A further thought about other registers - there are a few other entity types that could be - for example, a `SwitchEntity` for on/off-type registers and a `SelectEntity` for registers which allow for selecting one of several fixed values. For the Prog1-6 Time registers I found not find a suitable "time input" entity, so perhaps a `SelectEntity` with hours or 15 minute steps in a 24 hour day might be appropriate.

Thoughts?

kellerza, 2022-06-09:
It looks very promising @Ivan-L !

Like all changes, lets do it in incremental steps. Start with the NumberEntity, but without customizing min/max/changing definitions.
The only change to definitions might be to introduce a a new class NumberRWSensor (from RWSensor) to make your test more natural (and ensure it doesnt break when a new class is introduced

In future PR we could even include a text based input for RWSensor

If you can change the settings with scenes, that is simply an additional way to do this. The main reason I made the profiles was that settings like the bit values for programming becomes complex (multiple switch sensors per register) and the current profiles is still the most flexible&safest way to change this.

kellerza, 2022-07-11:
Hi @Ivan-L you have been very quiet,- are you still working on this?

Ivan-L, 2022-07-11:
Hey @kellerza apologies, a few things happened and I have been spending a lot of time sorting out my inverter + battery, so unfortunately I have not had a chance to work on this. I will pick it up this week.

Last I worked on it there was some weirdness where the `mode` of the number entities kept on resetting back to `auto` despite me specifying otherwise in the python code. I did the inheritance thing where I selectively opt-in certain registers to use `NumberRWSensor`. The `Prog1..n_Capacity` ones were easy since their min/max values are in line with the Home Assistant defaults and I can PR that, however another issue I ran into, which could be tackled later, was with the maximum values for the `Prog1..n_power` entities - there I cannot hardcode a value of 5000 watts (for 5kW inverters) because that would limit inverters with higher capabilities. There is a specific register that I think gives us the max power to use for the program power registers, but then that register would likely need to be treated the same as the serial number sensor - read on startup and use for the maximum values of other sensors. Furthermore, the UI on the inverter does not allow you to set a `Prog1..n_Capacity` lower than the low battery capacity register, so that is also something to maybe consider, i.e the minimum is not actually a hard-coded 1 or 0.

This is all in the name of not allowing the user to set values outside of the allowable ranges that the inverter would allow through Home Assistant.

A final spanner is that the `Prog1..n_Capacity` registers should not be editable if the battery is configured to use voltages rather than % or lithium mode, though that is probably not the end of the world if they are, the user would just not be performing a useful action if editing these entities.

I'd love to hear your thoughts on the above.

I can also push what I have on my fork for you to look at / use.

kellerza, 2022-07-11:
Hi, register 16&17 seems to be rated power (0.1W) so this might be an option for capacity. Else you can add the min/max to the config of the dev addon while we're testing it

Ivan-L, 2022-07-19:
Hi @kellerza, I put up a PR for the above. It ended up being bigger than I thought - let me know if you'd like me to split it up into smaller PRs, I'd be happy to do so.

Ivan-L, 2022-07-28:
Hi @kellerza I have implemented some more writable sensors. No PR yet as I just want to make sure you're happy with #48 first before proceeding.

Here is a screen grab:
[image]

kellerza, 2022-07-29:
@Ivan-L so far its running smoothly. A couple of entities changed from `sensor.x` to `number.x` and had to be renamed in the frontend

Updating System_mode seems a bit less responsive, but might have to rethink the concept (especially creating new profiles) now that we can directly change the values.

Ivan-L, 2022-07-29:
Hi @kellerza, great to hear it's running smoothly.

Yeah the sensors that become writable do migrate from `sensor.x` to `number.x` or (in the next PR) `select.x` and `switch.x` entities. What we can do is introduce an OPT flag which when set to `true` makes all sensors writable but if not then the sensors end up as the regular `sensor.x` entities - I made that suggestion in the initial PR, I guess it just needs to be decided whether to push a breaking change like this out or introduce the option flag so that users can elect to opt into writable sensors and when doing so they are warned that the entity IDs will change.

Updating system mode is slow yeah, I believe this is due to the round robin filter for all of the prog sensors. This can be changed to use a similar approach as the `write_sensors` function that I implemented where you read the written sensors and publish them again, and you'd get near instant feedback when switching system modes. The previous PR did not affect switching profiles. And yes, HA scenes can be used as an alternative to profiles, but I found profiles extremely useful whilst developing writable entities because I could restore my sensor values easily. With regards to creating new profiles - one enhancement I'd suggest with the current implementation is to get rid of the UPDATE option in the system mode select and replace it either with a [MQTT button](https://www.home-assistant.io/integrations/button.mqtt/) which triggers a profile update - perhaps that is the only way to create new profiles as currently I have many profiles automatically being created if I tweak some writable entities.

I will put up some PRs over the weekend for `SelectRWSensor`, `TimeRWSensor` (updated) and `SwitchRWSensor`.

Ivan-L, 2022-08-04:
Hi @kellerza just checking if you saw #52 is up? I have two more PRs after that one lined up.

Ivan-L, 2022-08-05:
Hi @kellerza, PR #53 is up. Please note the following important comment:

When the value of a TimeRWSensor changes, the other TimeRWSensors which depend on it through their min or max fields will not have their options updated. This is a larger issue that we started discussing in a previous PR, where it was debated whether we should subscribe to sensor updates so that we can update HASS discovery info when that happens. In my view this would be desirable not only for solving this issue but also reacting to sensor changes if they are done on the inverter itself. I can put up a separate PR with a proposed solution if you'd like, or I can update this PR with such functionality.

Ivan-L, 2022-08-07:
PR #54 was added to elicit some feedback on updating HASS auto discovery info when sensor values change.

Ivan-L, 2022-08-15:
Hi @kellerza, just a gentle reminder about #53 and #54 if you can have a look when you get a gap.

kellerza, 2022-08-16:
How dynamic are these updates in the HASS UI?

Ivan-L, 2022-08-16:
With #54 the updates to the affected entities are instant.

For example, assume the following `Prog1 Time` and `Prog 2 Time` configuration (ignore the icons, I was playing around with them):
[image]

Because `Prog 1 Time` is `1:00`, the `min` value of `Prog 2 Time` is `1:00`.
This is reflected here:
[image]

If I open `Prog 1 Time` and change it from `1:00` to `2:00`, then the `min` value of `Prog 2 Time` is updated without the screen even refreshing.
[image]
It works really well.

This is true if I walk to the inverter and change the values there as well, although it takes a few seconds to take effect due to the round robin filter.

kellerza, 2022-08-28:
Hi @Ivan-L I guess we can close this?

Ivan-L, 2022-08-28:
Hi @kellerza , I have one last PR that is almost ready. Will put it up tonight.

kellerza, 2022-08-30:
🎉 thanks @Ivan-L !!

PR #57 (2022-08-30): More writable sensors
Resolves #37 

This is the final PR dealing with writable sensors in #37.

The salient changes here are:

  - Introduced new `Battery Shutdown voltage`, `Battery Restart voltage` and `Battery Low Voltage` sensors
  - Introduced new `Float voltage` sensor upon which the `Prog1-6 Voltage` sensors depend for their `max` value
  - Made `Prog1-6 Charge` and `Prog1-6 Voltage` sensors writable
    [image]
  - Made `Equalization voltage` and `Absorption voltage` writable
  - Made `Battery Shutdown Capacity`, `Battery Restart Capacity` and `Battery Low Capacity` writable
  - Corre [... cut]
closed 2022-08-30 by kellerza