kellerza/sunsynk #38: Sunsynk library docs/example code

gfwilliams, 2022-06-07:
### Issue related to the Python sunsynk library

I've just found this, and it looks perfect for my Sunsynk 8kw inverter.

However I'm really struggling to find any documentation on the library other than `pip install sunsynk`. Is there anything available?

Ideally, I'd really like just a simple 'hello world' app - a command-line tool that displayed one or two of the inverter's registers. 

Once I had that I could dig into the code enough to figure out whatever else I needed, but it would also be a hugely helpful debugging aid. I've just made up the cable and I'd like to see if it was working before I have to worry about whether Home Assistant is set up correctly.

Thanks - and sorry if this isn't the best way to get in touch.

kellerza, 2022-06-07:
Hi, the sunsynk library is used extensively by the homeassistant addon, in the same repo:

[Init the driver](/kellerza/sunsynk/blob/main/hass-addon-sunsynk-dev/run.py#L82) and then simply read values with `await read_sensors(list_of_sensors)`

You can see this library as a wrapper (with sunsynk register info) over either one of the two modbus drivers: umodbus or pymodbus

gfwilliams, 2022-06-07:
Thanks for the speedy response! I just tried this, but sadly no luck:

```
import asyncio
import pprint
from sunsynk.definitions import ALL_SENSORS, DEPRECATED
from sunsynk.sunsynk import Sensor, Sunsynk
from sunsynk.pysunsynk import pySunsynk
from sunsynk.usunsynk import uSunsynk

#pprint.pp(ALL_SENSORS)
SERIAL = ALL_SENSORS["serial"]
sensors = [ALL_SENSORS["pv1_power"], ALL_SENSORS["pv2_power"]]

SUNSYNK = uSunsynk() 
#SUNSYNK = pySunsynk()
SUNSYNK.port = "serial:///dev/ttyUSB0"
#SUNSYNK.server_id = 1
#SUNSYNK.timeout = 10
#SUNSYNK.read_sensors_batch_size = 60

async def main():
  print("Connecting...");
  await SUNSYNK.connect()
  print("Connected. Reading...")
  await asyncio.wait_for(SUNSYNK.read_sensors([SERIAL]), 10)
  print("Serial "+SERIAL.value)
  #print("Reading 2...")
  #await asyncio.wait_for(SUNSYNK.read_sensors(sensors), 10)
  #print(sensors[0].value)
  #print(sensors[1].value)
  print("Finished");  
  
  
LOOP = asyncio.get_event_loop()
LOOP.set_debug(True)
LOOP.run_until_complete(main())
LOOP.close()
```

It gives me a timeout:

```
Connecting...
Connected. Reading...
Traceback (most recent call last):
  File "test.py", line 34, in <module>
    LOOP.run_until_complete(main())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "test.py", line 23, in main
    await asyncio.wait_for(SUNSYNK.read_sensors([SERIAL]), 10)
  File "/usr/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
```

It's a SunSynk 8,8kW, wired to the RS485 connector, with the wiring and USB adaptor as shown (using pins 1+2) in the pics on https://github.com/kellerza/sunsynk#hardware (there's no GND since the adaptor doesn't have it).

Does it look like the code I've got there is ok? Is there anything else I need to set up in the inverter to enable RS485?

If/when I finally get this going, it is worth me doing a quick PR for it?

kellerza, 2022-06-23:
Hi, I would simpy change the read to ` await SUNSYNK.read_sensors(sensors)`

So your main method would be:

```python
async def main():
  print("Connecting...");
  await SUNSYNK.connect()
  print("Connected. Reading...")
  await SUNSYNK.read_sensors(sensors)
  print(", ".join(f"{s.name}={s.value} {s.unit}" for s in sensors))
  print("Finished");  
```

gfwilliams, 2022-06-27:
Thanks! And thank you for the documentation update on the Server ID. Mine was set to `00` by default. I set it to `01` and uncommented `SUNSYNK.server_id = 1` and that fixed it for me!

Out of interest, does it matter if the inverter is set as a master or slave?

I'll close this issue, but do you want me to add a PR for the script with the changes you suggested (or just feel free to copy it in yourself if it's easier)? I guess `sunsynk/scripts/comms-test.py` or similar, and a mention in the README?

closed 2022-06-27 by gfwilliams

kellerza, 2022-06-27:
PRs are always welcome, you can add comms-example.py

The master/slave setting is only used for a multi-inverter setup. From my understanding is is not related to or affects the modbus master/slave client/server in any way