kellerza/sunsynk #39: SunSynk Double Inverters

Maniac0Maniac, 2022-06-12:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue

I have 2 x SunSynk 8 KW inverters running in parallel, using the Modbus gateway to pass the data to Home Assistant - Works a treat!

The question is how do we handle paralleled inverters? The addon only accepts a single serial number

kellerza, 2022-06-12:
The only way for now would be to run 2x instances of the addon. So you can run the normal and dev versions

The downside is you will probably need 2x usb to rs485 converters (you could try connecting both through mbusd and only mbusd connecting to the serial port)

kellerza, 2022-07-11:
Did you try running 2x addons @Maniac0Maniac ?

If you have 2x USB to RS485 converters this should work well, but if you try to read info from multiple inverters on a single RS485 bus/converter they might compete for transmission time

N1c084, 2022-07-11:
Hi
Is it possible to send « off / on grid » command to inverter ?
It is for specific scheduler

kellerza, 2022-07-11:
Not sure I follow @N1c084 - why would you want to take the inverter off-grid?

This sensor is for monitoring the municipal supply.

The only way I can think of to stop using power from the grid, is to adjust the System mode to prefer your backup battery/PV during the time. It will stop using form the grid if you have alternatives available.

N1c084, 2022-07-11:
It is just to avoid totally feed back to grid , even with 0w or 100w …
limit setting from grid, all stop from peak consumer give a feed to grid
until inverter do the regulation and we are grid tie.
I want go to off grid when I have sufficient PV and battery and go grid tie
when there are not sufficient .
Normally we can drive the input relay of inverter to do that command .

N1c084, 2022-07-11:
Also during storm weather to protect the installation 😊

kellerza, 2022-07-11:
@N1c084 I'm not sure of any such relay on the inverter itself. 

You can probably add a relay before your installation, but I wont use anything less than a contactor there! And still not claiming that it's safe/complies with regulations.

If you are concerned about lightning you probably want to disconnect live, neutral and earth from the grid supply and then you can only rely on you local earth spike for the PV part of the install and any lightning surge from the load side.

N1c084, 2022-07-11:
Yes , there 2 internal Panasonic relay (neutral/phase cut) in série to
handle the grid tie function that are also vde compliant and sufficient to
handle on/ off grid power , and yes I have in first idea to drive a
contactor , but if we can drive this by the integrated grid tie function it
will be more easier and reliable.
Thanks

Maniac0Maniac, 2022-07-11:
Hi @kellerza 

Short version:
Yep. Got both inverters chatting happily in the end - Thanks Mate!

Long version:
Tried all the deployment methods but found I had a problem with the comms "stuttering" - data flow was intermittent. Never figured out why, but as soon as I changed to Modbus TCP it was a lot more reliable. Left it at that and moved on.

Two converters (one for each Inverter) plugged into an original model PI works a treat. Each instance of mbusd on the PI was configured with its own port, talking to the addons (one normal and one dev) . Data is flowing, Grafana graphs looking great and automations purring along!

Me and a mate did it at the same time and I will share what we learnt from the experience. Feel free to use it / add it to documentation / ignore!

I initially could not get the comms going between the inverters and mbusd. Tried and checked everything and kept failing. Also randomly lost coms after a few days ago after power down without changing anything. What fixed it in the end is a cold inverter power up with the battery disconnected, wait 5 mins, power up the PI, and then power up the battery. I also disconnected the new "Wifi Stick" for good measure and just hot added it in again after everything was working. Not your code but the inverters seems a little flaky here.

My mate didn't have a dual inverter setup and struggled to get comms going. Ends up that if the MODBUS_SERVER_ID is 0 you have to edit the YAML config file to '00' for it to work - might be worth adding that to the readme? Mine was 1 & 2 to get the inverters running in parallel and I used mbusd so never had the issue.

It would be great if we could the addon to be able to handle multiple inverters. What I can recommend (if this is achievable) is reading the register blocks for each inverter at the same time. With the 2 x addons strategy the inverters are read at different times and it creates havoc on the Grafana graphs, especially when you start comparing what the one inverter is doing compared to the other because the times don't sync up.

All that being said - Thank you for giving this Addon to the community - it's great piece of work!!!

kellerza, 2022-07-11:
Thanks for sharing.

`MODBUS_SERVER_ID` : Ideally you have to set it to something non-zero - this was added to the repo's README

Multiple inverters seems like a fairly significant amount of work,- and I have no way of testing that locally, so it will not be easy to implement. What might help is to NOT use any of the default filters - they can delay results for minutes. If you simply read every fixed amount of seconds from both inverters it might behave better in Grafana

N1c084, 2022-07-16:
> Yes , there 2 internal Panasonic relay (neutral/phase cut) in série to handle the grid tie function that are also vde compliant and sufficient to handle on/ off grid power , and yes I have in first [...]

Found a simple solution:

290 | Grid Hz Low | R/W | [4500 -6500] | 0.01Hz |  

Just need to write a out of tolerence value and inverter go off grid :-) 
Tested with qmod master

kellerza, 2022-07-16:
Not sure I’m comfortable rewriting out of tolerance values. Wont this cause the inverter to throw an error and shutdown for a minute or two?

N1c084, 2022-07-16:
> Not sure I’m comfortable rewriting out of tolerance values. Wont this cause the inverter to throw an error and shutdown for a minute or two?

No, at all, the inverter will just disconnect from the grid, my inverter has been in this state since 9:00.
I need to be able to schedul a scheduler to write 290 based on SOC/Load/PV prod

If you can add these sensors to rewrite: 20, 290, 210, 211, 293, 232, 245

N1c084, 2022-07-16:
> No, at all, the inverter will just disconnect from the grid, my inverter has been in this state since 9:00. I need to be able to schedul a scheduler to write 290 based on SOC/Load/PV prod
>
> If you [...]

here the proof of this working without behavior:
[image]

When off grid, all peak are handle by inverter, so no more peak from or to grid during this mode
[image]

Also asyou can see, i have some sensor named deye2 like my conf file and other with SS? Also not so important the device name is sunsynk and not deye

Thanks

cods4, 2022-08-22:
> The only way for now would be to run 2x instances of the addon. So you can run the normal and dev versions
> 
> The downside is you will probably need 2x usb to rs485 converters (you could try connecting both through mbusd and only mbusd connecting to the serial port)

I am running three deye inverters and am wondering if there is a way to run three instances of the add-on so I can monitor them all separately?

kellerza, 2022-08-22:
@cods4 are they all on the same RS485 bus (with different IDs), or separate?

cods4, 2022-08-22:
> @cods4 are they all on the same RS485 bus (with different IDs), or separate?

I was planning to connect them all to the same bus on different IDs and talk to them via a Waveshare RS232/485 to ETH media converter. But I have since read that I might need to have three different media converters to avoid collissions on the modbus.

I currently have things working with just my master inverter connected to the media converter, although I get frequent "ERROR   Timeout reading: " messages in the log.

kellerza, 2022-08-22:
> But I have since read that I might need to have three different media converters to avoid collissions on the modbus

Not sure where you read this, but this is true if you have 3 separate programs querying.

If the code was modified to support multiple modbus server ID's this would not be the case.

Can you check that these all use different Modbus IDs (think its slave ID in the Sunsynk panel) and then try address them individually by changing the `SERVER_ID` in the Homeassistant addon,- They might already be on a single bus.

cods4, 2022-08-22:
> > But I have since read that I might need to have three different media converters to avoid collissions on the modbus
> 
> Not sure where you read this, but this is true if you have 3 separate programs querying.
I read it here, towards the bottom of the page, comment by 'iiznh' https://powerforum.co.za/topic/11806-sunsynk-connection-to-home-assistant/
> If the code was modified to support multiple modbus server ID's this would not be the case.
>
> Can you check that these all use different Modbus IDs (think its slave ID in the Sunsynk panel) and then [...]

Yes they should all be configured with different Modbus IDs which is required for them to work in parallel. I can't communicate with the other inverters at this stage since I am only connected to the master inverter's RS485 port. But I will add the wiring from the other two inverters to my media converter and report back in the next few days.

One thing I am a little uncertain about is whether the inverters have multiple modbus networks. I.e one for BMS RS485, one for METER RS485 and one for parallel connection.
I guess the fact that I can't reach the other inverters through the master inverters BMS port indicates that this is separate. But it then seems a bit strange that changing the modbus server ID for parallel connection also affects the server ID for the BMS RS485 connection.

cods4, 2022-08-28:
So I have wired my other two inverters up to the RS485 to ETH converter in parallel and I have set up another instance of the HA Sunsynk Add-on to try and monitor a second inverter at the same time.

Not long after I started the second instance up I started getting errors on it and it eventually shutdown. Meanwhile the original instance kept running the whole time and just showed up repeated Timeout errors which it always does.

I configured both inverters to poll all known sensors.

Here are some logs and the configs for each inverter.
[INV1 log1.txt](https://github.com/kellerza/sunsynk/files/9439421/INV1.log1.txt)
[INV2 log1.txt](https://github.com/kellerza/sunsynk/files/9439423/INV2.log1.txt)
[INV2 log2.txt](https://github.com/kellerza/sunsynk/files/9439424/INV2.log2.txt)
[INV2_config.txt](https://github.com/kellerza/sunsynk/files/9439425/INV2_config.txt)
[INV1_config.txt](https://github.com/kellerza/sunsynk/files/9439426/INV1_config.txt)

cods4, 2022-08-28:
So I have wired my other two inverters up to the RS485 to ETH converter in parallel and I have set up another instance of the HA Sunsynk Add-on to try and monitor a second inverter at the same time.

Not long after I started the second instance up I started getting errors on it and it eventually shutdown. Meanwhile the original instance kept running the whole time and just showed up repeated Timeout errors which it always does.

I configured both inverters to poll all known sensors.

Here are some logs and the configs for each inverter
[INV1 log1.txt](https://github.com/kellerza/sunsynk/files/9439429/INV1.log1.txt)
[INV2 log1.txt](https://github.com/kellerza/sunsynk/files/9439430/INV2.log1.txt)
[INV2 log2.txt](https://github.com/kellerza/sunsynk/files/9439431/INV2.log2.txt)
[INV2_config.txt](https://github.com/kellerza/sunsynk/files/9439432/INV2_config.txt)
[INV1_config.txt](https://github.com/kellerza/sunsynk/files/9439433/INV1_config.txt)
.

kellerza, 2022-08-28:
My guess is they are way too chatty for two instances to work on the same bus.

If you run only 1 at a time, are they working as expected?

cods4, 2022-08-28:
Yes, they work normally if just one instance of the add-on is running.
What do you think the solution will be?
Can the add-on be modified to address each inverter sequentially and avoid these sorts of errors? Or will three rs485-eth converters be required?

Perhaps I could time how long it takes for one inverter to finish transmitting all the data, and we could work out how many inverters can be polled per minute.
For someone with up to 16 inverters they might need to reduce the polling rate right down to x minutes, or use multiple modbus adaptors.

cods4, 2022-08-31:
I've done a bit more testing, and I get parallel canbus errors (F29) if I parallel the modbus ports from each inverter.

So the only way to do this is use three modbus rtu - modbus tcp or something else converters

kellerza, 2022-08-31:
In that case (F29) trying to read multiple inverters on the same bus will not work...

Potentially this can be done if we can hook up to the CAN bus, but will require a completely different hardware and some sort of a CANBUS - Ethernet gateway (similar to the Modbus RTU gateway / mbusd)

cods4, 2022-08-31:
Yeh I was thinking the same thing. Canbus would be too much work.

I think the solution should be to use a separate modbus network for each inverter. I guess it's a matter of adding a method of addressing multiple inverters from the one app. Or otherwise running three instances of the app.

I have three USR DR-302 din rail mount rs485-eth converters on their way which I will wire to each inverter.

N1c084, 2022-08-31:
But add a can bus read/write for BMS address (pylon tech emulator)will be a very good option to an homemade battery pack

kellerza, 2022-08-31:
> Can bus is only for BMS, no user access .

In @cods4 case it also seems to carry inverter to inverter comms as well (causing the F29 error)

> I have 2 inverters with 1 Ethernet to multiple modbus rtu ports. I will try with two instance of addon.

:+1: as long as these ports are isolated, this seems like a good piece of HW once you parallel. What HW is that?

> Can we add multiple modbus port and inverter on same instance easily ?

fairly easy, if you still treat them separately...

> Use one instance can help to obtain better data reliability for example load/grid charge ...power need to be addition of all inverters power .

This part is a bit more complicated through! as you are talking about a sensor that is not linked to an inverter, but rather to a logical group of inverters (technically that is a new device in HASS).

There is also no way to synchronize reads & filters across multiple inverters. Well maybe reads can be synchronized

N1c084, 2022-08-31:
[image]

This one I have 4 rs485 or rs232 to 4 différents ip output but only one port on device(like an Ethernet switch)

cods4, 2022-08-31:
> In @cods4 case it also seems to carry inverter to inverter comms as well (causing the F29 error)

Yeh it's all a bit strange as in the settings you have to set a unique modbus address for each inverter, and obviously messing with the RS485 port afffects the parallel connection. But the error you get when you have a parallel communications problem is a canbus error.

Possibly the parallel connection uses a can bus and mod bus on separate twisted pairs. The canbus is probably used for fast communications such as watchdogs, voltage, current flows etc (just the stuff the inverters need to operate in synchronism to react to loads etc). And the modbus is probably used for transferring larger and less time critical data packets like settings etc.

> fairly easy, if you still treat them separately...

That's great news. I think that will be good enough for me. It should be possible to create a template sensor in HA to add the values together or average them out or whatever is needed.

> This one I have 4 rs485 or rs232 to 4 différents ip output but only one port on device(like an Ethernet switch)

Nice one, it would be good for what I need as I would only need to use one ethernet cable back to my main switch, but unfortunately I have already ordered the 3 separate USR converters.

N1c084, 2022-08-31:
> fairly easy, if you still treat them separately...

How to add second inverter?

first cfg is:
PORT: 192.168.0.20:502
DEVICE: /dev/ttyS0
DRIVER: pymodbus
SUNSYNK_ID: "xxxxxxxxxxxxx"
SENSOR_PREFIX: deye2

second need to be:
PORT2: 192.168.0.15:502
DEVICE2: /dev/ttyS0
DRIVER: pymodbus
SUNSYNK_ID2: "xxxxxxxxxxxxxx"
SENSOR_PREFIX2: deye1

N1c084, 2022-08-31:
I run the standard add on on my deye 1 inverter and dev on my deye 2 inverter and both working fine.
The com and data are ok.
If we could easily add the possibility of having several inverter servers on the same add on, it would be easier to manage.
Maybe that way we could add some sum data representing the global system

cods4, 2022-08-31:
> How to add second inverter?

I think he was meaning it wouldn't be hard to add the ability to address multiple inverters. I don't think the code can handle that at the moment.

Maniac0Maniac, 2022-09-13:
Hi @kellerza

Still happily running my dual dev/release add-ons to monitor both inverters and I have bumped into some problems. Would appreciate your thoughts.

1) The sample times, load_power in this screenshot, seems random. Sometimes they come in quick succession, other times a minute or so apart. Not ideal when this happens alongside another inverter as well. They are out of sync and using the sum of the two in dashboards becomes a pain. Any idea why this would be? Syncing the readings so that they happen at the same time would be nice.

[image]

2) Would it be possible to sample some key values like load_power sampled quicker (say every 10 secs) and other ones like battery charge less frequent? Would make dashboard dials a lot cooler!

kellerza, 2022-09-15:
@Maniac0Maniac you could try to add cusom filters to your sensor definitions. They are partly documented [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-dev/DOCS.md#sensor-modifiers---minmaxnowaveragestep). Some of them you can customize some of the timers by adding another colon `:` and a value.

If you edit your config in YAML, you simply append these to the sensor name

```yaml
SENSORS:
  - grid_voltage:FILTER:A:B
```

(You will have to look at the code to see exactly how A&B behaves, not sure I documented it. I might also confuse it with my SMA-EM addon :wink:.... but it's surely possible)

Maniac0Maniac, 2022-09-17:
Hi @kellerza - No idea how I missed that! Thanks for the pointer - worked a treat!

cods4, 2022-10-02:
Any progress on adding ability to connect to 2 or more inverters?
I now have my 3 separate media converters connected to the 3 inverters. I can connect to each independantly or monitor two at a time by running the normal build and the dev build. But I can't monitor all three at once.

Is there a quick and easy way to get another instance of the add-on running?
[image]

amctear, 2023-01-30:
Johann,
I once had a FOX twin inverter set up with batteries which didn't work I may add, hence moved to the light with twin sunsynks. 

Anyway, 
I had an HA connection to both inverters via a custom modbusLAN.yaml. I dont know the 1st thing about how it worked but wonder if its possible to do something similar with multiple Sunsynks ?

[image]

kellerza, 2023-01-31:
> Anyway, I had an HA connection to both inverters via a custom modbusLAN.yaml. I dont know the 1st thing about how it worked but wonder if its possible to do something similar with multiple Sunsynks ?

Not today, your best option would be to have multiple addons running (normal & the dev)

kellerza, 2023-01-31:
> Is there a quick and easy way to get another instance of the add-on running?

You can create a third instance by running a local addon. If you share /addons through samba the scripts\copy2local.cmd script copies the dev version to the /addons share

amctear, 2023-01-31:
> Thanks,
Got both addons running, a few errors here and there, but running well.  Now begins the process of correlating this new data with the smart meter consumption and learning more of HA to add the peak & off peak helpers :-)

yksxjan, 2023-02-19:
Solar-assistant (to my understanding) also needs separate parallel inverters to be connected to a another USB port. However, the number is not limited (maybe cannot handel 16 :)).

The whole master-slave setup should copy the settings automatically from master to all slaves. At least this is how it should work in my understanding. So for setting inverter parameters, is it OK just to connect to master and the only problem is, we cannot monitor the individual inverters and sum the data?

cods4, 2023-02-19:
I haven't done much testing with changing settings using this add-on, but that was my understanding of how it should work. You make the changes to the master inverter and it would automatically propagate to the other inverters. At least that's how it works when making settings changes on the inverter HMI.

bruceborrett, 2023-03-10:
I am also interested in running this addon with 3 inverters, I want to try the approach described here - [https://community.home-assistant.io/t/cloning-the-z-wave-js-to-mqtt-add-on-to-run-multiple-z-wave-networks/415863](url)

I will feedback with my results.

bruceborrett, 2023-03-10:
Or alternatively, I could perhaps assist with modifying the addon to support multiple inverters. Ive had a quick look through the code and it seems the following is needed:

1. Modify DEVICE and SUNSYNK globals to be lists
2. Iterate those lists where necessary
3. Prefix all sensor names with the inverters name
4. Modify addon settings to allow definition of multiple inverters

Am I missing any other major changes needed?

cods4, 2023-03-10:
> Or alternatively, I could perhaps assist with modifying the addon to support multiple inverters. Ive had a quick look through the code and it seems the following is needed:
>
> 1. Modify DEVICE and [...]

It would be awesome if you could have a go at that. I'm currently running three instances of the add-on in HA, but this makes it fairly difficult to update.

I'm no coder, but what you are saying makes sense.

It would probably be good if you could make it so that it doesn't affect anyone who has the add-on running with a single inverter. I.e we don't want it appending a number to their inverter prefix which would stuff up their HA configs.

Also it would be nice if it was possible to define a separate list of parameters to pull from the master inverter and the slave inverters, as some values aren't required from the slaves such as grid current, power etc.

bruceborrett, 2023-03-11:
Unfortunately this would be a breaking change and it would be very difficult to make it backwards compatible. This repository currently contains 3 add-ons (mbusd, main and dev), so in my opinion the best option is to create a 4th add-on specifically for multiple inverters, then existing installs will be unaffected.

kellerza, 2023-03-12:
@bruceborrett you are welcome to assist with adding multiple inverters, I have started on this earlier this year and merged my work in #106 

The main change in this is that the sensors definitions & state have been split. In the past, since state was part of the sensor definition, it was not possible to read from multiple inverters. Now a sensor definition has simple `value_to_reg` and `reg_to_value` functions. State is now kept by the Sunsynk class and this enables the concept you mention below:

> * Modify DEVICE and SUNSYNK globals to be lists

mostly done - only Sunsynk is a list afair, device is simply part of Sunsynk

> * Iterate those lists where necessary

currently I have a fixed reference to the first item in the list / a static [0]. Iterating needs to be added

> * Prefix all sensor names with the inverters name

this is part of the configuration (see below) - but ideally there should be a check to make sure this prefix is unique, which is outstanding.

> * Modify addon settings to allow definition of multiple inverters

mostly done

If you want to contribute you can test this new version - for now try it as a single addon. And if you are happy then you are welcome to submit a PR to start the iterations.

bruceborrett, 2023-03-13:
I get the following error when I try to save the config using the new multi addon, the only :

`Failed to save add-on configuration, invalid value or type (must have a partial ordering).`

Only thing I changed is the device:

```
- SERIAL_NR: "007"
  HA_PREFIX: SS
  MODBUS_ID: 1
  PORT: ""
  DEVICE: "/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_D-if00-port0"
```

bruceborrett, 2023-03-14:
I managed to get rid of the error by changing `MODBUS_ID: int(1,10)` to just `MODBUS_ID: int` in the config schema, not sure why that range limit was causing issues.

So now it is connecting to the first inverter and reading values correctly.

Will look into adding the iterations now.

kellerza, 2023-03-14:
Glad you found the issue. So far I've only tested it locally on my laptop and never tried to build it on Home Assistant

[int(1,10)](https://developers.home-assistant.io/docs/add-ons/configuration#options--schema) seems quite valid ;-)

bruceborrett, 2023-03-15:
Unfortunately this seems a lot more complicated then simply iterating the list in the right places. There is a lot of cross referencing between the OPT.inverters list and the SS list, and in many places the serial number and modbus ID are used in ways that are not dynamic (ie assumed to be unchanging). Im also not sure how write_sensors would work since the write would need to be addressed to the appropriate inverter.

This is a bit more work then I am willing to commit to right now, but I also am unable to simply make 3 copies of the single inverter addon since I need the 3-phase definitions, so I need to install the sunsynk library locally, but the current version of the sunsynk library is now incompatible with the original addon, so seems things have gotten a bit messy now.

kellerza, 2023-03-19:
There is one more addon to try - the new multi addon

This one can handle other definitions & custom sensors, but no option to do multiple inverters yet... Just one more instance to run.

New docs here - https://kellerza.github.io/sunsynk/guide/intro

kellerza, 2023-03-19:
> 3-phase definitions, so I need to install the sunsynk library locally, but the current version of the sunsynk library is now incompatible with the original addon, so seems things have gotten a bit messy now.

Once you use the copy2local script the addon uses the local version of the library

kellerza, 2023-05-05:
The latest multi addon should support multiple inverters in one addon. Please try it out and let me know.

There is no syncing of filters today, so `:now` is probably still important for power graphs etc

All inverters will use the same set of sensors, and in the future I'll add some support to have additional sensors for the first inverter (i.e. configuration)

N1c084, 2023-05-05:
Hi
Can we add multiple standalone inverter ?
I have 2 unit but not in //
I will check this later
Thanks for the job👍🏼

kellerza, 2023-05-05:
It can be any combinations of inverters. I test it by adding one/more underscores to my device's serial.. but they read the same inverter and even the same port

```yaml
INVERTERS:
  - SERIAL_NR: "2105150000"
    MODBUS_ID: 1
    HA_PREFIX: "SS"
    PORT: "tcp://192.168.1.230:502"
  - SERIAL_NR: "2105150000_"
    MODBUS_ID: 1
    HA_PREFIX: "SS2"
    PORT: "tcp://192.168.1.230:502"
```

N1c084, 2023-05-15:
Hi 
Just testing now on both inverters !
Working great , now I have my 2 inverters connected to HA.
Thanks again

amctear, 2023-05-15:
Working perfect for my 2 parallel inverters and almost zero read errors.   Great Job, Thanks

kellerza, 2023-05-16:
Great, thanks for the feedback guys!

[1 more reply]

closed 2023-05-16 by kellerza