kellerza/sunsynk #40: Reading more sensor values in Home Assitant

Bravewildheart, 2022-06-12:
Hi,

Could someone please be so kind to explain to me step by step how to get all the values from the Sunsynk Inverter in Home Assistant.

I am getting the following:
[image]

I would like aditional Values like this:
[image]

Any help would be greatly appreciated.

Thank you

CunningSquirrel, 2022-06-22:
I’m having the same issue, can’t figure out how to pull more info / sensors. Not sure where this integration stores it’s files and folders within Home Assistant, e.g configuration.py etc. if I can find the folder, I could try and figure it out myself.

Looking forward to a reply.

kellerza, 2022-07-11:
@CunningSquirrel @Bravewildheart - did you manage to add more?

Bravewildheart, 2022-07-11:
> From "Johann Kellerman" ***@***.***>
> To "kellerza/sunsynk" ***@***.***>
> Cc "Bravewildheart" ***@***.***>; "Mention"
> Date 2022/07/11 09:14:42
> Subject Re: [kellerza/sunsynk] Reading more sensor values in Home
> Assitant (Issue #40)

Yes thank you.

closed 2022-07-11 by kellerza