kellerza/sunsynk #42: I can't start dev addon

agtconf, 2022-07-12:
### Issue related to

Sunsynk / Home Assistant (dev) Add-On

When I try to start the addon it returns the following error

Device '/dev/ttyUSB0' does not exist in Sunsynk Inverter Add-on (dev) (4bb530c7_hass-addon-sunsynk-dev).

I try to delete device entry on yaml, but it always appears again.

If I try to save the yaml with device variable 

Failed to save add-on configuration, Device '/dev/ttyUSB0' does not exist in Sunsynk Inverter Add-on (dev) (4bb530c7_hass-addon-sunsynk-dev). Got {'PORT': 'tcp://192.168.1.141:502', 'DEVICE': '/dev/ttyUSB0', 'DRIVER': 'umodbus', 'SUNSYNK_ID': '2012104132', 'SENSOR_PREFIX': '', 'SENSORS': ['pv1_power', 'grid_power', 'load_power', 'dc_transformer_temperature', 'radiator_temperature', 'battery_power', 'total_active_power', 'total_pv_power', 'total_battery_charge', 'total_battery_discharge', 'total_grid_export', 'total_grid_import', 'battery_soc', 'device_type', 'pv1_current', 'pv1_voltage', 'grid_connected_status'], 'READ_SENSORS_BATCH_SI...

I don't have any ttyUSB connected to the server where HA is.

kellerza, 2022-07-12:
Try selecting a device that exists from the UI, like `/dev/ttyAMA0` (which is wrong obviously, but it wont use it).

I'll continue working on this tonight. Unfortunately it's not easy to "release" addons, and HASS OS always pulls the latest from GitHub.

That is why there is a `-dev` and a normal version. Golden rule, don't just update the -dev version. Especially if I commit after 10pm in the evening ;-)

kellerza, 2022-07-12:
I've made the default blank, so you could try: 
1. First UNINSTALL the addon. 
2. Refresh the store and 
3. REINSTALL.

Remember to save your yaml config before you uninstall.

kellerza, 2022-07-12:
Glad it works. You can probably go into yaml mode in the GUI as well and then delete the offending `DEVICE: /tty/x` entry, or just set it to an empty string

closed 2022-07-12 by kellerza