kellerza/sunsynk #44: Sensor "day load energy " and "grid status" not working

N1c084, 2022-07-16:
Hi

 Inverter: deye / 8k eu hybrid/ hmi=c358 main=6014-0717

sensor "day_load_energy" is reported as unknow

i already have a look at closed issue about the grid status sensor but even with filter adjustment" :last" it not working.

Thanks

```
 Loading HASS OS configuration
2022-07-16 12:57:27,666 INFO    ############################################################
2022-07-16 12:57:27,666 INFO             Unknown sensor in config: day_load_energy
2022-07-16 12:57:27,667 INFO    ############################################################
2022-07-16 12:57:27,667 INFO    ############################################################
2022-07-16 12:57:27,667 INFO        Unknown sensor in config: grid_connected_status:last
```

kellerza, 2022-07-16:
Both these will only be on the dev version of the addo

closed 2022-07-16 by kellerza

N1c084, 2022-07-16:
sorry for borring you for nothing :-/  no error now in dev  version , why these 2 sensor show depracated in dev version only?

Sorry, i'm new, how to request add new RW sensors? i need to do some wscheduler on W sensors depending of state of other rw sensors.
>need to open issue or pull request?

2022-07-16 13:27:55,122 INFO    ############################################################
2022-07-16 13:27:55,122 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2022-07-16 13:27:55,123 INFO    ############################################################
2022-07-16 13:27:55,123 INFO    ############################################################
2022-07-16 13:27:55,123 INFO      Sensor deprecated: total_load_power -> total_load_energy
2022-07-16 13:27:55,123 INFO    ############################################################
2022-07-16 13:27:55,124 INFO    Filter *last used for battery_soc, day_load_energy, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_load_power, total_pv_power
2022-07-16 13:27:55,124 INFO    Filter *step used for battery_current, battery_power, grid_power, inverter_frequency, inverter_power, overall_state, pv2_current, pv2_power
2022-07-16 13:27:55,124 INFO    Filter *avg used for battery_temperature, battery_voltage, dc_transformer_temperature, grid_voltage, inverter_voltage, pv2_voltage, radiator_temperature
2022-07-16 13:27:55,125 INFO    Filter *round_robin used for fault
2022-07-16 13:27:55,125 INFO    Filter last used for grid_connected_status
2022-07-16 13:27:55,129 INFO    Protocol made connection.

kellerza, 2022-07-16:
Those sensors still work, just renamed.

Technically they report energy and not power, so just change yours