kellerza/sunsynk #49: MQTT / mbusd gateway via Tasmota?

Maniac0Maniac, 2022-07-20:
Question related to
Sunsynk / mbusd Home Assistant Add-On

Describe the issue
Hi KellerZA

I'm thinking of trying to replace my PI with an ESP device - It would be great and have loads of benefits. I considered trying to write my own code, but thought why re-invent the wheel? Did a quick google to see if Tasmota would be able to do something out of the box and I stumbled across these:

TCP Gateway - arendst/Tasmota#9586
MQTT Gateway - arendst/Tasmota#15703

I think they might have potential but honestly, don’t know where to begin. I don't have enough understanding of modbus, your addon, etc. (yet) to know if it's feasable. Think you can give them a quick once over?

If your verdict is that they have potential I'm happy to put in the time to see if we can add another deployment option.

kellerza, 2022-07-20:
@Maniac0Maniac There is someone that buil an MQTT gateway this on Powerforum with ESPHome, but you are re-implementing the entire definitions file in ESPHome. It wont be as flexible, you won't have filters and its doubtful if/how you'll be able to write settings.

The TCP gateway could potentially work with this addon, so you could potentially look at this

I'm not such a big fan of Wifi for streaming telemetry data like this. It is intentionally polling very frequently in order to be responsive. And the golden rule for anything in my smarthome is no soldering. I've built my own Electric Imp boards in the past, but after a lightning strike I now buy ready made devices (and generally keep spares as well :wink:)

My ideal setup would be a Home Assistant Server with [this](https://www.robotics.org.za/W16529) as gateway. Today I'm using a Pi Model B + USB adaptor to replicate the RS485 gateway, but it work flawlessly so far

Maniac0Maniac, 2022-07-20:
Hi KellerZA

Ah! You make an excellent point that I missed. Wi-Fi streaming is probably not ideal especially if you're running automations off it. Guess my PI and RS485 converters are just get tucked into the trunking as is :)

Thanks

jeroenst, 2022-07-20:
Mqtt modbus gateway is now in development branch.

Documentation is also in development branch of tasmota docs.

jeroenst, 2022-07-20:
The modbus tcp bridge is on its way, it's currently in testing phase before a pr will be provided.

Probably it will be finished this week.

closed 2022-07-31 by kellerza