kellerza/sunsynk #50: rw sensor not change state in real

N1c084, 2022-07-30:
Hi 
in last revision
I changed the low  batt value , we can see the write state chaging but the read state stay the same, so inverter do not take the value.

[image]

Is it possible to Read/write value from nodeRed to RW sensors?

thanks

Ivan-L, 2022-08-01:
Hi @N1c084, just to understand, did you change the "Battery Low Capacity" register on the inverter itself and you do not see the effect in the add-on?
Currently the "Battery Low Capacity" sensor is not a R/W sensor in this add-on, so I am just trying to understand how you were able to change it if not on the inverter?

N1c084, 2022-08-04:
Hi @Ivan-L 
I added the sensor to set it to HA, but I confused the "prog 2 capacity" and "low battery capacity" sensors which is not R/W, sorry for the confusion.

closed 2022-08-04 by N1c084