kellerza/sunsynk #59: New version of Deye/Sunsynk Modbus Protocol Document

cods4, 2022-09-07:
I have come across a new version (V118) of the Deye Modbus RTU Protocol Document.
The previous version of this document that I had access to was V115. The changes in relation to V115 are the following

#248 - More options added to time of use selling, can now enable and disable time of use selling for each day of the week. I'm not 100% sure how this translates to the inverter settings, as you can only enable or disable grid charging or generator depending on the day of the week in the inverter HMI. (Plus there is some special option for Spain).

#274-279 - More access to time of use settings has been added, including enabling/disabling grid charge, enabling/disabling generator charge, and something about BM mode, BU mode and CH mode.

#181 - Looks like it should be Gen port voltage, not certain.

#62 - Changed from Day grid work time, to Day gen energy Wh

#92 & 93 - Changed to low and high values of total energy generation of the generator.
[Modbus储能-组串-微逆宁波德业V118.pdf](https://github.com/kellerza/sunsynk/files/9504873/Modbus.-.-.V118.pdf)

kellerza, 2022-09-15:
Thanks @cods4 do you know if this requires new firmware, or was it simply mistakes in v115?

Not sure if we need more sensors? There has been some slight shifts in the later version of the library and probably best to use the dev version of the addon when doing any comparisons

cods4, 2022-09-15:
> Thanks @cods4 do you know if this requires new firmware, or was it simply mistakes in v115?
> 
> Not sure if we need more sensors? There has been some slight shifts in the later version of the library and probably best to use the dev version of the addon when doing any comparisons

My system is still not quite commissioned, so haven't done too much testing yet. But I guess that some of these changes would only apply to the latest firmware.

Agreed, not much need for these additional sensors, but they might be useful for someone in the future.

Zuikkis, 2022-11-06:
This document has interesting modbus register 205 "Empty_V"

By default there's 45000 value, = 45.0V

The register is not documented and can't be changed from inverter menus. However this setting is the voltage level that corresponds to SOC 0%. 

[image]

Here's a screenshot when happens when battery voltage drops below Empty_V... SOC drops from 72% to 0% instantly. 

Even if you have configured your battery with voltage levels instead of %, inverter really doesn't work properly when voltage drops below Empty_V. Only way to draw power from battery below Empty_V seems to be the "grid peak limit" feature.

With Empty_V = 41000 I was able to get better use of my 12S lithium pack which only has a nominal voltage of 44V..

akoxa, 2023-11-13:
I am pretty confused with modbus document V118 when reflecting to definitions3ph.py. I don't get how modbus addresses and document correlates between each other. For example (there are many of those): 

**definitions3ph.py**
`Sensor(633, "Inverter L1 power", WATT, -1),`

**modbus document**
633 - PACK 3 - Total Capacity

How this should be interpreted ?

Zuikkis, 2023-11-13:
@akoxa The linked pdf is for 1-phase inverters only, so the matching file is definitions.py

Apparently 3-phase uses completely different protocol.

akoxa, 2023-11-14:
Thanks, but this is very misleading as the document is defining 3 phase values like phase power / phase. I was also reading issue #63 relating to the 3 phase definitions. 

Can @lubster85 or @sebmueller help on where to find the correct document for modubs parameters ? 

There was also a document https://github.com/kellerza/sunsynk/files/9671512/sunsynk_modbus.-.no.chinese.docx, in the same discussion, but it's register addresses don't match with definitions3ph.py.

Zuikkis, 2023-11-14:
> Thanks, but this is very misleading as the document is defining 3 phase values like phase power / phase. 

I'm sorry but I have to disagree? There simply is no mention of 3 phases in the pdf. Also the .docx you linked seemes to be only for single phase setup.

However the values in definitions3ph.py seems to be correct, what more do you need?

kellerza, 2023-11-14:
There was a document in #63 
The one in this issue is for single phase only

Alternatively search on the solar forums

akoxa, 2023-11-14:
Thanks, I found the document from German forum. 

@Zuikkis: I am building multi-battery system and it seems I can read battery statuses with Deye, but those parameters was not in definitions3ph.py and I am trying to build my custom sensors as per instuctions.

closed 2025-03-18 by kellerza