kellerza/sunsynk #60: Only Some Sensors not reading

hazardrr, 2022-09-11:
Hi There

i have HA installed on a Raspberry Pi 4. I purchased this USB to RS485 and its connected to RS485 on my Sunsynk 8KW Inverter
https://www.robotics.org.za/index.php?route=product/product&product_id=5947

I installed the Add-on mid August and it worked fine even though i kept getting Timeout Errors in the log. A few days ago, i did maintenance on the inverter an shortened and re-crimped the CAT5 Cable running to the USB-RS485

Ever since then, i have been unable to get a reading on Total Grid import sensor. I re-install and redownloaded the addon (which will shows 0.14 by the way) and it still just timeouts with that sensor specifically.

I then added a few other sensors and the other sensors work fine except for Essential and non Essential Loads. So now 3 sensor readings keep timing out

Here is my log output:

2022-09-11 15:55:16,953 INFO    Loading HASS OS configuration
2022-09-11 15:55:16,970 DEBUG   could not patch umodbus: numpy not found
2022-09-11 15:55:16,975 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_load_power, total_pv_power
2022-09-11 15:55:16,975 INFO    Filter *step used for essential_power, grid_ct_power, grid_ld_power, grid_power, load_power, non_essential_power, pv1_power, pv2_power
2022-09-11 15:55:16,975 INFO    Filter *avg used for grid_voltage
2022-09-11 15:55:16,977 INFO    Connecting to serial:///dev/ttyUSB0
2022-09-11 15:55:17,073 INFO    ############################################################
2022-09-11 15:55:17,073 INFO                Inverter serial number '220628xxxx'
2022-09-11 15:55:17,073 INFO    ############################################################
2022-09-11 15:55:17,075 INFO    MQTT: Connecting to mosquittousr@core-mosquitto:1883
2022-09-11 15:55:17,107 INFO    MQTT: Connection successful
2022-09-11 15:55:28,123 ERROR   Timeout reading: 
2022-09-11 15:55:28,124 INFO    Retrying individual sensors: ['Total PV Power:last', 'Total Battery Charge:last', 'Total Battery Discharge:last', 'Total Grid Export:last', 'Battery SOC:last', 'Grid power:step:mean', 'Load power:step:mean', 'PV1 power:step:mean', 'PV2 power:step:mean', 'Total Load Power:last', 'Grid CT power:step:mean', 'Grid LD power:step:mean', 'Grid voltage:mean', 'Essential power:step:mean', 'Non-Essential power:step:mean', 'Total Grid Import:last']
2022-09-11 15:55:39,196 ERROR   Timeout reading: Essential power
2022-09-11 15:55:49,232 ERROR   Timeout reading: Non-Essential power
2022-09-11 15:55:59,264 ERROR   Timeout reading: Total Grid Import
2022-09-11 15:56:09,320 ERROR   Timeout reading: 
2022-09-11 15:56:19,335 ERROR   Timeout reading: 
2022-09-11 15:56:29,348 ERROR   Timeout reading: 
2022-09-11 15:56:39,363 ERROR   Timeout reading: 
2022-09-11 15:56:49,383 ERROR   Timeout reading: 
2022-09-11 15:56:59,401 ERROR   Timeout reading: 
2022-09-11 15:57:09,421 ERROR   Timeout reading: 

I also tried adding TIMEOUT: 20 to the config file but it still does it in 10sec intervals

Your assistance would be highly appreciated

Thank you

hazardrr, 2022-09-11:
just something i noticed as well, its actually not updating any values anymore. i reboot add-on then it pulls data only for that time. it doesn't seems to be updating on its own

hazardrr, 2022-09-14:
ok, i found the issue. For some reason it just stopped working after i re-routed the cable. Possibly interference. The cable is 36M long and even though it worked before, i suspect that closing it up in trunking caused some more interference because if I put my Home Assistant unit right by the inverter it works fine. 

I would like to increase the timeout to see if that can help because of the distance of the cable but when i change timeout, its just still 10seconds. How can i change it to 15 or 20 seconds?

kellerza, 2022-09-15:
Hi Hazard, I will have to look into why the timeout is not being used. 

I assume you use 3x wires (D+/D- and GND). the data lines MUST be on a twisted pair in the cable to reduce noise, but even then the interference might be too much. Another option might be a Shielded UTP cable. Then you ground one end of the shield to an external ground (You don't necessarily have to use metal STP connectors)

hazardrr, 2022-09-15:
Hi Johann

Thanks, much appreciated, that would be great with the timeout function. I am using Cat5e yes and I and using 3 wires as per diagram yes. Strange thing is that it worked before. Now that I neatly put away the cables in the trunking and cut off excess wires it just stopped working completely. It reads on start up but then just times out so it does definitely seems like some interference (for about 10 meters its running in the same trunking as the solar DC cables). This is why i would like to try to increase the timeout to see if that will possibly work to accommodate for distance and interference. 

If that doesn't work, i have another Raspberry Pi right by the inverter which I am using for reporting on my battery via Pylontech cable, I could maybe use that Raspberry Pi as a modbus gateway as a last resort

kellerza, 2023-01-19:
@hazardrr the latest dev version tries to propagate this timeout for the usunsynk driver in the dev version of the addon, so might help!

btw, I use an old Raspberry Pi 1b running mbusd (no Hass io) as a gateway with a wave USB adaptor

If I would start today I would likely choose [this](https://www.robotics.org.za/W16529?search=wave%20rs4) Wave RS485 to Ethernet unit

closed 2023-01-19 by kellerza

jabezz, 2023-01-19:
This latest published version today broke my add-on. It cannot start up.
Not infront of a pc to debug yet.

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-01-19 14:34:05,573 INFO    Loading HASS OS configuration
2023-01-19 14:34:05,574 INFO    PROFILES were deprecated. Please remove this configuration.
2023-01-19 14:34:05,601 DEBUG   could not patch umodbus: numpy not found
2023-01-19 14:34:05,613 INFO    ############################################################
2023-01-19 14:34:05,614 INFO      Sensor deprecated: day_active_power -> day_active_energy
2023-01-19 14:34:05,614 INFO    ############################################################
2023-01-19 14:34:05,614 INFO    ############################################################
2023-01-19 14:34:05,615 INFO        Sensor deprecated: day_load_power -> day_load_energy
2023-01-19 14:34:05,615 INFO    ############################################################
2023-01-19 14:34:05,616 INFO    ############################################################
2023-01-19 14:34:05,616 INFO    Sensor deprecated: day_reactive_power -> day_reactive_energy
2023-01-19 14:34:05,616 INFO    ############################################################
2023-01-19 14:34:05,617 INFO    ############################################################
2023-01-19 14:34:05,617 INFO      Sensor deprecated: month_grid_power -> month_grid_energy
2023-01-19 14:34:05,617 INFO    ############################################################
2023-01-19 14:34:05,618 INFO    ############################################################
2023-01-19 14:34:05,618 INFO      Sensor deprecated: month_load_power -> month_load_energy
2023-01-19 14:34:05,618 INFO    ############################################################
2023-01-19 14:34:05,619 INFO    ############################################################
2023-01-19 14:34:05,619 INFO        Sensor deprecated: month_pv_power -> month_pv_energy
2023-01-19 14:34:05,619 INFO    ############################################################
2023-01-19 14:34:05,620 INFO    ############################################################
2023-01-19 14:34:05,620 INFO    Sensor deprecated: total_active_power -> total_active_energy
2023-01-19 14:34:05,620 INFO    ############################################################
2023-01-19 14:34:05,621 INFO    ############################################################
2023-01-19 14:34:05,621 INFO      Sensor deprecated: total_load_power -> total_load_energy
2023-01-19 14:34:05,622 INFO    ############################################################
2023-01-19 14:34:05,622 INFO    ############################################################
2023-01-19 14:34:05,622 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2023-01-19 14:34:05,623 INFO    ############################################################
2023-01-19 14:34:05,623 INFO    ############################################################
2023-01-19 14:34:05,624 INFO       Sensor deprecated: year_load_power -> year_load_energy
2023-01-19 14:34:05,624 INFO    ############################################################
2023-01-19 14:34:05,624 INFO    ############################################################
2023-01-19 14:34:05,625 INFO         Sensor deprecated: year_pv_power -> year_pv_energy
2023-01-19 14:34:05,625 INFO    ############################################################
2023-01-19 14:34:05,627 INFO    Filter *step used for aux_power, battery_current, battery_power, control_mode, day_reactive_power, essential_power, grid_ct_power, grid_current, grid_frequency, grid_l2_power, grid_ld_power, grid_power, inverter_frequency, inverter_power, load_l1_power, load_l2_power, load_power, non_essential_power, pv1_current, pv1_power, pv2_current, pv2_power
2023-01-19 14:34:05,627 INFO    Filter *last used for battery_soc, day_active_power, day_battery_charge, day_battery_discharge, day_grid_export, day_grid_import, day_load_power, day_pv_energy, grid_connected_status, month_grid_power, month_load_power, month_pv_power, total_active_power, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_load_power, total_pv_power, year_grid_export, year_load_power, year_pv_power
2023-01-19 14:34:05,627 INFO    Filter *avg used for battery_temperature, battery_voltage, grid_voltage, inverter_voltage, pv1_voltage, pv2_voltage
2023-01-19 14:34:05,628 INFO    Filter *round_robin used for battery_absorption_voltage, battery_equalization_voltage, battery_float_voltage, battery_low_capacity, battery_low_voltage, battery_restart_capacity, battery_restart_voltage, battery_shutdown_capacity, battery_shutdown_voltage, load_limit, priority_mode, prog1_capacity, prog1_charge, prog1_mode, prog1_power, prog1_time, prog1_voltage, prog2_capacity, prog2_charge, prog2_mode, prog2_power, prog2_time, prog2_voltage, prog3_capacity, prog3_charge, prog3_mode, prog3_power, prog3_time, prog3_voltage, prog4_capacity, prog4_charge, prog4_mode, prog4_power, prog4_time, prog4_voltage, prog5_capacity, prog5_charge, prog5_mode, prog5_power, prog5_time, prog5_voltage, prog6_capacity, prog6_charge, prog6_mode, prog6_power, pro [... cut ...]
  File "/usr/local/lib/python3.9/site-packages/serialio/__init__.py", line 17, in serial_for_url
    return aio.serial_for_url(url, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/serialio/aio/__init__.py", line 19, in serial_for_url
    return Serial(url, *args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'connection_timeout'
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.`

kellerza, 2023-01-19:
@jabezz can you try updating again, it seems connection_timeout is not supported by serial drivers. I only have a TCP test system

jabezz, 2023-01-19:
Perfect. Working again. Much appreciated.