kellerza/sunsynk #63: Compatibility with Sunsynk 12kw 3-phase inverter?

lubster85, 2022-09-25:
Hi there

I recently had a Sunsynk 12kw 3-phase inverter installed, but I'm a little unsure if this integration works with this model.
I purchased a USB-RS485 cable from Solar-Assistant (https://solar-assistant.io/shop/products/sunsynk_rs485) and tried starting up the integration.

I've tried a few ports on the inverter with no success yet, but I was hoping to sense-check myself here as I don't recall the 3-phase inverter being included in the list of tested inverters.

[image]

My understanding is that the "Meter-485" port could be used for monitoring and also the "BMS" port.

Referencing Solar Assistant, they claim there was a firmware change recently.
> The inverter manufacturer released a firmware update which changed the port to read the inverter via RS485. An RJ45 splitter can be used to allow SolarAssistant to read the inverter while the inverter is also reading a battery via CAN.

Is there perhaps something else I may be missing or should try? 
If not, I'll try find a splitter and confirm if that makes a difference.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-09-24 08:45:31,191 INFO    Loading HASS OS configuration
2022-09-24 08:45:31,264 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_power
2022-09-24 08:45:31,273 INFO    Protocol made connection.
2022-09-24 08:45:31,274 INFO    Connected to /dev/ttyUSB0
2022-09-24 08:45:41,286 ERROR   Timeout reading: 
2022-09-24 08:45:41,287 INFO    ############################################################
2022-09-24 08:45:41,287 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2022-09-24 08:45:41,287 INFO    ############################################################
2022-09-24 08:45:41,288 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
[cmd] ./run.py exited 0
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

kellerza, 2022-09-25:
I have no idea if this will work, have not seen any document on how to wire up the RS485 on the 12kw unit. There might even be different registers (to measure power etc on different phases etc). Would be great if you can test and report back

lubster85, 2022-09-27:
Will certainly do so. I've ordered a splitter as I think that one "BMS" port is the one needed for communication.
Out of interest, I've seen elsewhere that when communicating over modbus, not to use Modbus SN id "00". Our inverter is configured with 00. What is the issue with using 0 as opposed to something like 1?

Yesterday I tried changing this through the Sunsynk app and shortly afterwards there was an error event with audible alert: F13: Grid_Mode_changed. I was a little taken back and changed it back to 0. Do you know if a change like this be a "real" error or is that normal behaviour for the inverter to log an event like this?

cods4, 2022-09-27:
> Will certainly do so. I've ordered a splitter as I think that one "BMS" port is the one needed for communication. Out of interest, I've seen elsewhere that when communicating over modbus, not to use [...]

As long as the the parallel box isn't checked (you only have the one inverter right?), Changing the Modbus SN ID to 01 wont cause any problems. The F13 error will clear after a few minutes and the inverter output will turn back on.

kellerza, 2022-09-27:
If you have multiple inverters installed, or parallel checked, it's highly unlikely that your Modbus SN is till 0!

Modbus address 0 is a broadcast address and you are not allowed to respond to any message you receive on 0. With older firmware versions the inverter always responded...

lubster85, 2022-09-27:
Thank you @kellerza and @cods4. Yes, only 1 x 3-phase inverter.
Will report back one I have some more info.

sebmueller, 2022-09-28:
I have the same problem with a Deye 12kw Inverter.
Solar Assistant reads it out, but this plugin with homeassistant doesnt read.

The only one is the Serial No.

Do you have an idea?

ronaldt80, 2022-09-28:
Planning to buy the same inverter (3fase/12kw) combined with batteries only (no solar) so very curious to hear about your progress.

Out of curiosity, with this add-on, is it possible to instruct the inverter (for instance to change discharge or charge from grid settings) or is the addon reporting state sensors only?

cods4, 2022-09-28:
> I have the same problem with a Deye 12kw Inverter. Solar Assistant reads it out, but this plugin with homeassistant doesnt read.
> 
> The only one is the Serial No.
> 
> Do you have an idea?

The fact that it's displaying the SN means it is communicating, so that's a good start.

Have you entered the inverter SN into the config? If so, try remove the quotation marks around the serial number.

sebmueller, 2022-09-29:
Hi,
thanks for the fast reply!

in the meanwhile, i figured out, what is the actual problem!

The Deye Sun-12k-SG04LP3 seems to "speak" another protocol.

Last night, I found another protocol description with different register addresses than the "other" one, you are using in your add on.

I tried a NodeRed Flow to read out the inverter, it worked!

I attach both documents, the old and the new one. 

[sunsynk_modbus - no chinese.docx](https://github.com/kellerza/sunsynk/files/9671512/sunsynk_modbus.-.no.chinese.docx)
[Modbus Deye amit posun.docx](https://github.com/kellerza/sunsynk/files/9671513/Modbus.Deye.amit.posun.docx)

found here, a german forum:
https://www.photovoltaikforum.com/thread/169125-answer-everything-about-deye-inverters/?pageNo=64

The Serial# registers are equal, also the Register: 02 - Communication Protocol Version.
When I read this, I get a 0x0104. What version do you get? 

Maybe, it is possible to implement both version of the protocol, switching after the read of the protocol version.

lubster85, 2022-09-29:
I'm not sure if the same port issue exists with the 3-phase Sunsynk and Deye, but I think the port could also be the issue here. I'm waiting on a delivery for a splitter and hope to test on the "BMS" port soon.

Interestingly I also managed to connect with the inverter using NodeRed (Using "Meter-485" port) and managed to pull some values  out of it --- however the values appeared not to change over time.

As soon as I have the splitter, my plan is to first test with this integration "out the box" and see what I get. Else I'll play around more with NodeRed.

kellerza, 2022-09-29:
@sebmueller thanks for the document. This document shows a new device type-which seems to be the 3ph inverter. So far we’ve tested the dev type 0x0300, while the 3ph is 0x0500.

The key would be to go through the new doc and map the register numbers that you are interested in. My guess is that the current implementation will either give you total(across phases) or only one of the phases, or worst case the single-phase registers will give you something entirely different.

Once we have the registers of the 3phase inverter it is possible to choose either based on the value of the device type

@sebmueller how did you connect to you inverter? (Might help @lubster85)

@ronaldt80 you can read and you can write RW type sensors.

[definitions.py](https://github.com/kellerza/sunsynk/blob/v0.2.4/sunsynk/definitions.py) will show you the registers and which are read &write — use the dev version of the addon for write

lubster85, 2022-09-29:
I've started having a look at the doc and I can confirm that I'm getting readings that appear to be correct. I'm still going through them to work out which is which.

ronaldt80, 2022-09-29:
Keen to hear about your experience @lubster85 and @sebmueller - my sunsynk 3ph 12kw has been ordered. I intend to switch to dynamic energy tariffs with the aim to buy low sell high :) - typically the pricing curve shows 2 peaks and 2 troughs so plenty of options to recover part of the investment. Key is however to also steer the export / import settings of the inverter - hope the r/w settings permit this

sebmueller, 2022-09-30:
@lubster85 you are welcome.
would you also share your nodered flow?

For Overall_State - you can use register 500.

[3ph-inverter.zip](https://github.com/kellerza/sunsynk/files/9681463/3ph-inverter.zip)

@kellerza is it possible to extent your implementation to read out also the 3ph inverter?

SimonF89, 2022-10-02:
@kellerza when do you think you will add the Deye 12kw? Depending on this, i would deside if i order the solar assistant or if i would use the home assistant :)

lubster85, 2022-10-03:
@sebmueller i will definitely share, not a problem. Just hoping to add some of the “3-phase” metrics first and will share. 

Out of interest - I used a splitter on the “BMS” port to split readings from the inverter (based on the solar-assistant site suggestion). I have a sneaky suspicion that this was not necessary in my case (I was just referencing the incorrect registers initially). 

Would be interesting to hear how others have connected.

jcwilson01, 2022-10-03:
@lubster85 If I am understanding it correctly, you only managed to start getting readings once you started using Node-Red? You didn't have any success using @kellerza add-on? I'm completely new to all of this and I'm also trying to get readings out of the 12kW Sunsynk without much success. I just assumed I was doing something wrong as I was getting the same error in the logs as you initially did, but I will try Node-Red and see what happens. Maybe I'm completely off-base here, but shouldn't we get something (even incorrect or 'zero' readings out of the add-on, even if the registers are different? Or did you start getting some sort of readings from the inverter when you swapped the Modbus SN to 01? I also managed to shut my inverter down by changing the SN to 01. Nearly had heart failure. But, after doing further reading, apparently you are supposed to isolate all loads from the inverter before changing anything on that settings page. Live and Learn.

sebmueller, 2022-10-04:
@lubster85 I just used the Modbus Port. But I also read, it depends on the firmware version, which port es usable. In the future, I want to connect my diy battery, but therefore I need a converter for the battery informations.

lubster85, 2022-10-04:
@jcwilson01, correct. This is because the integration developed by @kellerza was made-for and tested on the inverters listed here: https://github.com/kellerza/sunsynk#tested-inverters. Interestingly enough, when I changed the modbus serial id to '01', the integration picked up my inverter serial number as it started. 

I'm really no expert and I haven't looked under the hood of what this integration is doing, but I think the integration is looking for an expected response, not getting it, and safely stopping itself. If you look at the two docs that @sebmueller shared a few posts above, you'll see just how different the two are. I'm still going through it all and double-checking my initial "mappings". Once I'm done I'll share my flows (to be used at your own risk). :-)

lubster85, 2022-10-05:
Updated addresses for 3-phase inverter "monitoring" values.

Name | 3-Phase address
-- | --
Battery_Charge_Day | 514
Battery_Charge_Volts | 606
Battery_Discharge_Day | 515
Battery_Output_Current | 591
Battery_Output_Power | 590
Battery_SoC | 588
Battery_Volts | 587
DayActive PowerWh | 501
DayReactive_PowerWh | 502
Grid_Export_Day | 521
Grid_Grid_CT | 607
Grid_Grid_Frequency | 183
Grid_Import_day | 520
Grid_Inverter_Load | 608
Grid_Phase_Voltage_A | 598
Grid_Phase_Voltage_B | 599
Grid_Phase_Voltage_C | 600
Inverter_Phase_Voltage_A | 627
Inverter_Phase_Voltage_B | 628
Inverter_Phase_Voltage_C | 629
Inverter_total_power | 636
Run_State | 500
Solar_PV_Day | 529
Solar_PV1_Amps | 677
Solar_PV1_Voltage | 676
Solar_PV2_Amps | 679
Solar_PV2_Voltage | 678
Temperature Battery | 586
Temperature DC Transformer | 540
Temperature Environment | ?
Temperature Heat Sink | 541
Total_Load | 653
Total_Load_Phase_A | 650
Total_Load_Phase_B | 651
Total_Load_Phase_C | 652

Then these are some of the "settings" addresses I can find. Most of these are R/W addresses.
Interestingly, it looks like with the single-phase inverters, it is possible to set seperate time zones for "selling" and "charging" actions. I can't find this equivalent for the 3-phase model - it looks like there is only one set of times and then all other actions hinge on those times (feel free to double-check me on this).

Name | Address
-- | --
Read_Time_Z1 | 148
Read_Time_Z2 | 149
Read_Time_Z3 | 150
Read_Time_Z4 | 151
Read_Time_Z5 | 152
Read_Time_Z6 | 153
Read_SoC_Z1 | 166
Read_SoC_Z2 | 167
Read_SoC_Z3 | 168
Read_SoC_Z4 | 169
Read_SoC_Z5 | 170
Read_SoC_Z6 | 171
Read_GC_Z1 | 172
Read_GC_Z2 | 173
Read_GC_Z3 | 174
Read_GC_Z4 | 175
Read_GC_Z5 | 176
Read_GC_Z6 | 177
Read_PL | 141
Read_UseTimer | 146
Read_SolarExport | 145
Read_LoadLimit | 142
Read_BatShut | 115
Read_BatLow | 117
Read_BatRestart | 116
Read_BatMaxC | 108
Read_BatMaxDC | 109
Read_GridEn | 130
Read_GridC | 128
Read_ExportLimit | 143
Read_SM_Z1 | 148
Read_SM_Z2 | 149
Read_SM_Z3 | 150
Read_SM_Z4 | 151
Read_SM_Z5 | 152
Read_SM_Z6 | 153

ronaldt80, 2022-10-05:
The way i read the manual 148 - 177 should al be read/write:

 148-  153 accepts a time value, where 1745 would equate to quarter to 6.

154 - 159 should be the max watt power importing or exporting, 8000 being 8kw

160 - 165 is the voltage

166 - 171 is the target SoC level

172 - 177 is enable/disable of charge mode

Unfortunately I will only receive my deye in december.. so no possibility to test

lubster85, 2022-10-07:
@sebmueller see attached "my" NodeRed flow. I found this at [this forum post](https://powerforum.co.za/topic/8451-sunsynk-inverter-monitoring/) and adapted it to suit my needs. Much of what I have done is stripped some functionality I didn't need/understand and I also added some of the 3-phase metrics in (as referenced in the posts above). Still very much a work in progress so please treat it with caution. :-P
[flows (Sunsynk-3ph-MQTT).json.zip](https://github.com/kellerza/sunsynk/files/9737365/flows.Sunsynk-3ph-MQTT.json.zip)

@kellerza thank you for the quick replies and apologies - you'll see the references I've posted above are based on the NodeRed flow I found (and not in line with your [definitions.py](https://github.com/kellerza/sunsynk/blob/v0.2.4/sunsynk/definitions.py)) naming. Nonetheless I hope it proves helpful if/when you consider looking at this.

In the meantime, I thought I'd share some of the Home Assistant (apexcharts) charts I've put together for our setup :-) We've been away from home so the usage is lower than normal, but its been particularly interesting seeing the usage by phase. It looks like most of our plugs + lights are on Phase A, pool pump and some other things on B and the Stove/Oven on C.

[image]

sebmueller, 2022-10-10:
@lubster85 thanks for sharing your flows! :)
could you also share your HA charts?

@kellerza are there some news about integrating a second set of registers in your addon? 

kind regards

lubster85, 2022-10-10:
> @lubster85 thanks for sharing your flows! :) could you also share your HA charts?

No problem - you'll see there are a few custom cards and a theme that you'll need to download/change. I'm sure it's also not perfect, but its getting there :-)

```
views:
  - title: Inverter
    path: inverter
    theme: Mushroom
    badges: []
    cards:
      - type: vertical-stack
        cards:
          - type: custom:mushroom-title-card
            title: Energy Summary
          - square: false
            columns: 2
            type: grid
            cards:
              - type: custom:mushroom-entity-card
                name: PV
                entity: sensor.sunsynk_solar_total_power
                icon_color: '#FCCA1B'
                icon: mdi:solar-power-variant
              - type: custom:mushroom-entity-card
                name: Load
                entity: sensor.sunsynk_total_load
                icon_color: '#F78F24'
                fill_container: false
                icon: mdi:power-plug
              - type: custom:mushroom-entity-card
                name: Battery
                entity: sensor.sunsynk_battery_output_power
                icon_color: '#0B84B3'
                icon: mdi:battery-charging-100
              - type: custom:mushroom-entity-card
                name: Grid
                entity: sensor.sunsynk_grid_ct
                icon_color: '#EA285B'
                icon: mdi:transmission-tower-export
              - type: custom:mushroom-entity-card
                name: Battery
                entity: sensor.sunsynk_battery_soc
                icon_color: '#1D8D7D'
          - type: custom:apexcharts-card
            header:
              title: Overview
              show: false
              show_states: false
              colorize_states: true
            all_series_config:
              stroke_width: 2
              opacity: 0.3
              type: area
            color_list:
              - F78F24
              - 0B84B3
              - EA285B
              - FCCA1B
              - 1D8D7D
              - F39089
              - '274656'
              - 0ECAEE
              - 2BCDBC
            apex_config:
              chart:
                height: 350
                zoom:
                  type: x
                  enabled: true
[... 337 lines cut ...]
                opposite: true
                decimals: 0
                apex_config:
                  tickAmount: 4
                min: -2000
                max: 2000
            series:
              - entity: sensor.sunsynk_battery_output_power
                name: Battery output
                invert: true
                group_by:
                  func: avg
                  duration: 10min
                yaxis_id: second
                type: column
                opacity: 1
              - entity: sensor.sunsynk_battery_soc
                name: Battery %
                group_by:
                  func: avg
                  duration: 60min
                yaxis_id: first
                type: line
                opacity: 1
```

sebmueller, 2022-10-11:
@lubster85 
I played around with my inverter, wondering why grid_inverter_load is still 0.
I think, the register 619 is the correct one, reading the current grid load. could you have a look at your readings?

sebmueller, 2022-10-11:
@lubster85 
[InverterState.json.txt](https://github.com/kellerza/sunsynk/files/9754328/InverterState.json.txt)

I also figured out, that the Overall State decoding in you flow was buggy.
Please try this one

sebmueller, 2022-10-11:
@lubster85

thanks for sharing your dashboard, looks nice :)

[image]

sebmueller, 2022-10-12:
@lubster85 
I found m

Register 606, in the Flow it is used for "battery_charge_volts", this is in the "old / other" protocol.
in the new one, 606 is "C phase power on the inner side of the grid" but I am not sure, what the values mean.

lubster85, 2022-10-15:
> @lubster85 I played around with my inverter, wondering why grid_inverter_load is still 0. I think, the register 619 is the correct one, reading the current grid load. could you have a look at your readings?

Thank you, this looks right now. Mine was also pulling in 0

lubster85, 2022-10-15:
> @lubster85 I found m
> 
> Register 606, in the Flow it is used for "battery_charge_volts", this is in the "old / other" protocol. in the new one, 606 is "C phase power on the inner side of the grid" but I am not sure, what the values mean.

Thanks for that too - what about 210?

sebmueller, 2022-10-20:
@lubster85 : Reg 210, I get only '0'.

another question, was you able to write a register in the inverter? I tried, but I was not successfull.

lubster85, 2022-11-07:
> @lubster85 : Reg 210, I get only '0'.
> 
> another question, was you able to write a register in the inverter? I tried, but I was not successfull.

Apologies @sebmueller for not replying. I somehow missed this.
Initially I also thought I was not able to write back to the inverter (my initial test was to change something like the time by + 1). But it looks like values in 5 minute increments work OK.

It's also worth noting that it takes a few seconds to "read" the current values. But I've definitely been able to write back settings now. Below are my current times that I've set.

[image]

NismoBoy34, 2022-11-21:
Thanks for the digging work carried out so far . Have managed to get a copy working directly from node red first time attempt so my abilities are abit Limited.. soc was reporting the wrong value changed It to 588 I think and we got the right info . I'm in the process of trying to create some automation in nodered but fetching and using the soc value from home assistant which doesn't seem to work . Get a red square below the node with a statement not found.  I am using the entity id deye battery soc from the drop down list. Will keep tinkering but happy to contribute with more values and findings to get this project moving forward

NismoBoy34, 2022-11-21:
got it working just needed to add some form of delays, The next issue is debug in node red seems to show communication drop outs randomly must be a timing issue on the 485 
[image]

brendanvz, 2022-11-28:
I found the following when downloading the operation data for my plant from https://www.sunsynk.net/workdata/list
The excel column headers were suffixed with what looks like an address.
Hope it's useful.

| name | address |
| --- | --- |
| chargeVolt(V) | 210 |
| dischargeVolt(V) | 211 |
| chargeCurrentLimit(A) | 212 |
| dischargeCurrentLimit(A) | 213 |
| realtimeCap(%) | 214 |
| realtimeVolt(V) | 215 |
| realtimeCurrent(A) | 216 |
| realtimeTemp(℃) | 217 |
| runStatus(NA) | 500 |
| netoday(kWh) | 502 |
| htoday(S) | 503 |
| netotal(kWh) | 506-507 |
| batChargeToday(kWh) | 514 |
| batDischargeToday(kWh) | 515 |
| batChargeTotal(kWh) | 516-517 |
| batDischargeTotal(kWh) | 518-519 |
| gridBuyToday(kWh) | 520 |
| gridSellToday(kWh) | 521 |
| gridBuyTotal(kWh) | 522-523 |
| gridSellTotal(kWh) | 524-525 |
| dailyUsed(kWh) | 526 |
| totalUsed(kWh) | 527-528 |
| pvEtoday(kWh) | 529 |
| todayPv1(kWh) | 530 |
| todayPv2(kWh) | 531 |
| todayPv3(kWh) | 532 |
| todayPv4(kWh) | 533 |
| pvHistory(kWh) | 534-535 |
| genToday(kWh) | 536 |
| genTotal(kWh) | 537-538 |
| genDailyTime(h) | 539 |
| dcTemp(℃) | 540 |
| heatSinkTemp(℃) | 541 |
| acRelayStatus(NA) | 552 |
| batteryTemp(℃) | 586 |
| batteryVolt(V) | 587 |
| batteryEnergy(%) | 588 |
| batteryPower(W) | 590 |
| batteryCurrent(A) | 591 |
| correctedEnergy(Ah) | 592 |
| vac1(V) | 598 |
| vac2(V) | 599 |
| vac3(V) | 600 |
| aLineVolt(V) | 601 |
| bLineVolt(V) | 602 |
| cLineVolt(V) | 603 |
| aInsidePower(W) | 604 |
| bInsidePower(W) | 605 |
| cInsidePower(W) | 606 |
| gridInsideTotalPac(W) | 607 |
| gridInsideTotalSac(VA) | 608 |
| gridFac(Hz) | 609 |
| iac1(A) | 610 |
| iac2(A) | 611 |
| iac3(A) | 612 |
| aOutsideCurrent(A) | 613 |
| bOutsideCurrent(A) | 614 |
| cOutsideCurrent(A) | 615 |
| aOutsidePower(W) | 616 |
| bOutsidePower(W) | 617 |
| cOutsidePower(W) | 618 |
| gridOutsideTotalPac(W) | 619 |
| gridOutsideTotalSac(VA) | 620 |
| pf(NA) | 621 |
| aPower(W) | 622 |
| bPower(W) | 623 |
| cPower(W) | 624 |
| gridTotalPac(W) | 625 |
| invAVolt(V) | 627 |
| invBVolt(V) | 628 |
| invCVolt(V) | 629 |
| invACurrent(A) | 630 |
| invBCurrent(A) | 631 |
| invCCurrent(A) | 632 |
| invAPower(W) | 633 |
| invBPower(W) | 634 |
| invCPower(W) | 635 |
| invTotalPower(W) | 636 |
| invTotalSac(VA) | 637 |
| invFac(Hz) | 638 |
| upsLoadAPower(W) | 640 |
| upsLoadBPower(W) | 641 |
| upsLoadCPower(W) | 642 |
| upsLoadTotalPower(W) | 643 |
| loadAVolt(V) | 644 |
| loadBVolt(V) | 645 |
| loadCVolt(V) | 646 |
| loadAPower(W) | 650 |
| loadBPower(W) | 651 |
| loadCPower(W) | 652 |
| loadTotalPac(W) | 653 |
| loadTotalSac(VA) | 654 |
| genAVolt(V) | 661 |
| genBVolt(V) | 662 |
| genCVolt(V) | 663 |
| genAPower(W) | 664 |
| genBPower(W) | 665 |
| genCPower(W) | 666 |
| genTotalPower(W) | 667 |
| ppv1(W) | 672 |
| ppv2(W) | 673 |
| ppv3(W) | 674 |
| ppv4(W) | 675 |
| vpv1(V) | 676 |
| ipv1(A) | 677 |
| vpv2(V) | 678 |
| ipv2(A) | 679 |
| vpv3(V) | 680 |
| ipv3(A) | 681 |
| vpv4(V) | 682 |
| ipv4(A) | 683 |

lubster85, 2023-01-19:
@brendanvz this is incredibly helpful, thanks.

Does anyone know what the difference between "inside" and "outside" power is when looking at metrics like aOutsidePower (616) vs aInsidePower (604)?

TheKamakaZi, 2023-01-21:
@lubster85, From the looks of things in my testing Outside applies to exporting to grid, and Inside to importing from grid.

I _may_ be wrong. I'm still trying to figure out some oddness in my numbers when the batteries pulling or pushing high current.

NismoBoy34, 2023-01-23:
for the record ive managed to get my readings fairly accurate with the above flows and no time outs now to work out what the settings read and write are any pointers so i can start to contribute ?

kababoom, 2023-01-24:
Anyone working on an adjusted definitions.py for a 3phase model?

sebmueller, 2023-01-24:
@kababoom: in my ha-node-red the flow is working well. Whats not working in your node-red?

kababoom, 2023-01-24:
@sebmueller I did import on a new flow and got 'gate' and 'cycle' not found hence could not deploy and assumed it was non-ha version.

kellerza, 2023-01-24:
If there is someone willing to build a `definitions.py` / `definitions12k.py` for the 12kW we can easily integrate it,- I can add code to select between the two.

The first step would be to edits the definitions.py and to test it locally on their inverter - it should be easy to do with [copy2local.cmd](https://github.com/kellerza/sunsynk/blob/main/scripts/copy2local.cmd)

kababoom, 2023-01-24:
@kellerza I got the 8kW version SUN-8K-SG04LP3 but assume the registers to be the same as the 12kW. I'm, mostly interested in setting the 'time of use' schedule (6 lines with time, charge, power, voltage/percentage). So I don't have walk 'al the way' to the garage. Oh and also because the menu is weird, setup and operation are mixed, you need to go through 'dangerous' setup items to the operating settings.

So it's adjusting, run cmd (perhaps skip some lines), restart the add-on, check log?

kababoom, 2023-01-24:
@kellerza that's the 1phase version, again assuming ;-) 
Mine looks 100% like the picture from topic starter.

Will start with reading, does that also require a boomswitch?

kellerza, 2023-01-25:
To install an addon on you local hass server: copy all the relevant files to you shared /addon folder (use the samba addon to share this)

the copy2local.cmd script copies all the required files

the you can go to the addon store, refresh it and you should have a local sunsynk addon.
Once installed, you simply run copy2local and the Rebuils the addon

kababoom, 2023-01-25:
@kellerza Thank you for your help. 

Done exactly that including installing unix-utils for windows to support cp + sed the cmd needs but the sed -i is not working correctly. Tried different quotes got rid of an error message but dockerfile is not created.

Will look for another sed version or create the dockerfile otherwise.

GNU sed version 4.2.1
Copyright (C) 2009 Free Software Foundation, Inc.

kellerza, 2023-01-26:
Try msys2 for all these unix shell tools.

And you cannot go wrong with Vscode for editing.

duanemck, 2023-01-26:
WSL (Windows Subsystem for Linux) is also great for running unix shell tools on Windows.
https://learn.microsoft.com/en-us/windows/wsl/install

archi, 2023-01-27:
I used bendranz' register map to update some numbers in the `definitions.py`. Since my Deye 12k only has grid+modbus connected as of now, I can not really test this - but it's a first step. At least the numbers I saw are now more sane (read: `0`). Maybe someone with some PV and a battery connected to the inverter wants to check+improve this ;-)

https://github.com/archi/sunsynk/blob/deye12k/sunsynk/definitions.py

n.b.: I know you wanted to put this into a file of its own, but for testing this was the easiest way ;-)

kababoom, 2023-01-27:
@archi looks a lot like mine, I replaced mostly with same registers.
it's a first step but not perfect, need more time and some debug options would be nice.

archi, 2023-01-27:
@kababoom great to hear! is yours on github? Else, feel free to comment on my commit, or even open a PR to my fork.

Regarding debugging, the addon can be run locally:

1. I checked out the repo on my Arch Linux machine
2. symlinked the `sunsynk` folder into `hass-addon-sunsynk-dev`
3. ran `./run.py` in there and since that failed
4. installed missing python packages (some from the official Arch Linux repos, some using `pip` an non-root) until the program started, but complained about missing parameters.
5. Changed the example `config.yaml` (PORT, DEVICE, SUNSYNK_ID, SENSOR_PREFIX, MQTT_USERNAME)
6. Run `./run.py my.mqtt.server.tld topsecretpassword` and see what happens

The `run.py` will always expect a MQTT server and password on the command line when run that way, so that's why these are passed. It also sets debug to 1. You can change that in `run.py` at lines 226 to 228.

kababoom, 2023-01-27:
@archi some version is on git but not latest (not enough time, hope this weekend).

Thank you for describing these steps, do you use ip or plug the usb in your dev machine?

archi, 2023-01-27:
I use [this Modbus Gateway](https://eckstein-shop.de/WaveShare-Industrial-Serial-Server-RS485-to-RJ45-Ethernet-TCP-IP-to-Serial). But USB should work as well, though I didn't try that.

kababoom, 2023-01-28:
@archi Was able to debug with USB connected to my MBPro but have ordered the Wave tcp gateway anyway. Bit chilly in the garage...

Have added the system program settings and have tested the write functionality (simple time change), which seems to work just fine. I'm not sure about the 'Prog mode' options though, is that 1phase specific? or firmware specific, I'm still on C027 waiting for an update.
Mine just got the general 'Time of use' checkbox and:
Grid charge - Gen charge - Time from - Time to - Power - Voltage/Soc%

There are still some unknown registers and not yet translated ones, I have reused your comment: # 12k?? 

Todo: test and add the following as RW 
Export limit 143
Bat max charge 108
Bat max discharge 109
Use timer 141

Also noticed, although in several lists, anything in 200 range returns 0 from my inverter.

Here's a link to the definitions file, when I've sorted more I will send pull.
https://github.com/kababoom/sunsynk/blob/main/sunsynk/definitions.py

Here's a list of the sensors and the one skipped:

    - battery_soc
    - battery_voltage
    - battery_power
    - battery_current
    - inverter_power
    - inverter_l1_power
    - inverter_l2_power
    - inverter_l3_power
    - inverter_voltage
    - inverter_frequency

    - grid_frequency
    - grid_power
    - grid_l1_power
    - grid_l2_power
    - grid_l3_power
    - grid_frequency
    - grid_power
    - grid_current
    - grid_ct_power

    - gen_power
    - gen_l1_power
    - gen_l2_power
    - gen_l3_power

    # - rated_power
    - device_type
    # - fault
    # - overal_state
    # - sd_states
    - serial
    - dc_transformer_temperature
    - environment_temperature
    # - radiator_temperature
    # - grid_connected_status

    # - control_mode
    - grid_charge_battery_current
    - grid_charge_enabled
    # - battery_charging_voltage
    - bat1_soc
    # - bat1_cycle
    # - battery_equalization_voltage
    # - battery_absorption_voltage
    # - battery_float_voltage
    # - battery_shutdown_capacity
    # - battery_restart_capacity
    # - battery_low_capacity
    # - battery_shutdown_voltage
    # - battery_restart_voltage
    # - battery_low_voltage

    - priority_mode
    - load_limit

    - prog1_time
    - prog2_time
    - prog3_time
    - prog4_time
    - prog5_time
    - prog6_time

    - prog1_power
    - prog2_power
    - prog3_power
    - prog4_power
    - prog5_power
    - prog6_power

    - prog1_capacity
    - prog2_capacity
    - prog3_capacity
    - prog4_capacity
    - prog5_capacity
    - prog6_capacity

    - prog1_charge    
    - prog2_charge    
    - prog3_charge    
    - prog4_charge    
    - prog5_charge    
    - prog6_charge    

    # - prog1_mode
    # - prog2_mode
    # - prog3_mode
    # - prog4_mode
    # - prog5_mode
    # - prog6_mode

    - prog1_voltage
    - prog2_voltage
    - prog3_voltage
    - prog4_voltage
    - prog5_voltage
    - prog6_voltage

archi, 2023-01-28:
Hey, great :) I will screenshot my settings for the gateway - I found some of the stuff a bit counterintuitive.
Since I only have the 3 phase inverter, I don't know. Maybe kellerza has an idea?
I don't have a lot of time for testing right now (many more house improvements to be done, including getting the electrical work up to code), but once my battery arrives I'll try to see if can setup a small temporary solar array for offgrid testing. Though testing configuration parameters match the display might already be possible (I did not yet dare playing with that, read-only is safer and personally enough for me).

kababoom, 2023-01-28:
Hmm prog mode is the combination of gen charge - grid charge..

kellerza, 2023-01-31:
> Hmm prog mode is the combination of gen charge - grid charge..

Yes, but if you define your own, you can also separate the using a bitmask=xx on the sensor definition

sebmueller, 2023-01-31:
> 

> @sebmueller I did import on a new flow and got 'gate' and 'cycle' not found hence could not deploy and assumed it was non-ha version.

@kababoom : You have to install the add ons for gate and cycle in HA. 

[image]
[image]

archi, 2023-01-31:
Ah, @kababoom nice work on the PR :) Looking forward to test it.

Regarding the Waveshare: Keep in mind the non-PoE variant is lacking any isolation, while the PoE variant is isolated both on the Ethernet port and on RS485. The think the chance of the inverter unloading into the modbus is pretty slim, but in case of a lightning strike I didn't want to take any chances (plus, I have PoE anyway).

I can also report the configuration software runs well on my Linux machine (using `wine`). I did not test the automatic scan (the gateway has its own subnet and now strictly firewalled), but manually adding by IP worked well.

kababoom, 2023-01-31:
> > @sebmueller I did import on a new flow and got 'gate' and 'cycle' not found hence could not deploy and assumed it was non-ha version.
> 
> @kababoom : You have to install the add ons for gate and cycle in HA.
> 
> [image] [image]

@sebmueller Thank you, packages could not be found but got it working by adding the packages to the config of nodered. Looks good..

[15 more replies]

closed 2023-04-18 by kellerza