kellerza/sunsynk #64: could not patch umodbus: numpy not found

rcotgrov, 2022-09-26:
I see this message when attempting to use this add-on in HA

kellerza, 2022-09-26:
The addon does not use numpy, so simply ignore the warning.

Timeouts is mainly cabling& interference. 

Check the projects github readme for some pointers…. even I get them a couple of times per day

rcotgrov, 2022-09-26:
Thank you so much for the very swift response and clarification. My first mistake was to assume the Cat5e patch cable I adapted was wired to T568A and in fact this is not the case, it's T568B I think. 

Once I got the right cabling it started working. I am getting a Timeout condition every 1 or 2 mins but I'm upgrading the cable to a solid copper core and will use STP/SFTP type cabling instead, just in case of interference. 

My inverter manual says Pin1 and Pin 7 are RS485B and Pin 2 and Pin8 are RS485A, Ground being Pin 3 and 6. 

Is it important to cable up both wires in each case.... Could that help with the timeouts do you think?

Do I need to do anything to confirm my testing of my specific model - SUNSYNK-3.6K-SG01LP1

rcotgrov, 2022-09-26:
r.e Modbus Server ID, I have a single inverter and yet my installer has set the inverter as "Slave" - the parallel box is not checked. Could this be causing any issue?  The ModBus SN is 0 and that is what I have configured.

kellerza, 2022-09-26:
Server ID must not be 0. Try 1. I doubt that the master checkbox makes a difference here…

I would only wire up a single RS485 pair (and make sure its on a twisted pair)

rcotgrov, 2022-09-27:
I've changed the cable to a top quality SFTP (shielded) solid core cable and still the timeouts in the log appear approx every 1-2mins however the data does appear to be flowing nicely into MQTT and HA. 

What do you think the issue is?   The actual inverter taking longer than expected by the code or my USB-RS485 Wave adapter? (the one you recommended in your notes)

kellerza, 2022-09-30:
Do you still have the Solarman Dongle reading data? It might also cause contention to read information...

rcotgrov, 2022-09-30:
Sort of. They don’t come with Solarman now. I have the new Sunsynk Wi-Fi dongle and app- I think it’s some kind of OEM’ing of software. I have asked Sunsynk support if this could be intervening but no response yet. I will chase. I can try removing the dongle I guess and looking at the logs.

kellerza, 2022-09-30:
I would try removing it for some test period… it queries the same Inverter MCU, very likely it’s over a shared bus as well

e1ace, 2022-10-22:
@rcotgrov Were you able to improve on the amount of timeouts?

rcotgrov, 2022-10-22:
I actually have not tried removing the Wi-Fi dongle as yet. Other things in life have taken priority. Sent from my iPhone

kellerza, 2022-11-09:
Ok to close this? From the logs, the timeout every 10 seconds seems to be two different pollers competing for the RS485 bus (i.e. This Addon, every second and the Wifi dongle, every 10 seconds)

rcotgrov, 2022-11-09:
Yes, please close. I suspect you are right.

closed 2022-11-09 by kellerza