kellerza/sunsynk #65: RS232 instead of RS485?

RoganDawes, 2022-09-29:
I'm struggling to get my RS485 dongle to communicate with the inverter, and am contemplating using the RS232 port, by disconnecting the Sunsynk dongle.

https://github.com/kellerza/sunsynk/issues/18#issuecomment-1036315960 mentions doing something similar, and I just wanted to confirm whether the protocol in use between the dongle and the inverter is indeed modbus, and in particular, which variant of it?

I'm actually contemplating replacing the firmware on the dongle with something custom, possibly ESPHome based, since the dongle is simply an ESP32 with a RS232 interface to the inverter. So if it is as simple as constructing the appropriate ESPHome yaml for the modbus integration, that would be pretty sweet! Then it wouldn't be as big a deal to poll the inverter continuously over Wifi, it would be the dongle driving the polling, and simply reporting changes to HA as necessary, when the values actually change more than whatever threshold you choose.

kellerza, 2022-09-30:
The comment you referred to mentions disconnecting the Solarman dongle to reduce timeouts.

You should be able to run modbus over RS232, but it's not something I'm able to assist you with unfortunately.

On Powerforum someone did use ESPhome and a RS485 connected to report values to HomeAssistant directly - see [here](https://powerforum.co.za/topic/8646-my-sunsynk-8kw-data-collection-setup/?do=findComment&comment=124813)

RoganDawes, 2022-09-30:
Thanks for the response. I guess I misunderstood the post. Thanks anyway!

closed 2022-09-30 by RoganDawes

RoganDawes, 2023-01-02:
Just to close the loop on this:

The protocol spoken over the RS232 link to the Sunsynk dongle is indeed Modbus RTU, exactly the same as the RS485 link. This means that you can run mbusd exactly the same as you do for a RS485 dongle, and interrogate or control the inverter without any difference. For some details, please see:
https://powerforum.co.za/topic/12604-sunsynk-wifi-dongle-hacking/?do=findComment&comment=145476
and
https://powerforum.co.za/topic/12604-sunsynk-wifi-dongle-hacking/?do=findComment&comment=145505

This implies that you can either:
a) Use a simple RS232 dongle attached to the externally exposed dongle port to monitor and manage your inverter.
b) Reprogram the Sunsynk dongle using ESPHome and the StreamServer external integration, and access the inverter wirelessly.
c) Reprogram the Sunsynk dongle using ESPHome and the Modbus integration to expose the various registers as suitable values.

It's probably a good idea to take a copy of the original firmware first, though, to make sure that you can write it back if you need to in the future.

kellerza, 2023-01-02:
Nice, thanks for confirming