kellerza/sunsynk #67: Connect to Port by ID instead of Device Path

e1ace, 2022-11-02:
Right now it seems like the Port settings has to be in the Device Path format "/dev/ttyUSB0". 

Is it possible to support the device ID instead "/dev/serial/by-id/usb-FTDI_USB-RS485_Cable_FTZ29WM6-if00-port0"?

The device path seems to change frequently when you have multiple devices in Home Assistant OS.

kellerza, 2022-11-02:
If you insert a “port” string it ignores the selected device

[image]

kellerza, 2022-11-02:
[image]

Try serial:///dev/abcded in the port field

e1ace, 2022-11-03:
Thanks. I've moved to the dev branch and I can now see the settings above.

closed 2022-11-03 by e1ace