kellerza/sunsynk #68: Specified environment_temperature but only getting -100 reading

louispires, 2022-11-15:
### Issue related to the Python sunsynk library

### Describe the issue/bug

Specified environment_temperature but only getting -100 reading

I added the probe that came with my inverter to port 1 and 2 as this diagram:
[image]

### Expected behavior

Get an actual reading

### Logs (if applicable)

```log
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-11-15 15:21:17,672 INFO    Loading HASS OS configuration
2022-11-15 15:21:17,685 DEBUG   could not patch umodbus: numpy not found
2022-11-15 15:21:17,689 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import
2022-11-15 15:21:17,689 INFO    Filter *avg used for battery_temperature, battery_voltage, dc_transformer_temperature, environment_temperature, grid_voltage, inverter_voltage, radiator_temperature
2022-11-15 15:21:17,689 INFO    Filter *step used for battery_current, battery_power, essential_power, grid_ct_power, grid_current, grid_frequency, grid_power, inverter_frequency, inverter_power, non_essential_power, overall_state
2022-11-15 15:21:17,689 INFO    Filter *round_robin used for fault
2022-11-15 15:21:17,702 INFO    Reading startup sensors serial, rated_power
2022-11-15 15:21:17,813 INFO    ############################################################
2022-11-15 15:21:17,813 INFO                Inverter serial number 'xxxxxxxxxx'
2022-11-15 15:21:17,813 INFO    ############################################################
2022-11-15 15:21:17,814 INFO    MQTT: Connecting to mosquittomqtt@10.0.0.249:1883
2022-11-15 15:21:17,821 INFO    MQTT: Connection successful
```

louispires, 2022-11-15:
[image]

cods4, 2022-11-19:
I have the same behavior.

Instead check the value of battery_temperature. I expect this will be what you are looking for.

louispires, 2022-11-20:
> I have the same behavior.
> 
> Instead check the value of battery_temperature. I expect this will be what you are looking for.

No not actually.

Battery Temp is being supplied via CAN from the BMS
I am attempting to use the probe as a general Room Temperature monitor.

[image]

To be clear, the 3 temps were there before I connected the probe.

kellerza, 2022-12-01:
I do not have this probe connected and also get -100. 

Unfortunately, that is the "environment" register (# 95) in the Sunsynk documentation https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py#L139

There is one other register # 92 that I never included; it's called "**Inductor 1 temperature (Void)**"

louispires, 2022-12-05:
@kellerza I've had a look at the registers and #95 is meant to be Total_Gen_powerWhH

The only other temp sensor I am seeing is #319:
[image]

I am quickly going to build this locally and test it as I have a probe connected and should actually see a value...

louispires, 2022-12-05:
Doesn't work... Still getting negative 100

kellerza, 2022-12-30:
Probably not much we can do here, except if we get a firmware update (or another register number). Let's close this for now

closed 2022-12-30 by kellerza