kellerza/sunsynk #70: Sunsynk Data Reading issues

RowanZee18, 2022-11-18:
I am experiencing the same issues as described in this ticket:
https://github.com/kellerza/sunsynk/issues/33

I have tried removing the termination resistor however the issue still persists. Could the problem be due to trying to run both the Sunsynk dongle and this addon? 
I cannot get a clear answer on this

kellerza, 2022-11-19:
Do you mean you get timeouts?

I suspect they query the inverter on the same bus. Please remove the Sunsynk dongle and let me know if things improve.

louispires, 2022-11-30:
I am successfully running both devices on my Inverter

rzimanTJY, 2022-12-01:
Thanks, The modbus ID is set to 1 and I'm also using the new Sunsynk dongle. I'll try another RS485 adapter and see if that helps.

kellerza, 2022-12-01:
I have the [Wave dongle](https://www.robotics.org.za/W17286) but anything that **includes a GND** should be ok. The data line must be on a twisted pair (e.g. CAT5 cable). Only wire up 1 pair, some people suggest wiring up more than 1. I have about 15m of solid CAT5 on the RS485 bus and then I run mbusd as a gateway on an old Pi.

There are still timeouts, but typically <1 per hour 

```
2022-11-22 05:18:38,524 ERROR   Timeout reading: 
2022-11-22 09:20:11,089 ERROR   Timeout reading: 
2022-11-22 10:20:33,943 ERROR   Timeout reading: 
2022-11-22 12:21:19,449 ERROR   Timeout reading: 
2022-11-22 13:21:42,342 ERROR   Timeout reading: 
2022-11-22 22:25:09,935 ERROR   Timeout reading: 
2022-11-22 23:25:32,454 ERROR   Timeout reading: 
2022-11-23 03:27:06,330 ERROR   Timeout reading: 
2022-11-23 05:27:51,524 ERROR   Timeout reading: 
2022-11-23 07:28:37,387 ERROR   Timeout reading: 
2022-11-23 09:29:23,413 ERROR   Timeout reading: 
2022-11-23 10:29:45,963 ERROR   Timeout reading: 
2022-11-23 12:30:33,037 ERROR   Timeout reading: 
2022-11-23 14:56:39,003 ERROR   Timeout reading: 
2022-11-23 15:57:01,304 ERROR   Timeout reading: 
2022-11-23 20:58:57,376 ERROR   Timeout reading: 
2022-11-23 21:59:20,494 ERROR   Timeout reading: 
2022-11-23 22:59:43,846 ERROR   Timeout reading: 
2022-11-24 01:00:30,210 ERROR   Timeout reading: 
2022-11-24 04:01:38,685 ERROR   Timeout reading: 
2022-11-24 05:02:01,381 ERROR   Timeout reading: 
2022-11-24 11:04:17,943 ERROR   Timeout reading: 
2022-11-24 13:05:04,832 ERROR   Timeout reading: 
2022-11-24 14:05:28,047 ERROR   Timeout reading: 
2022-11-24 15:26:34,684 ERROR   Timeout reading: 
2022-11-24 16:06:13,952 ERROR   Timeout reading: 
2022-11-24 17:06:36,440 ERROR   Timeout reading: 
2022-11-24 19:07:20,473 ERROR   Timeout reading: 
2022-11-24 21:08:06,801 ERROR   Timeout reading: 
2022-11-25 01:09:36,443 ERROR   Timeout reading: 
2022-11-25 04:10:43,813 ERROR   Timeout reading: 
2022-11-25 06:11:27,371 ERROR   Timeout reading: 
2022-11-25 07:11:49,767 ERROR   Timeout reading: 
2022-11-25 08:12:11,931 ERROR   Timeout reading: 
2022-11-25 11:13:18,063 ERROR   Timeout reading: 
2022-11-25 16:15:11,399 ERROR   Timeout reading: 
2022-11-25 17:15:34,952 ERROR   Timeout reading: 
2022-11-25 18:15:57,613 ERROR   Timeout reading: 
2022-11-26 01:18:31,308 ERROR   Timeout reading: 
2022-11-26 03:19:14,845 ERROR   Timeout reading: 
2022-11-26 09:21:32,840 ERROR   Timeout reading: 
2022-11-26 12:22:41,407 ERROR   Timeout reading: 
2022-11-26 13:23:05,413 ERROR   Timeout reading: 
2022-11-26 14:23:27,354 ERROR   Timeout reading: 
2022-11-26 20:25:45,689 ERROR   Timeout reading: 
2022-11-26 21:26:09,102 ERROR   Timeout reading: 
2022-11-26 22:26:31,425 ERROR   Timeout reading: 
2022-11-27 01:27:40,420 ERROR   Timeout reading: 
2022-11-27 03:28:26,261 ERROR   Timeout reading: 
2022-11-27 06:29:35,313 ERROR   Timeout reading: 
2022-11-27 07:29:59,096 ERROR   Timeout reading: 
2022-11-27 21:30:26,378 ERROR   Timeout reading: 
2022-11-27 22:30:48,487 ERROR   Timeout reading: 
2022-11-28 00:31:34,364 ERROR   Timeout reading: 
2022-11-28 01:31:56,748 ERROR   Timeout reading: 
2022-11-28 03:32:43,826 ERROR   Timeout reading: 
2022-11-28 04:33:06,845 ERROR   Timeout reading: 
2022-11-28 07:34:15,777 ERROR   Timeout reading: 
2022-11-28 08:34:38,095 ERROR   Timeout reading: 
2022-11-28 09:35:01,728 ERROR   Timeout reading: 
2022-11-28 14:36:58,002 ERROR   Timeout reading: 
2022-11-28 15:37:20,943 ERROR   Timeout reading: 
2022-11-28 16:37:44,170 ERROR   Timeout reading: 
2022-11-28 18:38:31,729 ERROR   Timeout reading: 
2022-11-28 19:38:54,846 ERROR   Timeout reading: 
2022-11-29 02:41:35,715 ERROR   Timeout reading: 
2022-11-29 03:41:59,020 ERROR   Timeout reading: 
2022-11-29 05:42:44,673 ERROR   Timeout reading: 
2022-11-29 07:43:31,753 ERROR   Timeout reading: 
2022-11-29 13:46:07,089 ERROR   Timeout reading: 
2022-11-29 14:46:29,863 ERROR   Timeout reading: 
2022-11-29 19:48:24,259 ERROR   Timeout reading: 
2022-11-29 21:49:11,185 ERROR   Timeout reading: 
2022-11-29 22:49:33,476 ERROR   Timeout reading: 
2022-11-29 23:49:56,360 ERROR   Timeout reading: 
2022-11-30 01:50:42,954 ERROR   Timeout reading: 
2022-11-30 02:51:06,133 ERROR   Timeout reading: 
2022-11-30 04:51:51,847 ERROR   Timeout reading: 
2022-11-30 13:55:20,527 ERROR   Timeout reading: 
2022-11-30 15:56:06,174 ERROR   Timeout reading: 
2022-11-30 17:56:51,211 ERROR   Timeout reading: 
2022-11-30 20:58:01,208 ERROR   Timeout reading: 
2022-11-30 21:58:22,952 ERROR   Timeout reading: 
2022-11-30 22:58:45,123 ERROR   Timeout reading: 
2022-11-30 23:59:07,403 ERROR   Timeout reading: 
2022-12-01 00:59:31,100 ERROR   Timeout reading: 
2022-12-01 01:59:54,197 ERROR   Timeout reading: 
2022-12-01 09:41:46,457 ERROR   Timeout reading: 
```

rzimanTJY, 2022-12-01:
I am also using the wave. Perhaps I am misunderstanding the outputs. I assumed I would see some sort of reading confirmation in the output in a similar way the serial number is printed. Based on your dump I guess this is incorrect

kellerza, 2022-12-01:
Values are reported via MQTT to HASS. You should see that it connects to the Sunsynk (verifies & prints the serial), then it connects to MQTT and that's it from a logging perspective.

joes3029, 2022-12-02:
I battled with this as well, finally came right after much trial and error, also with intermittent readings of serial. My HA instance is a virtual machine in Proxmox, and it took a while to make sure that USB was properly passed through. I'm using the 1st RS485/ethernet converter listed in the Readme, and using dmesg | usb really helped make sure I was talking to the right port.
I had intermittent success with using pymodbus (and /dev/ttyUSB0) and now have:
   PORT: serial:///dev/ttyUSB0
   DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A10MLDQG-if00-port0
   DRIVER: umodbus
Which is working but not updating as quickly as I'd like (+/- 1min).
I did find that the newer Sunsynk dongle seemed to interfere with things, removing it helped.
I had to update my MODBUS_SERVER_ID on the inverter to 1 as it was set to 0 and that threw errors about an invalid ID - which made my (5.5 Sunsynk) inverter throw and f13 error and beep loudly for a minute or two before settling down.
It took quite a bit of trial and error but I am at least getting readings with only intermittent timeouts, so I'd recommend not to give up yet.

cakuhn, 2023-01-17:
> It is possible to please explain to me in a bit more detail, how I make the filters more aggressive?  Pv_power 1&2 is updating frequently however something like the battery SOC dont seem to update at all, if it does its hours later.  I would really appreciate if you can guide me on how to make the filters more aggressive please.  Thanks

kellerza, 2023-01-17:
You can have a look at the addon's docs [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-dev/DOCS.md#sensor-modifiers---minmaxnowaveragestep)

With step you can specify the size of the step, but seems like it's not documented.

some examples

```yaml
SENSORS:
  - battery_soc:step:2
  - some_power:step:10
  - other_power:now
```

I would expect battery_soc to take time to update. It typically changes slowly :wink:

juanka99, 2023-01-19:
Hi,
I'm trying to use [this](https://es.aliexpress.com/item/1005003019784618.html?spm=a2g0o.productlist.main.15.516c5014SF4RbH&algo_pvid=a4f9f735-e4d7-40e1-9f45-be78c7ede87e&algo_exp_id=a4f9f735-e4d7-40e1-9f45-be78c7ede87e-7&pdp_ext_f=%7B%22sku_id%22%3A%2212000029764481110%22%7D&pdp_npi=2%40dis%21EUR%218.14%217.25%21%21%21%21%21%402100b78b16741545542773760d06fd%2112000029764481110%21sea&curPageLogUid=HyJulHc6P5Fu) adapter, but have been an impossible mision... ¿Has somebody used this one?
Thanks

kellerza, 2023-01-19:
> Hi, I'm trying to use [...]

It looks exactly like the one I'm using - [here](https://www.robotics.org.za/W17286?search=wave%20rs485)  - which works fine🤷‍♂️

juanka99, 2023-01-19:
> It looks exactly like the one I'm using - [here](https://www.robotics.org.za/W17286?search=wave%20rs485) - which works fine🤷‍♂️

Do I have to do something special to make it work? 

I've tried everything I can think of and haven't been able to get it to work... :-(

kellerza, 2023-01-19:
I would suggest reading the main README in the repo, especially the "Fault finding" section.

And then open your own issue and try to include more details about your setup. Might be everything for you, but there is no context to help you unfortunately. It does seem to read something; you might even have something else on the RS485 bus

cakuhn, 2023-01-24:
> You can have a look at the addon's docs [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-dev/DOCS.md#sensor-modifiers---minmaxnowaveragestep)
>
> With step you can specify the [...]

Thanks

cakuhn, 2023-01-24:
Any ideas on the following reality I am experiencing:

When I start the App and look at the log I get this message, basically no response from the Inverter / USB port :**

_______________________________________________________________________________________________________________________
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-01-24 12:52:40,566 INFO    Loading HASS OS configuration
2023-01-24 12:52:40,585 DEBUG   could not patch umodbus: numpy not found
2023-01-24 12:52:40,595 INFO    ############################################################
2023-01-24 12:52:40,595 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2023-01-24 12:52:40,596 INFO    ############################################################
2023-01-24 12:52:40,597 INFO    Added hidden sensor battery_low_capacity as other sensors depend on it
2023-01-24 12:52:40,598 INFO    Added hidden sensor battery_low_voltage as other sensors depend on it
2023-01-24 12:52:40,599 INFO    Added hidden sensor battery_float_voltage as other sensors depend on it
2023-01-24 12:52:40,599 INFO    Filter now used for aux_power, battery_power, essential_power, grid_connected_status, grid_ct_power, grid_power, inverter_power, load_power, non_essential_power, overall_state, pv1_power, pv2_power, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_energy, total_pv_power
2023-01-24 12:52:40,600 INFO    Filter step:2 used for battery_soc
2023-01-24 12:52:40,600 INFO    Filter *round_robin used for prog1_capacity, prog1_charge, prog1_power, prog1_time, prog1_voltage, prog2_capacity, prog2_charge, prog2_power, prog2_time, prog2_voltage, prog3_capacity, prog3_charge, prog3_power, prog3_time, prog3_voltage, prog4_capacity, prog4_charge, prog4_power, prog4_time, prog4_voltage, prog5_capacity, prog5_charge, prog5_power, prog5_time, prog5_voltage, prog6_capacity, prog6_charge, prog6_power, prog6_time, prog6_voltage, prog6_voltage, prog6_voltage, prog6_voltage
2023-01-24 12:52:40,615 INFO    Reading startup sensors battery_float_voltage, prog6_time, battery_low_capacity, battery_low_voltage, prog4_time, prog1_time, prog3_time, rated_power, serial, prog2_time, prog5_time
2023-01-24 12:52:50,629 ERROR   Timeout reading: 
2023-01-24 12:52:50,631 INFO    ############################################################
2023-01-24 12:52:50,632 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2023-01-24 12:52:50,633 INFO    ############################################################
2023-01-24 12:52:50,634 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
_____________________________________________________________________________________________________________________

It then restart, or I manually stop and start and then it connects, see below:

_______________________________________________________________________________________________________________________
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-01-24 12:54:21,206 INFO    Loading HASS OS configuration
2023-01-24 12:54:21,226 DEBUG   could not patch umodbus: numpy not found
2023-01-24 12:54:21,235 INFO    ############################################################
2023-01-24 12:54:21,235 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2023-01-24 12:54:21,236 INFO    ############################################################
2023-01-24 12:54:21,237 INFO    Added hidden sensor battery_low_capacity as other sensors depend on it
2023-01-24 12:54:21,238 INFO    Added hidden sensor battery_low_voltage as other sensors depend on it
2023-01-24 12:54:21,238 INFO    Added hidden sensor battery_float_voltage as other sensors depend on it
2023-01-24 12:54:21,239 INFO    Filter now used for aux_power, battery_power, essential_power, grid_connected_status, grid_ct_power, grid_power, inverter_power, load_power, non_essential_power, overall_state, pv1_power, pv2_power, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_energy, total_pv_power
2023-01-24 12:54:21,239 INFO    Filter step:2 used for battery_soc
2023-01-24 12:54:21,239 INFO    Filter *round_robin used for prog1_capacity, prog1_charge, prog1_power, prog1_time, prog1_voltage, prog2_capacity, prog2_charge, prog2_power, prog2_time, prog2_voltage, prog3_capacity, prog3_charge, prog3_power, prog3_time, prog3_voltage, prog4_capacity, prog4_charge, prog4_power, prog4_time, prog4_voltage, prog5_capacity, prog5_charge, prog5_power, prog5_time, prog5_voltage, prog6_capacity, prog6_charge, prog6_power, prog6_time, prog6_voltage, prog6_voltage, prog6_voltage, prog6_voltage
[... cut ...]
2023-01-24 12:58:34,638 ERROR   Timeout reading: 
2023-01-24 12:58:44,644 ERROR   Timeout reading: 
2023-01-24 12:58:54,658 ERROR   Timeout reading: 
2023-01-24 12:59:04,667 ERROR   Timeout reading: 
2023-01-24 12:59:14,674 ERROR   Timeout reading: 

Can it be a hardware issue perhaps?  I am running a small virtual machine with 2G Ram and x2 CPU's allocated.
The funny part is if I change inverter timer settings, the updates almost instantly happens on the inverter side.

Any insights, views or advice will be appreciated

kellerza, 2023-01-24:
@cakuhn you could try different filters - 
```
2023-01-24 12:52:40,600 INFO Filter step:2 used for battery_soc
```
Maybe try `battery_soc:step:1` ? Or the default? This is mine on default

[image]

Interesting to see you have a timeout every 10 seconds. Do you have the solarman dongle connected? They probably compete for the bus while reading information. The addon reads every second, but Solaman much less.

Can you post you full config? (enclose it in three backticks ``) I would also expect settings to use the `round_robin` filter and that won't read so aggressively as the `mean` filter.

cakuhn, 2023-01-24:
> 

Sorry, not sure how to posit the full config :-). can you guide me please, still learning :-)

cakuhn, 2023-01-25:
> > 
> 
> Sorry, not sure how to posit the full config :-). can you guide me please, still learning :-)
> 
> I have a Sunsync Dongle connected not solarman, its updates every 5 minutes
> 
> Is it this part of the config you are looking for?
> 
> ```
> PORT: ""
> DEVICE: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0
> DRIVER: umodbus
> SUNSYNK_ID: "xxxxxxx"
> SENSOR_PREFIX: ""
> SENSORS:
>   - total_pv_energy:now
>   - total_battery_charge:now
>   - total_battery_discharge:now
>   - total_grid_export:now
>   - total_grid_import:now
>   - battery_soc:now
>   - essential_power:now
>   - non_essential_power:now
>   - grid_ct_power:now
>   - battery_power:now
>   - inverter_power:now
>   - pv1_power:now
>   - pv2_power:now
>   - aux_power:now
>   - grid_connected_status:now
>   - load_power:now
>   - grid_power:now
>   - overall_state:now
>   - total_pv_power:now
>   - prog1_time
>   - prog1_power
>   - prog1_capacity
>   - prog1_charge
>   - prog1_voltage
>   - prog2_time
>   - prog2_power
>   - prog2_capacity
>   - prog2_charge
>   - prog2_voltage
>   - prog3_time
>   - prog3_power
>   - prog3_capacity
>   - prog3_charge
>   - prog3_voltage
>   - prog4_time
>   - prog4_power
>   - prog4_capacity
>   - prog4_charge
>   - prog4_voltage
>   - prog5_time
>   - prog5_power
>   - prog5_capacity
>   - prog5_charge
>   - prog5_voltage
>   - prog6_time
>   - prog6_power
>   - prog6_capacity
>   - prog6_charge
>   - prog6_voltage
> READ_SENSORS_BATCH_SIZE: 60
> NUMBER_ENTITY_MODE: auto
> PROFILES: []
> MODBUS_SERVER_ID: 1
> MQTT_HOST: core-mosquitto
> MQTT_PORT: 1883
> MQTT_USERNAME: xx
> MQTT_PASSWORD: xx
> DEBUG: 0
> ```

Ok, So I rewired the USB to RS485 converter, made the exposed section outside of the Cat6 cable cover shorter, twisted the remaining exposed section as much as I could.  I also made sure my host machine have more ram allocated that the virtual machine and allocated 2CPU's.  This seems to have done the trick for me.  The App log immediately picked up the USB device (Unlike previously where I had to stop start restart etc).  The below graph confirm that the updates now happend much more frequently.  I however still get timeouts every 10 seconds but I am going to ignore that for now

[image]

duanemck, 2023-01-25:
I found that I was having a lot of timeouts until I enabled the mbusd addon and pointed the Sunsynk addon to the TCP port instead, seems to be much more reliable.

kellerza, 2023-01-25:
> I found that I was having a lot of timeouts until I enabled the mbusd addon and pointed the Sunsynk addon to the TCP port instead, seems to be much more reliable.

While I'm not sure what retry logic is being used in mbusd, this is also what I've found: umodbus driver --> mbusd --> usb

cakuhn, 2023-01-25:
> I found that I was having a lot of timeouts until I enabled the mbusd addon and pointed the Sunsynk addon to the TCP port instead, seems to be much more reliable.

Do I need another device to enable this route?  Seems like an additiona RPI is needed.  I run a virtual machine.

duanemck, 2023-01-25:
Nope, it's another add-on in the HA store. They can both run on the same machine. 
[image]

kellerza, 2023-01-25:
Even the same rpi can work, you just need IP connectivity to mbusd and Docker provides that (similar to connecting to mqtt)

I find using a fixed IP for my supervisor works well. But you can also resolve the addon name, like the core-mosquitto to connect to mqtt, then resolves to docker internally, but the actual IP works just as well

kellerza, 2023-01-31:
I’ve added a section covering timeouts to the readme. The next release of the addon will also have a timeout counter sensor in Home Assistant to monitor these

closed 2023-01-31 by kellerza

RowanZee18, 2023-05-01:
I haven't had time to revisit this integration for a while. I have just downloaded the new multi addon and got it working immediately. I have not changed a thing since my original comment, Sunsynk dongle still connected without issues.