kellerza/sunsynk #71: Any idea what the unlisted inverter faults mean

cods4, 2022-11-19:
My inverters are always showing some sort of fault on the fault sensor, but none of them are listed on the manual or anywhere else I can find.

Surely they have some meaning, does anyone know what they are?
The ones I have are F65 through to F73

louispires, 2022-11-30:
Also wondered the same thing.

Mine always shows these 3: 
[image]

I have in the past seen an error on my Inverter, BMS Failure but the HA integration never changed...

kellerza, 2022-12-01:
No idea to be honest, I really don't use this sensor, my sensor shows: F66, F67, F68, F70, F71. 
Everything seems ok on the inverter though.

I've never tried to test the [FaultSensor](https://github.com/kellerza/sunsynk/blob/main/sunsynk/sensor.py#L386) so no idea if it even works, it's simply my interpretation of the Sunsynk Modbus document. Open for PRs/suggestions

kellerza, 2023-04-21:
It seems this was solved by removing register 107 from the sensor.

Changes might only be in the multi version

kellerza, 2023-04-21:
Please check and let me know if still an issue?

cods4, 2023-04-21:
Yeh I noticed they were gone, which I guess solves it.

Thanks Johann

closed 2023-04-21 by kellerza