kellerza/sunsynk #72: System mode definitions

cc120689, 2022-12-02:
Hello Kellerza, first of all thank you very much for your fantastic addon.
I have a Turbo Energy sg03 5kw inverter.
My inverter in the ProgX_charge option accepts values "4", "8" and "16" that correspond to the System Work Mode options: "General Mode GM", "Backup Utility BU" and "Charge Mode CH".
With the last update of your addon, when selecting the different options of the ProgX_Charge (Allow Grid and Gen, No Grid No Gen etc.) it injects the inverter with values that do not work on my inverter: 6, 7...
Could you modify the addon so that it could select in ProgX_charge: "GM" (value 4) "BU" (value 8) and "CH" value 16?
Thank you very much.

kellerza, 2022-12-02:
Can look into it, can you define all the acronyms? 😁

system profiles still allow adding your own values today

cc120689, 2022-12-02:
Can look into it, can you define all the acronyms? 😁

General Mode: The system tries to consume as little as possible from the network until the battery reaches the selected level, at which time it leaves the battery in standby and consumes everything from the network
Backup Mode: The system tries to conserve the battery for the future by spending from the network up to the selected power level, for example if we have put 2500w, it will consume 2500W from the network and if we go from there it will take it from the battery as long as the level of this is above the SOC selected in that program
Charge Mode: Charge mode without exceeding the selected network power level

system profiles still allow adding your own values today.

Yes I still do but it would be great to be able to do it from a select box without having to connect to the server and edit the file manually.
It would only be necessary to inject the values 4, 8 or 16 in the same registers as the file created by system mode.
Thanks again in advance!!

cc120689, 2022-12-02:
Or maybe you can tell me which file must I edit to change this values

Salor10, 2022-12-27:
Hi @kellerza thanks for the addons, very hard work. I have a Turbo Energy like @cc120689, i'm interesting too in that inject values.

Ivan-L, 2022-12-30:
Hi @Salor10 and @cc120689, the values are defined [here](https://github.com/kellerza/sunsynk/blob/0acbd110d107826aab18eb1d603705c7a071bfc6/sunsynk/definitions.py#L230-L235).
```python
PROG_CHARGE_OPTIONS = {
    4: "No Grid or Gen",
    5: "Allow Grid",
    6: "Allow Gen",
    7: "Allow Grid & Gen",
}
```
The ProgX..Charge registers are not documented very well in the protocol document, so it may well be that there are extra bits set that we do not know about. Can you provide an exhaustive list of values that the registers can be set to by modifying them on the inverter and reporting what the values are logged as in the logs after increasing the log verbosity level? That would help to figure out what bits are being set by the inverter.

cc120689, 2022-12-30:
I don't know if I have understood you very well, I will tell you what I have done.
When I save the configuration of my inverter in "/share/hass-addon-sunsynk/system-mode......" using the selector "system mode: update" the file saves me a 4 if I have selected in the inverter "general mode", an 8 if I have selected "backup mode" and a 16 if I have selected "charge mode".
I tried cloning the repository, then modifying the definitions.py file:
PROG_CHARGE_OPTIONS = {
    4: "General Mode",
    8: "Backup Mode",
    16: "Charge Mode",
}

but once installed and configured from local the Home assistant UI selector continues showing me the original values:

PROG_CHARGE_OPTIONS = {
    4: "No Grid or Gen",
    5: "Allow Grid",
    6: "Allow Gen",
    7: "Allow Grid & Gen",
}

Ivan-L, 2022-12-30:
@cc120689 thanks, looking at the profile files saved on disk is another way to see what the inverter sets the register values to.

If you are running the add-on from local, there are a few extra steps to take if you modify anything in the `sunsynk` folder of your local copy of the repo. The reason is because any changes you make inside the `addons/local/sunsynk/sunsynk` folder will not get picked up by the add-on because it references the published `sunsynk` package. To get the local add-on (whether the `-dev` one or not) to use a customized local `sunsynk` module, you need to:

1. Create a subfolder called `sunsynk` within the addon folder, so either within `hass-addon-sunsynk` or `hass-addon-sunsynk-dev`
2. Copy `setup.cfg` and `setup.py` from the root of the repo into that subfolder
3. Copy your (customized) `sunsynk` folder from the root of the repo into the `sunsynk` subfolder within the addon folder, so you will basically copy `./sunsynk` into `./hass-addon-sunsynk-dev/sunsynk`, ending up with `./hass-addon-sunsynk-dev/sunsynk/sunsynk`
4. Update the `Dockerfile` within the addon folder by removing the `sunsynk[pymodbus,umodbus]=0.x.x` bit from line 11 and uncomment line 18. So line 11 should read something like `RUN pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1` and line 18 should read `RUN pip3 install -e sunsynk[pymodbus,umodbus]` (without the `# `)
5. Remember to update the local add-on version each time there is a change.

The above is automated by `scripts/copy2local.cmd` however I find the above manual steps work for me on a Mac.

Ivan-L, 2022-12-30:
Now for the ProgN..Charge register values discussion, this is the info I have thus far.

Based on my observations, the following works on my Deye 5kW inverter and I assume on the other Deye and Synsynk inverters as well.
```python
4: "No Grid or Gen",
5: "Allow Grid",
6: "Allow Gen",
7: "Allow Grid & Gen",
```

According to the modbus protocol document however, this is what is stated:
> Bit0 表示电网充电使能
> Bit1 表示发电机充电使能
> 0 disable 1 enable
> Bit2 GM模式
> Bit3 BU模式
> Bit4 CH模式

Doing a quick Google translate yields this:
> Bit0 indicates grid charging enable
> Bit1 indicates generator charging enable
> 0 disable 1 enable
> Bit2 GM mode
> Bit3 BU mode
> Bit4 CH mode

I am still trying to wrap my head around the modbus doc on the one hand and the values which work for my Deye inverter on the other.

Ivan-L, 2022-12-30:
Ok it is actually obvious now.

When bit 4 is set (GM mode), then the values which work for my Deye inverter make sense:

- 00100 in binary yields 4 in decimal, so no grid or gen with GM mode set
- 10100 in binary yields 5 in decimal, so grid is set, gen is not set, GM mode is set
- 01100 in binary yields 6 in decimal, so grid is not set, gen is set, GM mode is set
- 11100 in binary yields 7 in decimal, so grid is set, gen is set, GM mode is set

My only worry is that now this register is used to control 3 things and currently we do not have a way to split a register into 3 sensors:

- Whether grid charging is enabled or not
- Whether generator charging is enabled or not
- Which mode (GM, BU or CH) is to be used

@kellerza any ideas? Does this now warrant adding the ability for multiple sensors to work on one register?

cc120689, 2022-12-30:
Thanks a lot!!! It works perfectly now, my inverter is a turbo energy (from Spain).
Now I can select the different modes directly in each time slot.
Thanks again!

PROG_CHARGE_OPTIONS = {
4: "General Mode",
8: "Backup Mode",
16: "Charge Mode",
}

kellerza, 2022-12-31:
Seems like it should be enough if we can simply override/redefine some options.

For example

```yaml
REDEFINE:
  PROG_CHARGE_OPTIONS:
    4: General Mode
    8: Backup Mode
    16: Charge Mode
```

Then we could also use this in the future to redefine other registers/attributes of sensors in the definition file (testing purposes, adding the 12k inverter etc)

Ivan-L, 2023-01-10:
I don’t think we’d want to override the current values though. Currently the values of 4,5,6 and 7 try to deal with grid charge and grid gen only but inadvertently have the GM bit set. If we had two other sensors which separately deal with bits 1 and 2, and do not touch the other bits, those sensors should work for everyone. Then a third sensor can perhaps deal with GM, BU and CH.

cc120689, 2023-01-11:
My knowledge up to that point is very limited, I can only say that the inverter is working correctly with the new options, just as it would if I modified the options on the inverter screen.

Ivan-L, 2023-01-11:
@cc120689 could you maybe confirm whether for each program you can, in addition to setting the GM BU or CH mode, you can also set whether grid charge and generator charge are enabled or not? What are the values then if you set grid charge and gen charge to on?

closed 2023-01-12 by kellerza

cc120689, 2023-01-12:
ok I'll try when I have a while, thanks!

cc120689, 2023-01-13:
hello again those of us who want to use the three new modes should install the dev version or the normal version also has them?  Thank you

kellerza, 2023-01-13:
For now, the dev version only.

Then you can use the sensors:
```
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - prog1_mode
  - prog2_mode
  - prog3_mode
  - prog4_mode
  - prog5_mode
  - prog6_mode
```

Charge gives you the option to select Grid/Gen or combination. Same as the previous version

Mode gives you the option to switch between General/Backup/Charge Mode - or No Mode.

To move between normal & dev : 
- Copy your config and paste it in the dev version, in YAML mode! (in the config tab on the top right there are three dots with a menu)
- Stop the normal version (only one instance can access a USB port)
- Start the dev version

Ivan-L, 2023-01-13:
@kellerza you’re a legend 🎉

Salor10, 2023-01-13:
Great legend.

urbanbur, 2024-12-16:
This was added for single phase inverers. From MODBUS specifications same modes should be supported also for 3phase inverters (register addr. 172-177). It would be nice to update these specifications accordingly. BTW is it possible to overload the original specification by using mysensors.py?
In addition, there should also be customized registers from 1000 onwards enabling weekday-weekend diferentiation in scheduling modes.