kellerza/sunsynk #73: DEVICE isn't documented

reedy, 2022-12-02:
https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk/DOCS.md doesn't describe at all what `DEVICE` is (other than mentioning it in the bullet point), while giving docs that show the YAML.

kellerza, 2022-12-03:
DEVICE is a selectable USB device in the HomeAssistant UI.

PORT should be an empty string when you want to use the selected device, otherwise PORT is preferred

reedy, 2022-12-03:
But DEVICE seemingly can't be empty (which is more likely to happen if just setting via yaml), when it should be able to be when you're using one of the "proxies" or basically the RS485 adapter plugged into the device

kellerza, 2022-12-03:
Just select any DEVICE, PORT is preferred if present.

closed 2022-12-03 by kellerza

reedy, 2022-12-03:
Yeah, I know.

But it should be at least documented to say this..