kellerza/sunsynk #74: Connect via Node RED but not via HA Add-On

Zodiac-69, 2022-12-10:
I can connect to my Sunsynk 8kW inverter via Node RED in HA, but the Add-On show no response from Modbus Interface.
I am using this [USB-2-RS485 Converter](https://www.robotics.org.za/W17286?search=rs485&page=2)

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-12-10 19:25:20,552 INFO    Loading HASS OS configuration
2022-12-10 19:25:20,574 DEBUG   could not patch umodbus: numpy not found
2022-12-10 19:25:20,582 INFO    ############################################################
2022-12-10 19:25:20,582 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2022-12-10 19:25:20,583 INFO    ############################################################
2022-12-10 19:25:20,584 DEBUG   last unit:kWh, id:total_pv_power
2022-12-10 19:25:20,585 DEBUG   last unit:kWh, id:total_battery_charge
2022-12-10 19:25:20,586 DEBUG   last unit:kWh, id:total_battery_discharge
2022-12-10 19:25:20,586 DEBUG   last unit:kWh, id:total_grid_export
2022-12-10 19:25:20,587 DEBUG   last unit:kWh, id:total_grid_import
2022-12-10 19:25:20,588 DEBUG   last unit:%, id:battery_soc
2022-12-10 19:25:20,589 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_power
2022-12-10 19:25:20,591 DEBUG   Using selector: EpollSelector
2022-12-10 19:25:20,614 INFO    Reading startup sensors rated_power, serial
2022-12-10 19:25:20,696 ERROR   Read Error: (1,3,5) 50
2022-12-10 19:25:20,696 INFO    ############################################################
2022-12-10 19:25:20,696 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2022-12-10 19:25:20,696 INFO    ############################################################
2022-12-10 19:25:20,696 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.

Config is as follow:

[image]

I presume i am doing something stupid here...
Any idea what i am doing wrong?

Zodiac-69, 2022-12-10:
Seems that if you have Modbus in Node RED, HA Add-On does not work

closed 2022-12-10 by Zodiac-69

kellerza, 2022-12-11:
Correct, only ONE at a time