kellerza/sunsynk #75: Essential Power reading issue

lispwarez, 2022-12-13:
So not really a bug but an issue depending on how you use your AUX port

I use it as a generator input and when the generator is running the Essential Power goes into the negative
If it is being used as a non-essential load then the readon would be correct

Graph from yesterdays loadshedding when the generator kicked in
https://ibb.co/jvqgjyw

kellerza, 2022-12-29:
Today essential power is defined as: 

```
 essential_power = inverter_power + grid_ld_power - aux_power
```

from the definitions file
```python
 Sensor(175, "Inverter power", WATT, -1),
 Sensor(167, "Grid LD power", WATT, -1),
 Sensor(166, "AUX power", WATT, -1),
 MathSensor((175, 167, 166), "Essential power", WATT, factors=(1, 1, -1)),
```

maybe we could just change the definition to?
```
 MathSensor((175, 167, 166), "Essential power", WATT, factors=(1, 1, 1)),
```

so are you saying that `aux_power` should only be part of this equation when it's negative?

Ivan-L, 2022-12-30:
For what it's worth, my `aux_power` and `essential_power` sensors work correctly, though I am using the aux port as an output, not as a generator input.

kellerza, 2022-12-30:
Thanks @Ivan-L - I'm not using AUX for anything, so the final fix will probably need to check the direction/sign of the AUX power, or use the absolute value of AUX

kellerza, 2023-01-19:
@lispwarez and @Ivan-L can you both try the `essential_abs_power` sensor? For now it's a test sensor only in the dev version

It simply reports an absolute value of the current `essential_power` sensor and will hopefully address the graph above.

There is a second test sensor as well, called `essential_l2_power` which includes an l2 sensor. Not exactly sure what is being added, but a user on powerforum reported he needed to add the l2_power

lispwarez, 2023-01-24:
I quickly loaded up the dev version now during loadshedding to capture the graph. All is looking correct. Both _abs_ and _l2_ readings seem to be the same on my side

[image]

kellerza, 2023-01-24:
Great, I'm considering making `essential_l2_power` the only one as it seems to work in my case as well

kellerza, 2023-01-24:
Thanks again for the read/write contributions @Ivan-L !! Wish you all the best and happiness in your new home.

An inverter is probably not the first thing you **need** in your new country, unlike here in sunny South Africa 😉

kellerza, 2023-03-12:
Added it in https://github.com/kellerza/sunsynk/commit/89df18f459ba1a0b6f9c48ca51eae0a1ce4a9510

Should be released in the new `multi` inverter addon, but no exact timeline when that will be completed

closed 2023-03-12 by kellerza