kellerza/sunsynk #77: Problem reading progx_charge sensor

jabezz, 2022-12-15:
### You Inverter

Details about your Inverter: Sunsynk 8kw running SW ver (M 6.0.1.4 / S  1.7.2.4 / C E.4.2.4)

### Describe the issue/bug and what you expect

I've configured the 'system_mode' profile in the Sunsynk Inverter Add-on (dev) on Home Assistant.
Most everything seems to work nicely. But for some reason the progX_charge sensors are coming up with unknown value.
I suspect that it will also have issues trying to write settings (I have not tried this).

See below screenshots where the ProgX Charge sensors have unknown/no values
[https://tinypic.host/i/RzRuA](url)
[https://tinypic.host/i/RzT9i](url)

Am I perhaps specifying the sensors wrong? They look to be correct according to definitions.py

Is there perhaps anything I can do to make them work? Or should I rather remove them for now?

### My Config

```
PORT: ""
DEVICE: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
DRIVER: umodbus
SUNSYNK_ID: "xxxxxxxxx"
SENSOR_PREFIX: ss
SENSORS:
  - essential_power
  - non_essential_power
  - aux_power
  - battery_soc
  - battery_power
  - battery_voltage
  - battery_current
  - battery_temperature
  - grid_power
  - grid_frequency
  - grid_voltage
  - grid_current
  - grid_ct_power
  - grid_ld_power
  - grid_l2_power
  - grid_connected_status
  - inverter_power
  - inverter_frequency
  - inverter_voltage
  - pv1_power
  - pv1_voltage
  - pv1_current
  - pv2_power
  - pv2_voltage
  - pv2_current
  - load_power
  - load_l1_power
  - load_l2_power
  - day_active_power
  - day_battery_charge
  - day_battery_discharge
  - day_grid_export
  - day_grid_import
  - day_load_power
  - day_pv_energy
  - day_reactive_power
  - month_grid_power
  - month_load_power
  - month_pv_power
  - total_active_power
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - total_load_power
  - total_pv_power
  - year_grid_export
  - year_load_power
  - year_pv_power
  - control_mode
  - priority_mode
  - prog1_time
  - prog2_time
  - prog3_time
  - prog4_time
  - prog5_time
  - prog6_time
  - prog1_power
  - prog2_power
  - prog3_power
  - prog4_power
  - prog5_power
  - prog6_power
  - prog1_capacity
  - prog2_capacity
  - prog3_capacity
  - prog4_capacity
  - prog5_capacity
  - prog6_capacity
  - prog1_charge
  - prog2_charge
  - prog3_charge
  - prog4_charge
  - prog5_charge
  - prog6_charge
  - prog1_voltage
  - prog2_voltage
  - prog3_voltage
  - prog4_voltage
  - prog5_voltage
  - prog6_voltage
READ_SENSORS_BATCH_SIZE: 60
NUMBER_ENTITY_MODE: auto
PROFILES:
  - system_mode
  - system_mode_voltages
MODBUS_SERVER_ID: 1
MQTT_HOST: core-mosquitto
MQTT_PORT: 1883
MQTT_USERNAME: *
MQTT_PASSWORD: *
DEBUG: 0
```

### Logs (if applicable)

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2022-12-15 15:49:05,503 INFO    Loading HASS OS configuration
2022-12-15 15:49:05,522 DEBUG   could not patch umodbus: numpy not found
2022-12-15 15:49:05,529 INFO    ############################################################
2022-12-15 15:49:05,529 INFO      Sensor deprecated: day_active_power -> day_active_energy
2022-12-15 15:49:05,529 INFO    ############################################################
2022-12-15 15:49:05,530 INFO    ############################################################
2022-12-15 15:49:05,530 INFO        Sensor deprecated: day_load_power -> day_load_energy
2022-12-15 15:49:05,530 INFO    ############################################################
2022-12-15 15:49:05,531 INFO    ############################################################
2022-12-15 15:49:05,531 INFO    Sensor deprecated: day_reactive_power -> day_reactive_energy
2022-12-15 15:49:05,531 INFO    ############################################################
2022-12-15 15:49:05,531 INFO    ############################################################
2022-12-15 15:49:05,532 INFO      Sensor deprecated: month_grid_power -> month_grid_energy
2022-12-15 15:49:05,532 INFO    ############################################################
2022-12-15 15:49:05,532 INFO    ############################################################
2022-12-15 15:49:05,532 INFO      Sensor deprecated: month_load_power -> month_load_energy
2022-12-15 15:49:05,533 INFO    ############################################################
2022-12-15 15:49:05,533 INFO    ############################################################
2022-12-15 15:49:05,533 INFO        Sensor deprecated: month_pv_power -> month_pv_energy
2022-12-15 15:49:05,533 INFO    ############################################################
2022-12-15 15:49:05,534 INFO    ############################################################
2022-12-15 15:49:05,534 INFO    Sensor deprecated: total_active_power -> total_active_energy
2022-12-15 15:49:05,534 INFO    ############################################################
2022-12-15 15:49:05,535 INFO    ############################################################
2022-12-15 15:49:05,535 INFO      Sensor deprecated: total_load_power -> total_load_energy
2022-12-15 15:49:05,535 INFO    ############################################################
2022-12-15 15:49:05,535 INFO    ############################################################
2022-12-15 15:49:05,536 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2022-12-15 15:49:05,536 INFO    ############################################################
2022-12-15 15:49:05,536 INFO    ############################################################
2022-12-15 15:49:05,536 INFO       Sensor deprecated: year_load_power -> year_load_energy
2022-12-15 15:49:05,537 INFO    ############################################################
2022-12-15 15:49:05,537 INFO    ############################################################
2022-12-15 15:49:05,537 INFO         Sensor deprecated: year_pv_power -> year_pv_energy
2022-12-15 15:49:05,537 INFO    ############################################################
2022-12-15 15:49:05,539 INFO    Added hidden sensor battery_low_capacity as other sensors depend on it
2022-12-15 15:49:05,539 INFO    Added hidden sensor battery_low_voltage as other sensors depend on it
2022-12-15 15:49:05,539 INFO    Added hidden sensor battery_float_voltage as other sensors depend on it
2022-12-15 15:49:05,540 INFO    Filter *step used for aux_power, battery_current, battery_power, control_mode, day_reactive_power, essential_power, grid_ct_power, grid_current, grid_frequency, grid_l2_power, grid_ld_power, grid_power, inverter_frequency, inverter_power, load_l1_power, load_l2_power, load_power, non_essential_power, pv1_current, pv1_power, pv2_current, pv2_power
2022-12-15 15:49:05,540 INFO    Filter *last used for battery_soc, day_active_power, day_battery_charge, day_battery_discharge, day_grid_export, day_grid_import, day_load_power, day_pv_energy, grid_connected_status, month_grid_power, month_load_power, month_pv_power, total_active_power, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_load_power, total_pv_power, year_grid_export, year_load_power, year_pv_power
2022-12-15 15:49:05,540 INFO    Filter *avg used for battery_temperature, battery_voltage, grid_voltage, inverter_voltage, pv1_voltage, pv2_voltage
2022-12-15 15:49:05,540 INFO    Filter *round_robin used for priority_mode, prog1_capacity, prog1_charge, prog1_power, prog1_time, prog1_voltage, prog2_capacity, prog2_charge, prog2_power, prog2_time, prog2_voltage, prog3_capacity, prog3_charge, prog3_power, prog3_time, prog3_voltage, prog4_capacity, prog4_charge, prog4_power, prog4_time, prog4_voltage, prog5_capacity, prog5_charge, prog5_power, prog5_time, prog5_voltage, prog6_capacity, prog6_charge, prog6_power, prog6_time, prog6_voltage, prog6_voltage, prog6_voltage, prog6_voltage
2022-12-15 15:49:05,560 INFO    Reading startup sensors prog1_time, prog3_time, serial, prog6_time, battery_low_voltage, rated_power, prog4_time, prog5_time, battery_low_capacity, prog2_time, battery_float_voltage
2022-12-15 15:49:05,872 INFO    ############################################################
2022-12-15 15:49:05,873 INFO                Inverter serial number 'xxxxxxxx'
2022-12-15 15:49:05,873 INFO    ############################################################
2022-12-15 15:49:05,884 INFO    MQTT: Connecting to hass@core-mosquitto:1883
2022-12-15 15:49:05,906 INFO    MQTT: Connection successful
```

jabezz, 2022-12-15:
Just to note.
The progX_charge setting does write the correct values to the inverter when changing them, and it seems to remember the settings. Maybe the problem is just that initial value before actually setting it?

ASomerN, 2022-12-16:
Not sure if it helps, I've got this issue with all but one of the ProgX sensors and when I look at the options to set they are rounding  to 6:50,7:05,7:20 instead of 6:45,7:00,7:15

kellerza, 2022-12-16:
I suspect this is related to #72 your original prog_charge registers likely contained something else

@ASomerN no idea what inverter you have, but also completely different to this issu, so best to open your own with more info

jabezz, 2022-12-19:
Could be related to #72 yes.
It is strange that it would not read the original values (not sure what they were or how to check)
But I can set/change the values perfectly, and it reads the values that were manually set fine after that.

kellerza, 2022-12-19:
Some unrecognised bits in the register must have been set. Writing the new value changed them to a known state. 

Unfortunately there is no way to check that now, except to see if something changed in the settings on the inverter…

jabezz, 2022-12-19:
Thanks @kellerza 
I'll close this issue then.

I've left a comment on a closed PR, which I'm not sure was the best and most visible idea [https://github.com/kellerza/sunsynk/pull/57#issuecomment-1354280980](url)

How best to I raise this issue/question?
I'm willing to give the changes a stab if I can get some direction.

Ivan-L, 2022-12-30:
Hi @jabezz I've replied to your comment in that PR - since it's about adding new R/W sensors, can you create a new issue for it?

> Some unrecognised bits in the register must have been set. Writing the new value changed them to a known state.
> 
> Unfortunately there is no way to check that now, except to see if something changed in the settings on the inverter…

Fully agree with this. I have replied in #78 with a similar response.

@jabezz if you can up the verbosity level in the logs so that we can get an idea of what the values of the ProgX..Charge sensors are being set to by the inverter, we might be able to figure out what bits are being set.

kellerza, 2022-12-30:
I believe @jabezz original issue has been resolved. Let's close this one in favour of other more specific issues

closed 2022-12-30 by kellerza

ASomerN, 2023-01-09:
Hi,
I had only started using these sensors when they became available in the inverter add-on (not dev add-on).
I applied debug five to the non-dev addon and it crashed.

I've started the dev add-on with the same settings as the non-dev and this was logged:

```
2023-01-09 17:32:59,278 INFO    Loading HASS OS configuration
2023-01-09 17:32:59,280 INFO    PROFILES were deprecated. Please remove this configuration.
2023-01-09 17:32:59,347 INFO    ############################################################
2023-01-09 17:32:59,348 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2023-01-09 17:32:59,349 INFO    ############################################################
2023-01-09 17:32:59,349 DEBUG   last unit:kWh, id:total_pv_power
2023-01-09 17:32:59,350 DEBUG   last unit:kWh, id:total_battery_charge
2023-01-09 17:32:59,351 DEBUG   last unit:kWh, id:total_battery_discharge
2023-01-09 17:32:59,352 DEBUG   last unit:kWh, id:total_grid_export
2023-01-09 17:32:59,353 DEBUG   last unit:kWh, id:total_grid_import
2023-01-09 17:32:59,353 DEBUG   last unit:%, id:battery_soc
2023-01-09 17:32:59,354 DEBUG   step unit:W, id:grid_power
2023-01-09 17:32:59,355 DEBUG   step unit:W, id:pv1_power
2023-01-09 17:32:59,355 DEBUG   step unit:W, id:pv2_power
2023-01-09 17:32:59,356 DEBUG   step unit:W, id:battery_power
2023-01-09 17:32:59,357 DEBUG   step unit:W, id:load_power
2023-01-09 17:32:59,357 DEBUG   step unit:Hz, id:grid_frequency
2023-01-09 17:32:59,358 DEBUG   step unit:, id:control_mode
2023-01-09 17:32:59,359 INFO    Added hidden sensor battery_low_capacity as other sensors depend on it
2023-01-09 17:32:59,360 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_power
2023-01-09 17:32:59,360 INFO    Filter *step used for battery_power, control_mode, grid_frequency, grid_power, load_power, pv1_power, pv2_power
2023-01-09 17:32:59,361 INFO    Filter *round_robin used for priority_mode, prog1_capacity, prog1_charge, prog1_power, prog1_time, prog2_capacity, prog2_charge, prog2_power, prog2_time, prog3_capacity, prog3_charge, prog3_power, prog3_time, prog4_capacity, prog4_charge, prog4_power, prog4_time, prog5_capacity, prog5_charge, prog5_power, prog5_time, prog6_capacity, prog6_charge, prog6_charge, prog6_power, prog6_time
2023-01-09 17:32:59,362 DEBUG   Using selector: EpollSelector
2023-01-09 17:32:59,364 DEBUG   Connecting.
2023-01-09 17:32:59,387 DEBUG   Client connected to modbus server
2023-01-09 17:32:59,387 INFO    Protocol made connection.
2023-01-09 17:32:59,389 INFO    Connected to /dev/ttyUSB0
2023-01-09 17:32:59,389 INFO    Reading startup sensors serial, prog4_time, prog6_time, rated_power, prog1_time, prog5_time, battery_low_capacity, prog2_time, prog3_time
2023-01-09 17:32:59,394 DEBUG   send: 0x1 0x3 0x0 0x3 0x0 0x5 0x75 0xc9
2023-01-09 17:32:59,396 DEBUG   Adding transaction 1
2023-01-09 17:32:59,438 DEBUG   recv: 0x1
2023-01-09 17:32:59,438 DEBUG   Frame - [b'\x01'] not ready
2023-01-09 17:32:59,439 DEBUG   recv: 0x3 0xa
2023-01-09 17:32:59,440 DEBUG   Frame - [b'\x03\n'] not ready
2023-01-09 17:32:59,441 DEBUG   recv: 0x32
2023-01-09 17:32:59,442 DEBUG   Frame - [b'2'] not ready
2023-01-09 17:32:59,443 DEBUG   recv: 0x32 0x30
2023-01-09 17:32:59,444 DEBUG   Frame - [b'20'] not ready
2023-01-09 17:32:59,444 DEBUG   recv: 0x34
2023-01-09 17:32:59,445 DEBUG   Frame - [b'4'] not ready
2023-01-09 17:32:59,446 DEBUG   recv: 0x31 0x39
2023-01-09 17:32:59,447 DEBUG   Frame - [b'19'] not ready
2023-01-09 17:32:59,448 DEBUG   recv: 0x34 0x30
2023-01-09 17:32:59,448 DEBUG   Frame - [b'40'] not ready
2023-01-09 17:32:59,449 DEBUG   recv: 0x37
2023-01-09 17:32:59,450 DEBUG   Frame - [b'7'] not ready
2023-01-09 17:32:59,451 DEBUG   recv: 0x30 0x3e
2023-01-09 17:32:59,452 DEBUG   Frame - [b'0>'] not ready
2023-01-09 17:32:59,452 DEBUG   recv: 0xa
2023-01-09 17:32:59,453 DEBUG   Getting Frame - 0x3 0xa 0x32 0x32 0x30 0x34 0x31 0x39 0x34 0x30 0x37 0x30
2023-01-09 17:32:59,453 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2023-01-09 17:32:59,453 DEBUG   Frame advanced, resetting header!!
2023-01-09 17:32:59,454 DEBUG   Getting transaction 1
2023-01-09 17:32:59,455 DEBUG   Request registers: [3, 4, 5, 6, 7] glen=5. Response [12850, 12340, 12601, 13360, 14128] len=5. regs={3: 12850, 4: 12340, 5: 12601, 6: 13360, 7: 14128}
2023-01-09 17:32:59,456 DEBUG   send: 0x1 0x3 0x0 0x10 0x0 0x2 0xc5 0xce
2023-01-09 17:32:59,457 DEBUG   Adding transaction 1
2023-01-09 17:32:59,497 DEBUG   recv: 0x1
2023-01-09 17:32:59,498 DEBUG   Frame - [b'\x01'] not ready
2023-01-09 17:32:59,498 DEBUG   recv: 0x3
2023-01-09 17:32:59,499 DEBUG   Frame - [b'\x03'] not ready
2023-01-09 17:32:59,499 DEBUG   recv: 0x4
2023-01-09 17:32:59,500 DEBUG   Frame - [b'\x04'] not ready
2023-01-09 17:32:59,500 DEBUG   recv: 0x38
[... 9 lines cut ...]
2023-01-09 17:32:59,506 DEBUG   recv: 0x7b
2023-01-09 17:32:59,506 DEBUG   Getting Frame - 0x3 0x4 0x38 0x80 0x0 0x1
2023-01-09 17:32:59,506 DEBUG   Factory Response[ReadHoldingRegistersResponse: 3]
2023-01-09 17:32:59,507 DEBUG   Frame advanced, resetting header!!
2023-01-09 17:32:59,507 DEBUG   Getting transaction 1
2023-01-09 17:32:59,509 DEBUG   Request registers: [16, 17] glen=2. Response [14464, 1] len=2. regs={16: 14464, 17: 1}
2023-01-09 17:32:59,509 DEBUG   send: 0x1 0x3 0x0 0xdb 0x0 0x1 0xf4 0x31
2023-01-09 17:32:59,510 DEBUG   Adding transaction 1
[... cut ...]
2023-01-09 17:33:09,408 ERROR   Timeout reading: 
2023-01-09 17:33:09,408 INFO    ############################################################
2023-01-09 17:33:09,409 INFO    No response on the Modbus interface, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2023-01-09 17:33:09,409 INFO    ############################################################
2023-01-09 17:33:09,409 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
```

kellerza, 2023-01-10:
@ASomerN from the logs you could not connect to the inverter. Most likely you still had the non-dev addon running.

Only one app can access the USB port at a time.

The only way to get 2x to connect is to use the mbusd addon to connect to the USB port and then you connect via tcp://ip:502 to mbusd from the addons

ASomerN, 2023-01-19:
Hi,
I updated to the latest version and turned off the non-dev version. restarted the PI and then started dev with the config and it works.
Everything is working correctly on the latest dev version. The options are not flicking on/off. Updating each sensor is reflected right away on the inverter and in HA.
Looks to be resolved on the dev add-on.