kellerza/sunsynk #86: Failed to install add-on

timclark0000, 2023-01-19:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

Following error message when attempting to install add-on:

"The command '/bin/ash -o pipefail -c pip3 install --no-cache-dir --disable-pip-version-check paho-mqtt~=1.5.0 pyyaml~=5.4.1 sunsynk[pymodbus,umodbus]==0.2.5' returned a non-zero code: 1"

**Expected behavior**
A clear and concise description of what you expected to happen.

### Your environment

- Home Assistant version: 2022.12.1
- Addon version:2022.11.29-0.2.5
- Inverter make: Sunsynk
- Inverter Model: \_\_
- Inverter Firmware: \_\_

### Logs

```log
paste logs in 3x backticks...
```

kellerza, 2023-01-19:
You can try installing the dev version of the addon, but my first thought would be that you have bad DNS/Internet connectivity

Can you configure static IP&DNS on your HASS/raspberry pi?

timclark0000, 2023-01-19:
Thanks for your response. I've tried the dev version but it throws the same error. I've reconfigured my network with five different DNS servers now and each one also produces the same error. My Home Assistant instance is running in VirtualBox on a Windows 10 machine, if that helps at all. Happy to try changing various parameters at home to test this.

kellerza, 2023-01-27:
Any feedback?

amctear, 2023-01-27:
Hi All,
Not sure how to address my HF2221 Modbus server to Ethernet, - Port, Device & Driver ?  Any suggestions ?

[image]

)

kellerza, 2023-01-28:
The driver is ok, and device is ignored when you have a port

amctear, 2023-01-28:
> The driver is ok, and device is ignored when you have a port

Perhaps the wiring to the inverter id my problem.  I understand you tried a HF5142B, what was your wiring ?   
I've currently 

Green/White to T-
Green to T+
Orange/White to GND

Thoughts please 

[image]

amctear, 2023-01-29:
Got it working and so very happy and 3 or 4 days pulling my hair out. Feeling rather proud of myself now :-)

My mistake was the wiring of the CAT 5 cable to the Modbus Server,  it wasn't 
Green/White to T-
Green to T+
Orange/White to GND

Ended up being Wires 2 & 7 to A  and  1 & 8 to B.  Modbus Poll imediately :-) 👍

kellerza, 2023-01-31:
Ideally you should not have to wire both 1&2 AND 7&8, but if this is working for you its probably ok. The are other people on powerforum also doing this

kellerza, 2023-01-31:
Closing as there is no response from @timclark0000 and his problem is related to the environment rather than the addon

closed 2023-01-31 by kellerza