kellerza/sunsynk #94: Program Capacity not updating

duanemck, 2023-01-31:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

When setting the program mode capacity through HA, the change is not being effected on the inverter. I can see the message going out on MQTT but the change is not logged on the add-on and also the change doesn't happen to the inverter. I continued receiving updates via MQTT.

A restart of the add-on fixed it but I have some automations setup to manage batteries in prep for loadshedding and this caused me to have a flat battery this morning.

**Expected behavior**
The change in HA is applied to the inverter.

### Your environment

- Home Assistant version: 2023.1.7
- Addon version: 2023.01.19-0.2.9 (Dev version as prod has the bug with program modes being read wrong)
- Inverter make: Sunsynk
- Inverter Model: 5.5kW
- Inverter Firmware: Not sure

### Logs

```log
There was nothing showing in the logs, the last log entry was about a day old.
```

kellerza, 2023-02-03:
Try enabling debug & post the log.

Even without debug it should post a write message

duanemck, 2023-02-03:
I haven't seen it happen again since I restarted. I'll enable debug logging and keep an eye on it, if it happens I'll hopefully have some logs for you.

jacekowski, 2023-02-23:
I've seen the same issue as well, only happens after few days of running. I will try to log something as well

duanemck, 2023-03-06:
I'm seeing happen almost daily at the moment but I also have automations in place to detect it and restart the add-on when that happens. This seems to clear the logs though, is there a way for me to check the logs before it restarted?

(I am using automations to make sure the batteries are properly charged before loadshedding, so I'd prefer to not disable the auto-healing I have in place)

kellerza, 2023-03-12:
If you have such an automation in place, why not send yourself a notification and then you can save the log?

duanemck, 2023-03-13:
> If you have such an automation in place, why not send yourself a notification and then you can save the log?

Yeah, good idea. I'm already sending myself a telegram message, I'll just have to delay the restart for maybe 10 minutes so I have time to capture the log and it still works if I'm not around to check.

jacekowski, 2023-03-26:
I've just had the issue happen again, nothing in the logs.

But it seems like i might have found the culprit, it seems like whenever MQTT broker is restarted, the addon reconnects to the broker and publishes everything correctly, but does not subscribe to _set topics.

kellerza, 2023-03-26:
Not sure I understand exactly what you mean.

what (should) happen today:
- When config messages/values are published by the addon it should be persistent in MQTT
- Persistent MQTT messages should have a value in HASS as long as the device is available 
- Device availability is based on the fact that the addon is connected to MQTT
- Availability is removed when the device disconnects, or terminates unexpectedly through MQTT last_will
- Set messages is sent by HASS when a user changes input
- If the addon is not working properly it cannot update the value topic and the value should not change in HASS (because of persistentcy)
- I’m not sure when HASS will discard an attempt to set in the web ui if the value stays the same (because of not updating).
- Since you only read MQTT message as events (as it happens) or a flood of persistent ones once you connect,  it MIGHT be that HASS expects a message back to tell it the UI should revert to the correct message.

So if this happens again, can you try restarting HASS (not the entire OS/supervisor) and see if the value changes to the expected one? (Or the correct value as per the inverter)

jacekowski, 2023-03-26:
What happens now is:

- HA is up and running
- MQTT broker is up and running (and HA is connected to it)
- Sunsynk addon is started
- Sunsynk addon subscribes to *_set topics in MQTT broker
- Sunsynk addon starts publishing sensor topics
- MQTT broker is sending subscribed topics to Sunsynk addon as expected
- Everything works as expected at this point
- MQTT broker is restarted (or connection to MQTT broker is lost for some other reason)
- Sunsynk addon reconnects to MQTT broker 
- Sunsynk addon does not subscribe to *_set topics on reconnection 
- Sunsynk addon is publishing sensor topics as normal

kellerza, 2023-03-26:
Thanks @jacekowski! 

This should be better in the new multi version.

closed 2023-04-07 by kellerza