kellerza/sunsynk #100: Change grid charge sensor to rwsensor

jacekowski, 2023-02-23:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)

### You Inverter

5kW sunsynk

### Describe the issue/bug and what you expect

I would like to see a change to register 230 (Grid charge current) to be writeable ( https://github.com/kellerza/sunsynk/commit/ee156abe0dc3dfd3798eece9113b345584915b79 - this commit should achieve that) 

I'm on economy 7 tariff so normally have 7hrs to charge my battery so i keep charge current low to help with efficiency and battery life.  I'm also adjusting target SoC based on weather forecast, however, for example today, just 30 minutes before end of cheap period forecast changed to be significantly worse, if that happens i would like to adjust current to allow as much charging as possible and then reset it back to low current for next day.

kellerza, 2023-03-11:
The library has not been released (version needs to be bumped) and dockerfile updated for the addons

cc120689, 2023-03-11:
ok, we will wait patiently, thank you!

kellerza, 2023-03-12:
I updated the library, but did NOT bump the addon version (this is combined with some other changes I have not finished testing, - but it is not related to the dev addon) 

So auto-update will not work, if you delete the dev addon and reinstall the new version should be built in.

cc120689, 2023-03-12:
ok I'll try it and comment, thanks!

kellerza, 2023-03-14:
Great stuff!!

closed 2023-03-14 by kellerza