kellerza/sunsynk #102: Sum of PV1 power and PV2 power

kamransaleem, 2023-02-27:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)

### You Inverter

Sunsynk 8kw

### Describe the issue/bug and what you expect

A sum of pv1 and pv2 power so that I can create automations based on the combined power generated

jabezz, 2023-02-27:
You can do this in Home Assistant.

Add some template sensors in your configuration.yaml

```
template:
  - sensor:
      - name: "Total Solar Power"
        unit_of_measurement: 'W'
        state: >
          {{ (states("sensor.ss_pv1_power") | float + states("sensor.ss_pv2_power") | float) | round(2) }}
        device_class: power
        state_class: measurement
  - sensor:
      - name: "Total Solar Current"
        unit_of_measurement: 'A'
        state: >
          {{ (states("sensor.ss_pv1_current") | float + states("sensor.ss_pv2_current") | float) | round(2) }}
        device_class: current
        state_class: measurement
  - sensor:
      - name: "Total Solar Voltage"
        unit_of_measurement: 'V'
        state: >
          {{ (states("sensor.ss_pv1_voltage") | float + states("sensor.ss_pv2_voltage") | float) | round(2) }}
        device_class: voltage
        state_class: measurement

```

reboot and voila.

kamransaleem, 2023-02-28:
that worked great, thank you

closed 2023-02-28 by kamransaleem

BrendanW74, 2023-03-04:
> that worked great, thank you

Hi there kamransaleem

I was wondering if you could please help. I have been trying to get the above to work in HA for the past 3 days now.

The code is not the problem, I literally copied/pasted it into my "configuration.yaml" file, and ran the "CHECK CONFIGURATION" option under dev tools, and it states "Configuration will not prevent Home Assistant from starting!".

Once this has been added, where/how do you gain access to the PV1+PV2 result?
I am not sure where I am able to use the combined information within HA.

Any help would be most appreciated.
Thank you in advance.

jabezz, 2023-03-04:
The Entity will be available to add to any card.

[image]
[image]

or 

```
type: entities
entities:
  - entity: sensor.total_solar_power
    name: Solar Power
```
or
[image]
[image]

you can also check the values in Developer Tools
[image]

BrendanW74, 2023-03-05:
Hi there jabezz,

Thank you so much for the prompt reply, and the assistance offered, much appreciated!

Cheers for now

rkistner, 2023-03-16:
This is great for the total power.

Note however that adding the currents or voltages won't give a meaningful values - the combined voltage and current won't match the combined power. The inverter limits are applied to each PV string separately, not on total values.

kellerza, 2023-03-17:
> Note however that adding the currents or voltages won't give a meaningful values - the combined voltage and current won't match the combined power. The inverter limits are applied to each PV string separately, not on total values.

Good point, the inverter only uses the power output from the MPPTs and each string has its own MPPT

BrendanW74, 2023-03-18:
> This is great for the total power.
> 
> Note however that adding the currents or voltages won't give a meaningful values - the combined voltage and current won't match the combined power. The inverter limits are applied to each PV string separately, not on total values.

Hi rkistner,

Thanks for weighing in on this.

Cheers

kellerza, 2023-04-22:
With custom sensors (mysensors) supported in the MULTI addon, you could create your own sensor to output the combined PV power... 

See the discussion [here](https://powerforum.co.za/topic/16136-home-assistant-inverter-integration-options/?do=findComment&comment=162980)