kellerza/sunsynk #122: Monthly sensors

w1tw0lf, 2023-04-06:
There seems to be an error with the monthly sensors for the load and grid usage.

Load on sunsynk website
[image]
load on home assistant
[image]

Grid on sunsynk website
[image]
grid on home assistant
[image]

The values on sunsynk website correspond with energy dashboard, where I used the total gird/total load sensors. 

The solar is alos of by about 3 will keep an eye on it to see if it goes out by more.

Sunsynk Inverter Add-on (multi)
Current version: 2023.03.19b-0.3.1

Home assistant version 2023.4

kellerza, 2023-04-06:
Is this a single or three phase inverter?

The single phase definitions today are:

```python
    Sensor(84, "Day Load Energy", KWH, 0.1),
    Sensor(108, "Day PV Energy", KWH, 0.1),
    Sensor(61, "Day Reactive Energy", "kVarh", -0.1),
    Sensor(67, "Month Grid Energy", KWH, 0.1),
    Sensor(66, "Month Load Energy", KWH, 0.1),
    Sensor(65, "Month PV Energy", KWH, 0.1),
```

With the new addon you can test sensor definitions by defining [custom sensors](https://kellerza.github.io/sunsynk/reference/mysensors) - using definitions similar above / even overwriting them

So if these are important to you, please investigate these definitions & propose a PR.

kellerza, 2023-04-06:
You can also refer to #59 for a newer version of the Modbus protocol doc

w1tw0lf, 2023-04-07:
> Is this a single or three phase inverter?
>
> The single phase definitions today are:
>
> ```python
>     Sensor(84, "Day Load Energy", KWH, 0.1),
>     Sensor(108, "Day PV Energy", KWH, 0.1),
>     Sensor(61, [...]

it is single phase. Those are the definitions that I am using at this stage. Day sensors is correct, month sensors seems to be out with the values it is reporting.

w1tw0lf, 2023-04-07:
Noticed the drop in values around midnight
[image]
[image]

 "Month Grid Energy" doesn't show this behaviour.

kellerza, 2023-04-07:
Keep an eye on it, the inverter might combine this with other data (maybe plot the value of the day sensors for the same time)

Personally, I only use the total sensors + HA Energy management. It covers this use case (and more), so not something I'm going to investigate further.

w1tw0lf, 2023-05-19:
Switch to using HA utility meter and working 100s.