kellerza/sunsynk #124: Latest dev version "TypeError: 'method' object is not iterable"

jpmeijers, 2023-04-12:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

A clear and concise description of what the bug is.

Just updated to the latest and now the adon stops running.

### Your environment

- Home Assistant version: 2023.4.2
- Addon version: 2023.01.19-0.2.9 (dev)
- Inverter make: Sunsynk
- Inverter Model: \_\_
- Inverter Firmware: \_\_

### Logs

```log
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-04-12 09:22:40,471 INFO    Loading HASS OS configuration
2023-04-12 09:22:40,471 INFO    PROFILES were deprecated. Please remove this configuration.
2023-04-12 09:22:40,489 INFO    ############################################################
2023-04-12 09:22:40,489 INFO        Sensor deprecated: total_pv_power -> total_pv_energy
2023-04-12 09:22:40,489 INFO    ############################################################
2023-04-12 09:22:40,489 DEBUG   last unit:kWh, id:total_pv_power
2023-04-12 09:22:40,489 DEBUG   last unit:kWh, id:total_battery_charge
2023-04-12 09:22:40,489 DEBUG   last unit:kWh, id:total_battery_discharge
2023-04-12 09:22:40,489 DEBUG   last unit:kWh, id:total_grid_export
2023-04-12 09:22:40,490 DEBUG   last unit:kWh, id:total_grid_import
2023-04-12 09:22:40,490 DEBUG   last unit:%, id:battery_soc
2023-04-12 09:22:40,490 DEBUG   step unit:W, id:essential_power
2023-04-12 09:22:40,490 DEBUG   step unit:W, id:non_essential_power
2023-04-12 09:22:40,490 DEBUG   step unit:W, id:grid_ct_power
2023-04-12 09:22:40,490 DEBUG   step unit:W, id:battery_power
2023-04-12 09:22:40,490 DEBUG   avg unit:V, id:battery_voltage
2023-04-12 09:22:40,490 DEBUG   step unit:A, id:battery_current
2023-04-12 09:22:40,490 DEBUG   avg unit:°C, id:battery_temperature
2023-04-12 09:22:40,490 DEBUG   step unit:W, id:grid_power
2023-04-12 09:22:40,490 DEBUG   step unit:Hz, id:grid_frequency
2023-04-12 09:22:40,490 DEBUG   avg unit:V, id:grid_voltage
2023-04-12 09:22:40,491 DEBUG   step unit:A, id:grid_current
2023-04-12 09:22:40,491 DEBUG   step unit:W, id:inverter_power
2023-04-12 09:22:40,491 DEBUG   step unit:Hz, id:inverter_frequency
2023-04-12 09:22:40,491 DEBUG   step unit:W, id:pv1_power
2023-04-12 09:22:40,491 DEBUG   avg unit:V, id:pv1_voltage
2023-04-12 09:22:40,491 DEBUG   step unit:A, id:pv1_current
Traceback (most recent call last):
  File "/usr/src/app/./run.py", line 472, in <module>
    startup()
  File "/usr/src/app/./run.py", line 261, in startup
    setup_sensors()
  File "/usr/src/app/./run.py", line 307, in setup_sensors
    for dep in sen.dependencies:
TypeError: 'method' object is not iterable
[cmd] ./run.py exited 1
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

jpmeijers, 2023-04-12:
I have a single Sunsynk inverter, so using "multi" doesn't make sense.

I'm already using mbusd running on a raspberry pi.

kellerza, 2023-04-12:
Maybe this will fix it - https://github.com/kellerza/sunsynk/commit/0c9d99ef7accbe982a8c7f4f9371772391759aab

You will have to force an update since the version did not change

kellerza, 2023-04-12:
Glad that works for you. Note that only multi will get updates going forward

closed 2023-04-12 by kellerza

jpmeijers, 2023-04-12:
Changing to Multi works too. Thanks!

kellerza, 2023-04-12:
Ideally everyone should use Multi. But there is no way to seamlessly migrate, since the config changed.

Overall it should be better. It's only the serial driver that is broken in Multi 😉

kellerza, 2023-04-21:
This was fixed recently, but I still recommend the MULTI addon