kellerza/sunsynk #125: Profile Definition Example(s)

vexersa, 2023-04-16:
Existing sensor definitions can be found [here](https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions.py)

### You Inverter

Details about your Inverter: SunSynk 8Kw

### Describe the issue/bug and what you expect

I would like to make my Sunsynk load shedding aware to change some target SOC's etc depending on my load shedding schedule for the day. 

I believe Profiles can do this, but I don't have an example of how to configure the options. There isn't much information on how this works in the documentation other than a description of its functionality. 

### Logs (if applicable)

```
paste logs in 3x backticks...
```

kellerza, 2023-04-16:
Use the Multi version of the addon - progX_capacity sensors are writeable.

Here you can find a lovelace card to change settings (after you added the sensors)
https://kellerza.github.io/sunsynk/examples/lovelace-settings

duanemck, 2023-04-17:
@vexersa I am using Home Assistant and NodeRed to make my inverter Loadshedding aware. Happy to give some pointers. But as @kellerza says, it's really as simple as updating those progX_capacity sensors on HA and the values get written to the inverter by the addon.

vexersa, 2023-04-17:
Thanks guys let me give this a crack.

vexersa, 2023-04-17:
@kellerza 

Just wanted to check which version the multi is? Is it v0.3.0?

vexersa, 2023-04-17:
Nevermind, got it through the HA Addon UI

kellerza, 2023-04-17:
Updated the examples to include sensor names - https://kellerza.github.io/sunsynk/examples/lovelace-settings#example-1

closed 2023-04-18 by kellerza