kellerza/sunsynk #127: No Issue, hoping for some help from Johann

amctear, 2023-04-19:
My Sunsynk addon x 2 have been working wonderfully since installed and set, thanks to Johann.  An interesting Blueprint PV solar excess optimizer - auto control, has recently been released which requres some template skills to generate a new sensor **Export power to the grid in watts.**  
Sadly, I don't have the skills to strip out the **grid power out** portion from the **grip power sensor**.  Hoping someone can help please
[image]

Blueprint
[https://community.home-assistant.io/t/pv-solar-excess-optimizer-auto-control-appliances-wallbox-dish-washer-heatpump-based-on-excess-solar-power/552677](url)

kellerza, 2023-04-20:
```yaml
SENSORS:
 - grid_ct_power
```

The template could be something like this?
 
```
{{ -states("sensor.ss_grid_ct_power") if int(states("sensor.ss_grid_ct_power"))<0 else 0 }}
```

Let us know how it goes, I'll also have a look at the referenced project!

amctear, 2023-04-20:
Thanks, Will give it a try

closed 2023-06-04 by amctear