kellerza/sunsynk #128: ESP Integration | The Entity Is Unavailable | sensor.eskomsepushallowance

vexersa, 2023-04-20:
### Issue related to

Sunsynk / mbusd Home Assistant Add-On

### Describe the issue/bug

When following the guide to add the ESP lovelace card, the following error is presented:

[image]
[image]

Seems to be related to this: 
https://www.home-assistant.io/faq/unique_id

**Expected behavior**
The card to work correctly.

### Your environment

Home Assistant 2023.4.5
Supervisor 2023.04.0
Operating System 10.0
Frontend 20230411.1 - latest
Sunsynk Inverter Add-on (multi)
Current version: 2023.03.19b-0.3.1

### Logs

```log
paste logs in 3x backticks...
```

kellerza, 2023-04-20:
Updated the docs to make it a bit clearer, start here: https://kellerza.github.io/sunsynk/examples/overview#config-packages

Once you have the package loaded you should have new eskom entities - you can see these in Dev tools > States

[image]

If all of this is new, read the Home Assistant docs & join the HA Community forum

closed 2023-04-20 by kellerza

vexersa, 2023-04-20:
Great, that fixed it. Thanks for the knowledge 🙏