kellerza/sunsynk #131: Multi Version - Read Errors - Crash

cakuhn, 2023-04-21:
HI

On the latest version, installed this morning (Sunsynk Inverter Add-on (multi): Current version: 2023.04.20b-0.3.2 (Changelog)) I get the below errors as reported in the log.  The application then crash.  Any ideas on a possible fix and the reason for these?

Thanks

2023-04-21 07:32:39,121 ERROR   Read Error: <class 'Exception'>: (1,59,1) 'Serial' object has no attribute 'host'
2023-04-21 07:32:52,742 ERROR   Read Error: <class 'Exception'>: (1,268,6) 'Serial' object has no attribute 'host'
2023-04-21 07:33:03,835 ERROR   Read Error: <class 'Exception'>: (1,59,1) 'Serial' object has no attribute 'host'
2023-04-21 07:33:18,948 ERROR   Read Error: <class 'Exception'>: (1,59,1) 'Serial' object has no attribute 'host'
2023-04-21 07:34:38,121 ERROR   Read Error: <class 'Exception'>: (1,78,1) 'Serial' object has no attribute 'host'

kellerza, 2023-04-21:
Can you try the **pymodbus** driver?

kellerza, 2023-04-21:
@cakuhn can you try these options?

The actual value of the port below you can get from the `DEBUG_DEVICE` field in the UI (at the bottom of the settings)

1. Direct Serial + pymodbus
    ```yaml
    DRIVER: pymodbus
    INVERTERS:
      - PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0
    ```

2. Direct serial + umodbus
    ```yaml
    DRIVER: umodbus
    INVERTERS:
      - PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0
    ```

3. umodbus via mbusd (and mbusd to serial)
    This will work, but won't help me to test the serial implementation

Ko7Ad, 2023-04-24:
I had the exact same issue after migrating to the new release. I am glad I came across this thread as, touch wood, Direct Serial + pymodbus seemed to have resolved it. 

There was a thread on powerforum with a user with the same issue, I will link them here for resolution.

PS. thanks for your work on this addon!

hakonnissen, 2023-05-01:
Same failure on my side since upgrading to HassOS 10.1. I've tried all 3 methods above to no avail.
- When using Direct Serial + pymodbus I get no errors, but also no sensors available:
2023-05-01 17:01:07,796 INFO    ############################################################
2023-05-01 17:01:07,796 INFO                Inverter serial number '2209272082'
2023-05-01 17:01:07,796 INFO    ############################################################
2023-05-01 17:01:07,798 INFO    MQTT: Connecting to mqtt@127.0.0.1:1883

- When using Direct serial + umodbus I get similar error as above:
2023-05-01 17:06:40,723 ERROR   Read Error: <class 'Exception'>: (1,16,2) 'Serial' object has no attribute 'host'
2023-05-01 17:06:40,744 INFO    ############################################################
2023-05-01 17:06:40,744 INFO    No response on the Modbus interface serial:///dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc

Any help would be appreciated. Thnx

kellerza, 2023-05-01:
> * When using Direct Serial + pymodbus I get no errors, but also no sensors available:
>   2023-05-01 17:01:07,796 INFO    ############################################################
>   2023-05-01 [...]

This is good. You either have no sensors in the addon config, or no auto-discovery enabled in HA

hakonnissen, 2023-05-01:
Thnx for responding. I definitely haven't turned off auto-discovery in HA and my addon config lists the following sensors:
  - total_pv_energy
  - total_battery_charge
  - total_battery_discharge
  - total_grid_export
  - total_grid_import
  - battery_soc
However on starting the addon with pymodbus, show me:
2023-05-01 20:30:08,423 INFO    Filter *last used for battery_soc, total_battery_charge, total_battery_discharge, total_grid_export, total_grid_import, total_pv_energy

I'm not a complete newb, but I must admit, this has had me pulling my hair out the last week! Any help would be appreciated. Thnx

cakuhn, 2023-05-02:
> ```yaml
> DRIVER: pymodbus
> INVERTERS:
>   - PORT: ""
> ```

Thanks so much, This one worked for me "pymodbus - Using the device selected in DEBUG_DEVICE (at the bottom of the settings)".

All time out errors experienced on the original version is also gone now.  Using the NOW sensors, it also updates almost instantly by the second.  AWESOMENESS

kellerza, 2023-05-02:
> Thnx for responding. I definitely haven't turned off auto-discovery in HA and my addon config lists the following sensors:
>
> * total_pv_energy
> * total_battery_charge
> * total_battery_discharge
> * [...]

Best to open your own issue, as the driver seems to only be one of your problems. Please include all detail there including your mqtt server info

hakonnissen, 2023-05-02:
Thnx guys. After much struggling, the problem was the connection to the MQTT server! Prior to HassOS 10.1, connecting to HA internal MQTT server on 127.0.0.1 was working fine, but fails for some reason in 10.1.
FINALLY working again. Thanks for a great addon!

NeillHerbst, 2023-08-15:
> @cakuhn can you try these options?
> 
> 0. pymodbus - Using the device selected in `DEBUG_DEVICE` (at the bottom of the settings)
>    ```yaml
>    DRIVER: pymodbus
>    INVERTERS:
>      - PORT: ""
>    ```
> 1. Direct Serial + pymodbus
>    ```yaml
>    DRIVER: pymodbus
>    INVERTERS:
>      - PORT: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0
>    ```
> 2. Direct serial + umodbus
>    ```yaml
>    DRIVER: umodbus
>    INVERTERS:
>      - PORT: serial:///dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A9007CPR-if00-port0
>    ```
> 3. umodbus via mbusd (and mbusd to serial)
>    This will work, but won't help me to test the serial implementation.

 I have a similar issue as reported at the opening of the issue. All pymodbus options give me the following error:
```
2023-08-15 19:28:56,497 INFO    Loading HASS OS configuration
2023-08-15 19:28:56,736 INFO    Using Single phase sensor definitions.
2023-08-15 19:28:56,740 INFO    Connecting to /dev/serial/by-id/usb-Exar_Corp._XR21B1411_E5019832061-if00
2023-08-15 19:28:56,741 INFO    PyModbus 3.2.2 Serial: /dev/serial/by-id/usb-Exar_Corp._XR21B1411_E5019832061-if00
2023-08-15 19:28:56,745 INFO    Serial connected.
2023-08-15 19:28:56,747 INFO    Connected to /dev/serial/by-id/usb-Exar_Corp._XR21B1411_E5019832061-if00
2023-08-15 19:28:56,747 INFO    Reading startup sensors rated_power, serial
2023-08-15 19:28:59,756 INFO    Serial lost connection.
2023-08-15 19:28:59,758 INFO    Serial lost connection.
2023-08-15 19:28:59,759 WARNING Ignoring launch of delayed reconnection, another is in progress
2023-08-15 19:28:59,766 ERROR   timeout reading register 3 (5)
2023-08-15 19:28:59,767 INFO    ############################################################
2023-08-15 19:28:59,768 INFO    No response on the Modbus interface /dev/serial/by-id/usb-Exar_Corp._XR21B1411_E5019832061-if00, try checking the wiring to the Inverter, the USB-to-RS485 converter, etc
2023-08-15 19:28:59,768 INFO    ############################################################
2023-08-15 19:28:59,769 CRITICAL This Add-On will terminate in 30 seconds, use the Supervisor Watchdog to restart automatically.
2023-08-15 19:28:59,869 INFO    Serial connected.
2023-08-15 19:28:59,871 INFO    Connected to /dev/serial/by-id/usb-Exar_Corp._XR21B1411_E5019832061-if00
```

The umodbus options gives me the same error: `'Serial' object has no attribute 'host'`

The cable should work as I have tested it with a SolarAssitant setup, however, that was on a Deye 5kw inverter. I'm trying to install this add-on on a Sunsynk 8kW,

`mbusb` does not seem to connect.

Any ideas?

kellerza, 2023-08-18:
Try the faul finding guide & make sure you plug it into the correct port - see the adaptors & wiring page. Maybe this is a newer model with a shared port?

<https://kellerza.github.io/sunsynk/guide/fault-finding>

NeillHerbst, 2023-08-18:
I've tried both ports to test whether it is a combined port. I've also checked everything on the fault finding page.

As stated previously the cable wiring is definitely correct as it work on a previous SolarAssistant setup. Is there a way to check the port on the Inverter, i.e. that it is actually working.

kellerza, 2023-08-18:
Try from your laptop with modbuspoll/modbus CLI

If still not working ~R100 will get you a different RS485 adaptor & run a short wire while standing next to the inverter.

Personally, I would just redo the wire since you changed the Inverter

Leecavturbo, 2023-09-03:
Ok I preserved ( 11hrs solid lol ) .
I got a connection after entering the inverter serial number and changing port to “ “.
So the log went further than before but stopped at seemingly waiting for mqtt ? So I added mqtt broker and now it works except I’m missing some data like pv2 and total solar ( pv1+pv2 ). How can I add the sensor ? 
also power flow card doesn’t show data.
09:10:04 INFO                 Inverter serial number '****6757'
09:10:04 INFO    ############################################################
09:10:04 INFO    MQTT: Connecting to Home@core-mosquitto:1883

Then 

09:10:08 INFO    MQTT: Connection successful
09:10:08 INFO    Read every (inverter 1)
+-----+---------------------------------------------------------------------------------+
|  s  |                                     Sensors                                     |
+-----+---------------------------------------------------------------------------------+
|  2  | grid_ct_power, inverter_power, grid_power, pv1_power, battery_power, aux_power, |
|     |                       non_essential_power, essential_power                      |
|  5  |      prog1_power, prog5_capacity, prog3_charge, prog3_time, prog4_capacity,     |
|     |   prog1_time, prog1_charge, prog6_capacity, use_timer, load_limit, prog5_time,  |
|     |  prog3_capacity, prog6_power, prog2_time, battery_low_capacity, prog2_capacity, |
|     |     prog4_charge, priority_load, prog5_charge, prog1_capacity, prog2_power,     |
|     |  prog6_time, prog5_power, prog2_charge, prog4_power, prog6_charge, prog4_time,  |
|     |                                   prog3_power                                   |
|  15 |    grid_frequency, overall_state, battery_soc, battery_voltage, grid_voltage,   |
|     |   pv1_current, grid_connected, battery_current, inverter_current, pv1_voltage,  |
|     |                                  load_frequency                                 |
| 300 |            total_grid_import, day_battery_discharge, total_pv_energy,           |
|     |        total_battery_discharge, day_battery_charge, total_battery_charge,       |
|     |       day_pv_energy, day_grid_export, total_grid_export, day_grid_import,       |
|     |                                 day_load_energy                                 |
+-----+---------------------------------------------------------------------------------+
09:10:09 INFO    Report every (inverter 1)
+-----+----------------------------------------------------------------------------------+
|  s  |                                     Sensors                                      |
+-----+----------------------------------------------------------------------------------+
|  2  | grid_ct_power, inverter_power, grid_power, pv1_power, battery_power, aux_power,  |
|     |                       non_essential_power, essential_power                       |
| 300 |      overall_state, prog1_power, day_pv_energy, battery_soc, prog3_charge,       |
|     |     prog3_time, prog4_capacity, prog5_capacity, prog1_time, day_grid_import,     |
|     |      prog1_charge, day_load_energy, prog6_capacity, use_timer, load_limit,       |
|     |        total_battery_charge, prog5_time, day_grid_export, load_frequency,        |
|     |         prog3_capacity, prog6_power, day_battery_discharge, prog2_time,          |
|     |    battery_low_capacity, total_pv_energy, prog2_capacity, day_battery_charge,    |
|     |   priority_load, prog4_charge, battery_voltage, prog5_charge, prog1_capacity,    |
|     | battery_current, inverter_current, total_grid_export, pv1_voltage, prog2_power,  |
|     |    prog6_time, prog5_power, prog2_charge, total_grid_import, grid_frequency,     |
|     | total_battery_discharge, grid_voltage, pv1_current, grid_connected, prog4_power, |
|     |                      prog6_charge, prog4_time, prog3_power                       |
+-----+----------------------------------------------------------------------------------+

kellerza, 2023-09-03:
@Leecavturbo nice!

The docs will show you that `pv2_power` is another sensor. There is no combined sensor for pv1+pv2 today, but you can build one with mysensors or a template (mysensors is probably more accurate)

If power flow card shows nothing you probably have incorrect entity names. Check these under devices/dev tools- state

Leecavturbo, 2023-09-03:
> @Leecavturbo nice!
>
> The docs will show you that `pv2_power` is another sensor. There is no combined sensor for pv1+pv2 today, but you can build one with mysensors or a template (mysensors is [...]

Thanks any / all help pointers appreciated for a novice 👍

N1c084, 2023-09-03:
You can use total PV energy sensor for the flow chart .
System do the calcul automatically and show only the day prod .

Leecavturbo, 2023-09-03:
> You can use total PV energy sensor for the flow chart .
> System do the calcul automatically and show only the day prod .

Does that give Pv1+Pv2 power ? Not Kwh

N1c084, 2023-09-03:
Yes kWh but the HA flow chart only use kWh or you talking about other flow
chart?

Leecavturbo, 2023-09-03:
> Yes kWh but the HA flow chart only use kWh or you talking about other flow chart? Le dim. 3 sept. 2023 à 17:23, Leecavturbo ***@***.***> a écrit :
> […](#)

Pv1 power shows the instantaneous KW if that mppt ( string ) pv2 and the combined is what I’m after in kw

ech0-py, 2023-09-15:
> > You can use total PV energy sensor for the flow chart .
> > System do the calcul automatically and show only the day prod .
> 
> Does that give Pv1+Pv2 power ? Not Kwh

`SENSORS += MathSensor((186, 187), "PV power sum", WATT, factors=(1, 1))`
That works for me, in case you wanna sum of pv1+pv2 in a single sensor

Leecavturbo, 2023-09-16:
> > > You can use total PV energy sensor for the flow chart .
> > > System do the calcul automatically and show only the day prod .
> > 
> > 
> > Does that give Pv1+Pv2 power ? Not Kwh
> 
> `SENSORS += MathSensor((186, 187), "PV power sum", WATT, factors=(1, 1))` That works for me, in case you wanna sum of pv1+pv2 in a single sensor (see [this](https://kellerza.github.io/sunsynk/reference/mysensors) for details, it requires a some hand work)

Wouldn’t even know where to start and I’ve never seen pv2 (187) as a sensor so suspect move work before your math sensor 👍

kellerza, 2023-09-16:
I've been wondering if I should not simply add an `all` group? (especially for people not reading the docs :wink:)

`pv2_power` is now part of the defaults just to show people you can add other sensors. [here](https://github.com/kellerza/sunsynk/blob/main/hass-addon-sunsynk-multi/config.yaml#L35) - do we need to add more defaults?

kellerza, 2023-09-16:
> `SENSORS += MathSensor((186, 187), "PV power sum", WATT, factors=(1, 1))` That works for me, in case you wanna sum of pv1+pv2 in a single sensor (see [this](https://kellerza.github.io/sunsynk/reference/mysensors) for details, it requires a some hand work)

Nice solution. 

If you use the power flow card you can also visualize two separate sensors and see the total in the UI - [image]

Leecavturbo, 2023-09-18:
> `SENSORS += MathSensor((186, 187), "PV power sum", WATT, factors=(1, 1))` That works for me, in case you wanna sum of pv1+pv2 in a single sensor (see [...]

got samba working and established i can create the .py from windows as notepad save as > in the share/hass-addon . but the code from 'here' link doesn't give your example. just add text below to the code ?

Leecavturbo, 2023-09-18:
to update pv1 +pv2 sum is now working with SS add-on. Sensor = pv lowercase.

Leecavturbo, 2023-09-19:
Anyone know what the sensor for UPS load is load_2 or output_2 eg ?

ech0-py, 2023-09-20:
> Anyone know what the sensor for UPS load is ? load_2 or output_2 eg ?

If you meant backup power output I guess you should read issues about all that sensors ("essentials" is a load on backup output):
https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py#L91-L104

kellerza, 2026-08-18:
with v1.1.1 there is a new direct serial driver, which hopefully performs better than the sync pyserial

closed 2026-08-18 by kellerza