kellerza/sunsynk #132: New dev release fail to boot

N1c084, 2023-04-21:
Can't create container from addon_4bb530c7_hass-addon-sunsynk-dev: 409 Client Error for http+docker://localhost/v1.42/containers/create?name=addon_4bb530c7_hass-addon-sunsynk-dev: Conflict ("Conflict. The container name "/addon_4bb530c7_hass-addon-sunsynk-dev" is already in use by container "90b8509dc5377afb7351725e7b94675d34b5c3b8e52a625768df060d5ee69b57". You have to remove (or rename) that container to be able to reuse that name.")

New release don't boot 
Thanks

kellerza, 2023-04-21:
You should be running the MULTI version of the addon.

Do you have local addons installed?

This issue seems to be related to the HA supervisor. Have you tried to uninstall/resinstall?

N1c084, 2023-04-24:
hi,
strange maybe not related but after updated (dev) my HA webserver was dead and some difficult to recover it.

I need to do a clean HA os install and now a use multi, so no more pb.

thanks

closed 2023-05-13 by kellerza